MarlinFirmware/Marlin #16354: [BUG] AutoHoming with BL Touch / 3d touch sometimes occasionally fails.

alantoy, 2019-12-27:
### Bug Description
  I am not sure its a bug, but its strange.  I have installed bl touch clone (3d touch).. and configured it like Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN.
And occasionally (not always) during AutoHome operation motor in z axis doesn't stop on time ( may be endstop signal comes too late, or pulling occures too late)  and nozzle damages my heated bed. (I have already a 2 holes, because of this problem...)
I noticed this happening oft when the nozzle is already heated. So, my 3d Touch is not working really stable during AutoHome process, most often motor for z-axis stops in time, but sometimes not.. I have no ideas, how it could be fixed....

### My Configurations

Bigtreetech SKR 3 DIP.  Actual Marlin Bugfix-2.0.x version. Creality Ender 3 with 3d touch from
"two trees".

### Steps to Reproduce

Execute M28. Autohome. Sometimes it works correctly , sometimes not... So my autohome with bl touch clone is not stable enough and can damage my heated bed.

Seref, 2019-12-31:
I had the same problem with the clone, the easy and at the same time most expensive solution is to get a real BLTouch (there really is a difference).

Though what helped me with the clone was to increase
> #define SERVO_DELAY { 750 } //originally 300

in **[Configuration.h](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Configuration.h)**, additionally  uncommenting and setting 

>  #define BLTOUCH_DELAY 1000 // originally 500

in **[Configuration_adv.h](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Configuration_adv.h)** and also rotating the screw until it detect the pin.

boelle, 2020-01-07:
@alantoy still an issue?

Vertabreak, 2020-01-14:
Have you tried enabling endstop noise threshold I had a similar homing issue that was corrected by it.

boelle, 2020-01-18:
Lack of Activity from @alantoy 
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-18 by boelle