MarlinFirmware/Marlin #16355: [BUG] 2209 stepper driver on SKR Mini E3 1.2

bigtreeteche3, 2019-12-27:
Updated to marlin 2.0.1 now the stepper started to make a lot of noise. Downgraded to 2.0.X and the printer is silent again...  ?

sjasonsmith, 2019-12-27:
Can you provide configs and the commit numbers you are using for your comparison?

bigtreeteche3, 2019-12-27:
[201.zip](https://github.com/MarlinFirmware/Marlin/files/4005977/201.zip)

bigtreeteche3, 2019-12-27:
In the zip you can find the old and quite conf +adv and the new and noisy conf+adv

Lord-Quake, 2019-12-27:
@bigtreeteche3 In `Configuration_adv.h` set this option as follows `//#define HYBRID_THRESHOLD` and try again.

bigtreeteche3, 2019-12-27:
> @bigtreeteche3 In `Configuration_adv.h` set this option as follows `//#define HYBRID_THRESHOLD` and try again.

thx men, it worked... this option was uncomment by default....

Lord-Quake, 2019-12-27:
Well, it's actually deactivated by default in Marlin 2.0.1.

However, we do have an issue when activated.

bigtreeteche3, 2019-12-27:
> Well, it's actually deactivated by default in Marlin 2.0.1.
> 
> However, we do have an issue when activated.

nope its activated... i rechecked the downloaded zip folder under the example files of  bigtreetech , skr 1.2 ...

bigtreeteche3, 2019-12-27:
now that is working i have a new issue... same configuration files as above. I have activated my bl touch and set the edge distance to 40mm but the grid its taking is more like 80mm away from the edges ... it makes a little 15x15cm square in the middle of the plate during the props... how to fix this ?

Lord-Quake, 2019-12-27:
@bigtreeteche3 Yep, you're right if those files are used.

sjasonsmith, 2019-12-28:
I can't tell whether the original issue still exists or not based on the comments. Please do not mix multiple unrelated problems in one issue.

Lord-Quake, 2019-12-28:
@sjasonsmith The original issue still exist. I was able to replicate it. Activating `#define HYBRID_THRESHOLD` on my SKR Mini E3 1.2 causes the motors to make load screeching sounds.

@bigtreeteche3 Concerning your follow-up question I would recommend you ask around on Facebook or a Forum as it looks like a none bug related question.

sjasonsmith, 2019-12-28:
> @bigtreeteche3 Concerning your follow-up question I would recommend you ask around on Facebook or a Forum as it looks like a none bug related question.

The second issue is already described in https://github.com/MarlinFirmware/Marlin/issues/16348. I just posted a PR to fix it. Any follow-up discussion should happen in that issue and not here.

sjasonsmith, 2019-12-28:
> The original issue still exist. I was able to replicate it. Activating #define HYBRID_THRESHOLD on my SKR Mini E3 1.2 causes the motors to make load screeching sounds.

I just reproduced this, although on my SKR instead of my SKR Mini E3.

The screeching sounds totally normal to me. I made some test gcode files with transition up and down across the threshold, and it all seems to be working as intended to me. People just aren't used to it because HYBRID_THRESHOLD has been opt-in until these two SKR Mini E3 examples.

It seems the SKR Mini E3 examples are the only ones which enable HYBRID_THRESHOLD. We should probably just turn that off in the examples so that people only encounter SpreadCycle noises if they intentionally enable it.

Lord-Quake, 2019-12-28:
I agree with turning them off.

I'm not to sure about the screeching sounds. It sure made me jump when the printer went in to action..... personally I wouldn't consider it normal.

pmjdebruijn, 2020-01-01:
Switching to SpreadCycle briefly scrared me too, it's much louder than you'd expect, however to me it seems this doesn't need to be an all or nothing kind of thing.

I've never had skipped steps on my machine (Ender3Pro), and yet the firmware was already switching to SpreadCycle, so to me it seems the issue isn't that Hybrid Threshold is enabled, it's more of an issue that it triggers way to quickly.

Possible something like this, might be more useful:

#define X_HYBRID_THRESHOLD 150
#define Y_HYBRID_THRESHOLD 150
#define Z_HYBRID_THRESHOLD 10
#define E0_HYBRID_THRESHOLD 60

This way when doing regular prints, SpreadCycle should never activate, unless someone is really pushing a printer hard. But, if any user still encounters skipped steps for whatever reason, the thresholds can be adjusted via the menu, without the user needing to reflash the firmware.

boelle, 2020-01-07:
@bigtreeteche3 is this still an issue?

Dracrius, 2020-01-10:
I noticed this as well and to be honest it was right after my mainboard swap so I was well use to what SpreadCycle sounded like after many a nights trying to sleep through it. A SKR mini e3 running in  SpreadCycle whether it be from the hybrid threshold or not screaches louder then the stock Ender 3 Pro firmware and board ever did. I personally raised 

#define Z_HYBRID_THRESHOLD 10

 to 100 and this eliminated the most annoying/ scary part which was during homing the z axis. It was screaching loud enough I thought something was wrong with a threshold of 10. I honestly felt then and now especially like something else is set wrong even in SpreadCycle the ender 3's stock stepper motors shouldn't screach that loud.

After changing the z threshold I now only notice SpreadCycle kick in while probing or when the head moves from one object to another but now seeing as I'm not the only one who thought it was screaching I think its worth looking into!

boelle, 2020-01-18:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-18 by boelle