MarlinFirmware/Marlin #16358: [BUG] No extrusion when printing - Skr 1.3 5xTMC2205 Chimera dual extruder

Liger0, 2019-12-27:
### Bug Description

Having 2 extruders, if I manually extrude from the LCD, they do fine. But if I start a print, the pulley just goes forth and back and no extrusion is ever done.

### My Configurations
5x 2209, skr 1,3, cr10s, Chimera dual extruder
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4006102/Marlin.zip)

### Steps to Reproduce

Start a print

**Expected behavior:** 
Print

**Actual behavior:** 
Extruder goes back and forth of the same amount

#### Additional Information
https://www.youtube.com/watch?v=EpZp1Q9OTT4

sjasonsmith, 2019-12-28:
Can you uncomment the line in Configuration_adv.h with `MINIMUM_STEPPER_PULSE` and see if that improves things?

Liger0, 2019-12-28:
I actually uninstalled everything and reinstalled again and I don't have those configs anymore.
I was trying to rebuild the firmware but it doesn't work anymore on latest stable if I use the cr10s preconfig (I used the generic one when I built the one that doesn't extrude). #16361
So it's pretty hard to retry at this point unless I am able to flash the cr10s based one and see if I still have that issue.

Liger0, 2019-12-30:
It makes no difference.

Liger0, 2019-12-31:
It seems that I needed to use an M83 to switch back to relative extrusion after the ned of the start gcode as the relative mode was enabled in s3d. It's weird, with marlin 1 it worked and I didn't need to do it.

closed 2019-12-31 by Liger0