MarlinFirmware/Marlin #16359: [QUESTION] Shouldn't NOZZLE_PARK_?_FEEDRATE be limited to DEFAULT_MAX_FEEDRATE

GMagician, 2019-12-27:
Mine is just a question. Shouldn't any feedrate value be limited to max feed rate values?

I had a config with park to 300 (default) and max feed to 90 and after a park I had layer shift

InsanityAutomation, 2019-12-28:
Probably, I thought the way the command was injected it's clamped there anyway but it's been awhile.

boelle, 2019-12-30:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Roxy-3D, 2020-01-02:
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on this page.) For best results getting help with configuration and troubleshooting, please use the following resources:

@boelle     Both GMagician and InsanityAutomation are developers....   Please go a little bit easy on them.

@GMagician     I think I've seen some corner cases where the max feed rate isn't fully respected.   I think I've seen some crazy behavior during the Auto-Park Filament change sequence.    It will be a few weeks before I can look at that deeper.    But it is very possible there are some code paths where the wrong feed rate is happening.

GMagician, 2020-01-02:
@Roxy-3D I'm may investigate on it...my was just a question to understand if that was just a "senior" developper choice or a possible bug. InsanityAutomation and you just confirmed my suspects

GMagician, 2020-01-02:
It seems that all "move" calls end up into `Planner::_populate_block` and here there is a feedrate limit check so it seems impossible that speeds go over max...I'll investigate more...

GMagician, 2020-01-14:
I can't proceed with test in a short time...delayed. If something found I'll post a PR

closed 2020-01-14 by GMagician