MarlinFirmware/Marlin #16361: [BUG] Firmware doesn't compile using Cr10s example and skr 1.3

Liger0, 2019-12-28:
### Bug Description

Firmware doesn't compile using Cr10s example and skr 1.3

### My Configurations

2.0.1 stable or bugfix as of today

### Steps to Reproduce

1. Place the 4 cr-10s files inside the Marlin folder
2. In platformio.ini
env_default = LPC1768
3. In configuration.h
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3

**Expected behavior:** Compiles

**Actual behavior:**
It doesn't compile

Problems: 
```
{
	"resource": "/c:/3d/Marlin-bugfix-2.0.x/Marlin/src/lcd/dogm/status_screen_DOGM.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "unused variable 'isHeat' [-Wunused-variable]",
	"startLineNumber": 238,
	"startColumn": 16,
	"endLineNumber": 238,
	"endColumn": 16
}
```

Red/Yellow warnings
```

Marlin\src\lcd\dogm\status_screen_DOGM.cpp: In function 'void _draw_bed_status(bool)':
Marlin\src\lcd\dogm\status_screen_DOGM.cpp:238:16: warning: unused variable 'isHeat' [-Wunused-variable]
     const bool isHeat = BED_ALT();
In file included from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/./HAL_LPC1768/HAL.h:45,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/MarlinConfig.h:30,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../ui_api.h:45,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../compat.h:31,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../config.h:24,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../compat.h:20,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../ftdi_extended.h:25,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_bitmaps.cpp:22:
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/./HAL_LPC1768/MarlinSerial.h:27:10: fatal error: ../../inc/MarlinConfigPre.h: No such file or directory     
 #include "../../inc/MarlinConfigPre.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/./HAL_LPC1768/HAL.h:45,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/MarlinConfig.h:30,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../ui_api.h:45,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../compat.h:31,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../config.h:24,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../compat.h:20,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../ftdi_extended.h:25,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\standard_char_set.cpp:22:
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/./HAL_LPC1768/MarlinSerial.h:27:10: fatal error: ../../inc/MarlinConfigPre.h: No such file or directory     
 #include "../../inc/MarlinConfigPre.h"
Compiling .pio\build\LPC1768\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\marlin_events.cpp.o
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[... 20 lines cut ...]
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../config.h:24,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../compat.h:20,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../ftdi_extended.h:25,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\unicode.cpp:22:
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/./HAL_LPC1768/MarlinSerial.h:27:10: fatal error: ../../inc/MarlinConfigPre.h: No such file or directory
 #include "../../inc/MarlinConfigPre.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1768\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_size_t.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\unicode.cpp.o] Error 1
In file included from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/./HAL_LPC1768/HAL.h:45,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/MarlinConfig.h:30,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../ui_api.h:45,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../compat.h:31,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../config.h:24,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../compat.h:20,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../ftdi_extended.h:25,
                 from Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\western_char_set.cpp:22:
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\../../../../../../../inc/../HAL/./HAL_LPC1768/MarlinSerial.h:27:10: fatal error: ../../inc/MarlinConfigPre.h: No such file or directory     
 #include "../../inc/MarlinConfigPre.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1768\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\western_char_set.cpp.o] Error 1
```

ellensp, 2019-12-28:
Replicated this, It compiled fine. It does have the two warnings warning: unused variable 'isHeat'  and warning: 'u8g_i2c_opt' defined but not used.   Did you use vscode? Did you do platformio:Clean (trashcan icon in vscode, along the bottom)

Liger0, 2019-12-28:
Yes I did, It still gives me the same errors. I use Vscode of course.

ellensp, 2019-12-28:
Tried again on windows this time..  compiled first time. Zero issues.  Cannot reproduce this error.

Liger0, 2019-12-28:
Weird. If I try to compile the cr10s example directly I "just" have 45 problems in the problems tab but it compiles fine. 
If I select the skr 1.3 it triggers all of those warnings in the terminal and doesn't build at all.

ellensp, 2019-12-28:
cr10s example config also builds fine without any modifications...   with 2 warning.

Liger0, 2019-12-28:
It seems that using a shorted directory fixed it.

closed 2019-12-28 by Liger0

ugurgudelek, 2020-01-17:
shorter directory fixed my issues too

FDCoreV2, 2020-03-07:
Im facing a pretty similar issue at the moment, would you please explain what exactly you did?

thinkyhead, 2020-03-07:
> Im facing a pretty similar issue at the moment, would you please explain what exactly you did?

Delete all the HAL sub-folders that you don't need. If that isn't enough, move the Marlin folder closer to the root of your drive.