MarlinFirmware/Marlin #16366: Delta homing issues

DickyMcCockpants, 2019-12-28:
Backoff is structured to occur after each individual axis is homed. Deltas must home every axis before backing off (because axis are only set as homed on completion). The Backoff code initiates after each axis and for deltas since the home position has not been set, the result of the move calculation will attempt to drive the carriage further into the endstops. Working on a fix now.

boelle, 2019-12-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

DickyMcCockpants, 2019-12-31:
Since homing is handled by delta.cpp calling the homeaxis function of motion.cpp three times (once per tower/stepper-axis), and the backoff is handled inside the homeaxis function - both needed modification.

I put a conditional to skip backoff for delta in the homeaxis function (backing off still happens per-axis for Cartesian) as well as re-adding such code to delta.cpp after all three stepper-axis are homed.

I also finalized the homing in delta.cpp by calling set_axis_is_at_home and sync_plan_position; these are currently NOT called upon completion of homing by deltas because the placement of the calls in motion.cpp is nested within a conditional that excludes deltas - making homing impossible without this or similar code changes.

This modification is a quick fix and should be re-written for better object orientation. 
The two modified files are in the attached zip.

[Delta-Homing-fix.zip](https://github.com/MarlinFirmware/Marlin/files/4012883/Delta-Homing-fix.zip)

DickyMcCockpants, 2019-12-31:
Sorry for going balls deep instead of giving you a config file.

sjasonsmith, 2019-12-31:
Config files are still useful for anyone looking to verify tour change. If you’ve already solved the problem it would make sense to open a PR for the change.

DickyMcCockpants, 2019-12-31:
I used the delta example config file. Verification should be done by comparison with existing code and testing.

boelle, 2020-01-07:
@DickyMcCockpants did you solve this one?

DickyMcCockpants, 2020-01-07:
I guess so, it works for me.

sjasonsmith, 2020-01-07:
But it isn't actually fixed for everyone else yet, right? I think this one should probably stay open until the change is actually posted and merged. The user who filed #16445 may be looking into that.

boelle, 2020-01-14:
@Sewyboowy do this issue make a bell ring for you?

DickyMcCockpants, 2020-01-21:
Just look at the diff between my files and bug fix and commit my changes.

closed 2020-01-21 by DickyMcCockpants

sjasonsmith, 2020-01-23:
> Just look at the diff between my files and bug fix and commit my changes.

One major reason for wanting people to post pull requests for their own fixes, is that git can help with all the merging that has to happen. I just applied your changes to one of my branches, and I had to manually merge your changes with hundreds of other modified lines of code in those files, and guess which changes were yours.

I was able to test the resulting code on a delta, but if your fix doesn't align with what another willing developer happens to be sitting next to, they probably won't spend their time to apply your change without the ability to verify it.