MarlinFirmware/Marlin #16369: [FR] feedrate option for FILAMENT_UNLOAD_PURGE_LENGTH

rado79, 2019-12-28:
### Description

first of all i realy enjoy the ADVANCED_PAUSE_FEATURE (Bowden)
would be helpful to make an option to set a feedrate for: FILAMENT_UNLOAD_PURGE_LENGTH

### Feature Workflow

feedrate for FILAMENT_UNLOAD_PURGE_LENGTH is DEFAULT_MAX_FEEDRATE E:

// Quickly purge
  do_pause_e_move((FILAMENT_UNLOAD_PURGE_RETRACT + FILAMENT_UNLOAD_PURGE_LENGTH) * mix_multiplier,
                  planner.settings.max_feedrate_mm_s[E_AXIS] * mix_multiplier);

would be nice to have a configurable feedrate in cofiguration_adv, because if using high extruder feedrate, acceleration and high torque on stepper - DEFAULT_MAX_FEEDRATE E0 is harmful to bowden, fittings, filament, gears,... - in case of hardened Filament in heatbreak after retract and 5 seconds pause.

rado79, 2019-12-28:
my provisional solution: 

pause.cpp
// Quickly purge
  do_pause_e_move((FILAMENT_UNLOAD_PURGE_RETRACT + FILAMENT_UNLOAD_PURGE_LENGTH) * mix_multiplier,
                  (**PAUSE_PARK_RETRACT_FEEDRATE**) * mix_multiplier);

yet-another-average-joe, 2019-12-29:
The filament must harden in the heatbreak, and then be pushed + retracted as fast as possible in order to ram purge, or it will not be ram purging.

Doing this at PAUSE_PARK_RETRACT_FEEDRATE will do nothing special, as the filament will have time to melt again. It is simpler to set FILAMENT_UNLOAD_PURGE_RETRACT and FILAMENT_UNLOAD_PURGE_LENGTH to zero (deactivate ram purging).

But having "FILAMENT_UNLOAD_PURGE_FEEDRATE" and "FILAMENT_UNLOAD_PURGE_ACCEL" parameters wouldn't hurt.

So, better add a parameter for feedrate in Configuration_adv.h, and pass it to do_pause_e_move().
 
(i too enjoy the ADVANCED_PAUSE_FEATURE)

rado79, 2019-12-29:
> The filament must harden in the heatbreak, and then be pushed + retracted as fast as possible in order to ram purge, or it will not be ram purging.
> 
> Doing this at PAUSE_PARK_RETRACT_FEEDRATE will do nothing special, as the filament will have time to melt again. 

i don´t like to go lower than 25-40mm/s. but if you use a very high extruder feedrate/acceleration for linear advanced, it would make sence to lower down to a normal default feedrate. 
i think 21mm Filament (retract 13mm +purge 8mm) in less than a half second is stress to mechanics, makes filament curled between extruder and nozzle (bowden)

rado79, 2020-06-24:
meanwhile integrated in the code

closed 2020-06-24 by rado79