MarlinFirmware/Marlin #16370: [BUG] G29 auto bedleveling ignore M121 disable endstop and failed after hit the endstop

NetrunnerAT, 2019-12-28:
### Bug Description

sorry for my bad english!

If i use G29 it fail after hit the endstop, with activated M121 disabled endstop!

### My Configurations

Mechanical Config:
I use opto endstop and a over size heatbed. It mean ... z_min endstop is over the heatbed and nozzle dont touch the bed!

Marlin bugfix-2.0.x configuration.h:
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 5 <- endstop is 5mm over heatbed

//#define MIN_SOFTWARE_ENDSTOPS

#define NOZZLE_TO_PROBE_OFFSET { 0, 0, 0 }
#define Z_CLEARANCE_DEPLOY_PROBE    5 
#define Z_CLEARANCE_BETWEEN_PROBES  5 
#define Z_CLEARANCE_MULTI_PROBE     5 

#define AUTO_BED_LEVELING_BILINEAR

### Steps to Reproduce

1. G28
2. M121
3. G29

If i disable my endstop mechanical. G29 work! It mean ... G29 use endstop handling and ignore M121. 

My question. How i can disable endstop handleing in G29 code or make it responsible of M121 disable endstop?

boelle, 2019-12-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

XylenC4, 2020-01-03:
I've got the same issue. Attached you can find my configuration. It's an BLV MGN Printer with SKR1.3 and TMC2209. I'm using Dual Z Motor and Dual Z Endstops. The Endstops are above zero so i defined #define MANUAL_Z_HOME_POS 4.3.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4019926/Marlin.zip)

boelle, 2020-01-07:
closing until OP provide configs

closed 2020-01-07 by boelle