MarlinFirmware/Marlin #16371: Homing failed with Marlin 2.0.x

SPippo38, 2019-12-28:
### Bug Description
Homing action failed when nozzle if not close to endpoints.
Issue is the same using G28 or AutoHome from control panel.
If nozzle is close to endpoints then there is no issue, homing is done correctly.

Thanks for your help

### My Configurations
Enclosed both conf and conf_adv files
[Conf+conf_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4007454/Conf%2Bconf_adv.zip)

### Steps to Reproduce
1. G28 or selecting AutoHome from control panel with nozzle far from endpoints
Behavior :
Nozzle is starting to move, trying to reach the home position but stops after a short move and control panel mentions "Homing failed"
(Only reset button can be used at this moment)
This is not happening if nozzle is let's say around 2/3 cm from home position.

#### Additional Information
Printer is Geeetech/MeCreator2

Liger0, 2019-12-29:
It happened to me on my skr 1.3 with generics config, but instead it will just take as home position the point where it initially is.

Liger0, 2019-12-29:
So far it seems only the first time after flashing the firmware the X is able to home, then after saving the eeprom and restarting it never homes the X, it just moves a bit on the right.

yet-another-average-joe, 2019-12-30:
I also have this problem (randomly) and with 2.0.x only. Everything works fine for 2 weeks, and then could fail two times winthin five minutes, generally when the hotend is far from the endpoints (but not always and not only). I never reported this issue because I can't reproduce it. I've been using 2.0.x for 16-18 months, and since then had the issue from time to time, exactly as described by the video..

I suspect some interference : I always home/level with bed and hotend being heated. On CNC machine tools, endstop and motor wirings are usually shielded. Not printers. Maybe capacitors soldered on the switches could help.

2.0.x only : could the debouncing rountine be different from 1.1.x ?

AnHardt, 2019-12-30:
> could the debouncing rountine be different from 1.1.x ?

Yes. Very different. And by default, now, there is **NO** debouncing.

Try to activate
`#define ENDSTOP_NOISE_THRESHOLD 2`
or, if you board has no hardware lowpass on the endstop inputs try to add them.

Liger0, 2019-12-30:
> `#define ENDSTOP_NOISE_THRESHOLD 2`

It seems it worked!

Liger0, 2020-01-01:
Now sometimes it is happening that instead the Y is skipped, but usually it goes fine.

SPippo38, 2020-01-02:
Seems working for me but I don't understand why since my endstops got a capacitor installed

Liger0, 2020-01-03:
I installed the latest bugfix (of some minutes ago), with the options:

```
#define ENDSTOP_NOISE_THRESHOLD 2

#define INVERT_X_STEP_PIN true
#define INVERT_Y_STEP_PIN true
#define INVERT_Z_STEP_PIN true
#define INVERT_E_STEP_PIN false

#define MINIMUM_STEPPER_PULSE 1
```
that I all have to use because of both this bug and a bug that prevents my extruder from working, but now differently that 2.0.1 stable it's the Y axis never working...

SPippo38, 2020-01-05:
Having no more issue since several days, I consider this as fixed.
Thanks for your support guys

closed 2020-01-05 by SPippo38