MarlinFirmware/Marlin #16373: [BUG] Software endstops are ignored

phate191, 2019-12-29:
### Bug Description

After Homing with Software Endostops enabled, you can move out of the software endostops boundaries.

### My Configurations

Ender 3 Pro
Version 2.0.1, commit 432decd513f14d8b50ca4fec8b53e8cd9ff825b6
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4007646/Marlin.zip)

### Steps to Reproduce

1. M211 //Check if Software Endsotops are Enabled
2. Home XY [G91 | G28 X0 Y0 | G90]
3. M211 //Check, Still Enabled
4. G1 Y-1 F4000 // Move Y and the axis move without problem...

With Software Endstops enabled the axes after home should be stopped to go below MIN_SOFTWARE_ENDSTOP_*

From Octoprint Terminal:
> Send: M211
> Recv: echo:Soft endstops: ON  Min:  X0.00 Y0.00 Z0.00   Max:  X235.00 Y235.00 Z250.00
> Recv: ok
> [...]
> Send: G91
> Recv: ok
> [...]
> Send: G28 X0 Y0
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: X:0.00 Y:0.00 Z:10.00 E:0.00 Count X:0 Y:0 Z:4000
> Recv: ok
> Send: G90
> Recv: ok
> [...]
> Send: M211
> Recv: echo:Soft endstops: ON  Min:  X0.00 Y0.00 Z0.00   Max:  X235.00 Y235.00 Z250.00
> Recv: ok
> [...]
> Send: G1 Y-1 F4000
> Recv: ok

I've also tried to disable with M211 S0 and the repeat; the beavior is the same (and it's ok).

ellensp, 2019-12-29:
Endstops are ignored for all but homing commands. unless you enable
#define ENDSTOPS_ALWAYS_ON_DEFAULT in Configuration_adv.h

phate191, 2019-12-29:
Yes true, but phisical endstops must be separated from software endstops.
Theoretically they have different function.
If I remeber correctly, in old versions, software endostops if are enabled are always enforced.

tatusah, 2019-12-30:
Different board but for me soft endstops do work correctly with Marlin 2.0.1. What does M114 report as position after you issue the `G1 Y-1 F4000`?

phate191, 2019-12-30:
This is the output from Octoprint console:

> [...]
> Send: G1 Y-1 F4000
> Recv: ok
> [...]
> Send: M114
> Recv: X:0.00 Y:-1.00 Z:5.00 E:0.00 Count X:0 Y:-80 Z:2000
> Recv: ok

tatusah, 2019-12-30:
Does the LCD controls also allow going past soft endstops or is it only trough Octoprint?

phate191, 2019-12-30:
Ok I think I find the issue.
This is the ouput from octoprint if I only home X and Y:

> [...]
> Send: G91
> Recv: ok
> Send: G28 X0 Y0
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: X:0.00 Y:0.00 Z:10.00 E:0.00 Count X:0 Y:0 Z:4000
> Recv: ok
> Send: G90
> Recv: ok
> [...]
> Send: M114
> Recv: X:0.00 Y:0.00 Z:10.00 E:0.00 Count X:0 Y:0 Z:4000
> Recv: ok
> [...]
> Send: G1 Y-1 F4000
> Recv: ok
> [...]
> Send: M114
> Recv: X:0.00 Y:-1.00 Z:10.00 E:0.00 Count X:0 Y:-80 Z:4000
> Recv: ok

But this is the ouput after home of X, Y and **Z**:

> [...]
> Send: G91
> Recv: ok
> Send: G28 X0 Y0
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: X:0.00 Y:0.00 Z:10.00 E:0.00 Count X:0 Y:0 Z:4000
> Recv: ok
> Send: G90
> Recv: ok
> Send: G91
> Recv: ok
> Send: G28 Z0
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: echo:busy: processing
> [...]
> Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
> Recv: ok
> Send: G90
> Recv: ok
> [...]
> Send: M114
> Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
> Recv: ok
> [...]
> Send: G1 Y-1 F4000
> Recv: ok
> [...]
> Send: M114
> Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
> Recv: ok

So, if you home only X and Y, and then you move X or Y, software and hardware endstops are ignored for ALL axis :(

tatusah, 2019-12-30:
Well, there you are correct. Sorry, did only test general G28 previously. Software endstops are not enabled until all axis are homed and thus movement past endstops is possible.

tatusah, 2019-12-30:
And I remembered that I saw this thing before also. The issue connects to a bigger picture. It was last raised here https://github.com/MarlinFirmware/Marlin/issues/16283 so I think this needs some discussion about what if anything should be done about the situation.

kurtnelle, 2020-01-03:
I have confirmed that the software endstops are on but the joystick still allows the planner to send the axis beyond the frame of the machine.

boelle, 2020-01-07:
@phate191 still an issue?

kurtnelle, 2020-01-08:
I had to update to the lastest version of marlin. I think the software endstops work now. However the behaviour with the joysticks is so erratic, that I'm not sure if it actually is working.

kurtnelle, 2020-01-12:
I've tested it some more. Its not working.

boelle, 2020-01-18:
did you test it before or after pr #16533 was merged?

boelle, 2020-01-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

we can reopen if we hear from @phate191 again

kurtnelle, 2020-01-24:
@boelle , I haven't tested it after #16533 cause I can't figure out how to merge!

boelle, 2020-01-24:
@kurtnelle just download from https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

closed 2020-01-24 by boelle

italocjs, 2020-01-30:
Sorry for reopening, but i have noticed the same issue if using G90/G91 (moving as relative coordinate), the endstops (both phisical and software) are being ignored,  while this wont affect the printer, it i try to move the head around i can crash it.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4134667/Marlin.zip)

I have attached my config to see if someone can help me spot why,  this is a 2.0.x bugfix, from around 05/01/2020.

italocjs, 2020-01-30:
I have the same issue, i'll try burning the latest bugfix (from today) to verify if it was solved

italocjs, 2020-01-30:
I can confirm that #16533  has fixed the issue here!, thanks