MarlinFirmware/Marlin #16376: [BUG] SKR PRO Serial 3

akarnaukh, 2019-12-29:
When I want to select serial 3, a link error occurs.
#define SERIAL_PORT 1
#define SERIAL_PORT_2 3

Linking .pio/build/BIGTREE_SKR_PRO/firmware.elf
.pio/build/BIGTREE_SKR_PRO/src/src/Marlin.cpp.o: In function manage_inactivity(bool)': /Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/Marlin.cpp:483: undefined reference to Serial3'
.pio/build/BIGTREE_SKR_PRO/src/src/Marlin.cpp.o: In function setup': /Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/Marlin.cpp:1120: undefined reference to Serial3'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function serialprintPGM(char const*)': /Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/core/serial.cpp:37: undefined reference to Serial3'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function serial_echopair_PGM(char const*, char const*)': /Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/core/serial.cpp:41: undefined reference to Serial3'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o: In function serial_echopair_PGM(char const*, int)': /Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/core/serial.cpp:43: undefined reference to Serial3'
.pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o:/Users/akarnaukh/SKR_PRO/Marlin/Marlin/src/core/serial.cpp:44: more undefined references to `Serial3' follow

While switched to Serial 6.
But it's inconvenient, how do I fix it?

boelle, 2019-12-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

akarnaukh, 2020-01-01:
[Configuraion.zip](https://github.com/MarlinFirmware/Marlin/files/4014170/Configuraion.zip)

akarnaukh, 2020-01-01:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4014173/Configuration.zip)
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = BIGTREE_SKR_PRO

[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
build_flags = -fmax-errors=5 -g -D__MARLIN_FIRMWARE__ -fmerge-all-constants
lib_deps =
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  LiquidCrystal@1.3.4
  #TMCStepper@>=0.5.2,<1.0.0
  https://github.com/bigtreetech/TMCStepper
  Adafruit NeoPixel@1.2.5
  Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
  SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip

LinoBarreca, 2020-01-03:
I'm gonna push a fix for this.

LinoBarreca, 2020-01-03:
done.

akarnaukh, 2020-01-03:
With these changes, it build. 
But it doesn't work. There is no data in serial3.

LinoBarreca, 2020-01-03:
how are you sending the data in there?

akarnaukh, 2020-01-03:
USB-UART when Serial6 has responses from the Board to the m503 or M122 commands. There is no response to Serial3
RX and TX checked, everything is connected correctly

LinoBarreca, 2020-01-03:
I didn't get what you are trying to achieve.
you send the command through USB (Serial1) and expect it into Serial3?

LinoBarreca, 2020-01-03:
and with
serial_port 3
serial_port_2 -1
everything works as expected after the PR...
[image]

so...again...what are you trying to do?
are you on latest marlin sources or are you doing a weird merge of files and branches? because from what you wrote above it looks like you are using an outdated version from btt

akarnaukh, 2020-01-03:
No. I'm sending to Serial3 and waiting for a response there.
I don't use USB, it doesn't work on Mac OS for some reason. But this is a MacOS problem.
serial_port 1
serial_port_2 3
Maybe You're right and I'm using the wrong branch. 
I'll try https://github.com/LinoBarreca/Marlin/tree/HWSerials

LinoBarreca, 2020-01-03:
> 
> I don't use USB, it doesn't work on Mac OS for some reason. But this is a MacOS problem.

but before
> > how are you sending the data in there?
>
> USB-UART when Serial6 has responses from the Board to the m503 or M122 commands. 

anyway... SERIAL 1 IS the serial on the TFT connector. Is this what you really want to do?
..because I still didn't understand what you are trying to do...

LinoBarreca, 2020-01-03:
> I'll try https://github.com/LinoBarreca/Marlin/tree/HWSerials

@akarnaukh ok but ensure to comment line 1132 in /Marlin/src/Marlin.cpp

akarnaukh, 2020-01-04:
Yes, I use Serial 1 to connect to the computer instead of USB. The fact that it is signed by TFT in my opinion does not change the fact that it is just a Serail port.
Serial3 is more convenient for me to use to connect Onoff WiFi. This is more convenient, I only need TX and RX. Since I can't use Serial 3 at the moment, I'm using Serial 6. It's not critical and everything works.
But in my opinion there should be no error when choosing Serial 3.

LinoBarreca, 2020-01-04:
As far as I have understood you own a mac and you talk to the board through serial 1. How exactly? Does your mac have a serial port? Every serial (1,3,6), in that sense, are equal to the interfacing methods. They all have 2 pins+gnd. 
I agree with you that if there's something not working it must be fixed but I am not honestly understanding what doesn't work and in which situation. I could only reproduce the bug about not being able to compile which was what you reported in the op and which has been fixed.
If there's another bug tell us/me how to reproduce it exactly and we/I will look at it.

AnHardt, 2020-01-04:
Do you have a similar to "-DHAVE_HWSERIAL6" clause for Serial3 in Platform.ini?
What exact pins do you want to use with Serial3?

LinoBarreca, 2020-01-07:
@boelle you can close this 'cause the fix has been merged.

akarnaukh, 2020-01-07:
> As far as I have understood you own a mac and you talk to the board through serial 1. How exactly? Does your mac have a serial port? Every serial (1,3,6), in that sense, are equal to the interfacing methods. They all have 2 pins+gnd.

@LinoBarreca USB/UART CP2102

@AnHardt The first time I pointed it out, but it only allowed the firmware to compile, but not to work as it should.
[image]

akarnaukh, 2020-01-07:
There is no way to check the latest build yet. Be sure to write how the result will be.

closed 2020-01-07 by boelle