MarlinFirmware/Marlin #16377: [BUG] RUMBA32 - TMC5160 - SPI Problem

BoHoehne, 2019-12-29:
### Bug Description

Unable to control Motors, 

### My Configurations

Software: 
Marlin 2.0.x build as of  29.Dec.2019

Hardware: 
RUMBA32 with 6x TMC5160 (scl PIN removed),
Configuration has no SD-CARD, no Display, no EEPROM
Software SPI is not configured.
Powered from PSU with 24V (all sockets connected and checked)

M122 shows the following:
READ: 		X	Y	Z	E	E1	E2
READ: Enabled		true	true	true	true	true	true
READ: Set current	600	700	800	950	1000	1050
READ: RMS current	589	695	789	942	990	1049
...
READ: stealthChop	false	false	false	false	false	false
READ: msteps		256	256	256	256	256	256
...
READ: Stallguard thrs	15	15	15	0	0	0
...
READ: Driver registers:
READ: 		X	0x00:00:00:09
READ: 		Y	0x00:00:00:09
READ: 		Z	0x00:00:00:09
READ: 		E	0x00:00:00:09
READ: 		E1	0x00:00:00:09
READ: 		E2	0x00:00:00:09
READ: 
READ: 
READ: Testing X connection... OK
READ: Testing Y connection... OK
READ: Testing Z connection... OK
READ: Testing E connection... OK
READ: Testing E1 connection... OK
READ: Testing E2 connection... OK

Clearly the connection is not OK.
The motors don't move.
Cannot change to Stealthchop.

I ran out of options. Any help is appreciated.

### Steps to Reproduce

1. Have the Hardware
2. Compile Marlin (with configs provided)
3. Connect via USB and check M122

**Expected behavior:** [What you expect to happen]

Ability to control the stepper motors.

**Actual behavior:** [What actually happens]

Not able to controll stepper motors.

#### Additional Information
Config files:

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4008097/Marlin.zip)

BoHoehne, 2020-01-04:
The problem persists. Still unable to get the RUMBA32 to run.

Anyone seen this before ?

boelle, 2020-01-24:
i dont have same hardware so cant test, but how often do you update? bugfix is updated almost daily

could be that it has fixed it without anyone telling

boelle, 2020-01-30:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-30 by boelle