MarlinFirmware/Marlin #16378: [BUG] LCD 20x4 text moving constantly

jonathan-iap, 2019-12-29:
### Bug Description

The text on my LCD is constantly moving and it doesn't have the right indentation like if an overflow occur.

### My Configurations
Hardware : 
Printrbot plus v2-1
Motherboard printrboard revD
LCD reprap 20x2 
Software : 
Marlin 2.0 and bug fix
[conf_Marlin2_printrbotPlus.zip](https://github.com/MarlinFirmware/Marlin/files/4008200/conf_Marlin2_printrbotPlus.zip)

### Steps to Reproduce
1. upload firmware with joined conf.h files or just uncoment 
#define REPRAP_DISCOUNT_SMART_CONTROLLER 
in LCD character section 
2. show the result

**Expected behavior:** [What you expect to happen]
Boot screen appear correctly fallow to the menu with clear informations.  
**Actual behavior:** [What actually happens]
At boot screen the indentation is incorrect and the logo doesn't appear correctly. 
After, menu is showing but is constantly moving and mixing with unknown character like an overflow.
 
#### Additional Information
[image]
[image]
[image]
[image]

ellensp, 2020-01-01:
How did you compile it Arduino IDE or VSCODE? Are all your libraries and Boards up to date?
If Arduino IDE what version of arduino?

ellensp, 2020-01-01:
Also since it looks like a timing thing can you confirm what chip your board has? Most printrboards have a 16mhz AT90USB1286 but some have a faster 20mhz AT90USB1287

jonathan-iap, 2020-01-01:
Hi ellensp, i didn't  try with Arduino but  today it's work with Arduino 1.8.10 and board with 16Mhz crystal. 
The bug report is with PlatformIO and VScode.

jonathan-iap, 2020-01-01:
So PlatformIO download automatically all libraries needed, i don't how to force version ?

jonathan-iap, 2020-01-01:
I'm thinking i can continue with Arduino IDE but when i turn #define SDSUPPORT
My LCD display blank square, i think is an other bug in relation with the space memory.

ellensp, 2020-01-02:
sadly processors with less than 256MB of flash (such as AT90USB1286 and sanguino based boards) are just to small for modern firmwares.  It is really time to consider a nice 32bit upgrade

jonathan-iap, 2020-01-02:
So with the Marlin 1.1.9 i can use the LCD, SDSUPPORT, EEPROM and bed leveling (just for comparison). My Flash with the bootloarder is close to 99% but it's work.
With Marlin 2.0 LCD is not supported correctly when you compile from platformIO and if you pass by Arduino and you active only SDSUPPORT with LCD, LCD is lost.
I don't think all my problems is in like with space memory but Marlin 2.0 is not for my setup lol.

boelle, 2020-01-07:
@jonathan-iap is this one solved in that you know  your hardware is not up for it?

jonathan-iap, 2020-01-08:
yes you can close my issue

closed 2020-01-08 by jonathan-iap