MarlinFirmware/Marlin #16379: [BUG] 2.0 Does not compile with TMC2208

sergiyd, 2019-12-29:
### Bug Description

2.0.x does not compile with TMC2208.

> avr-gcc: error: CreateProcess: No such file or directory
> 
> Multiple libraries were found for "U8glib.h"
>  Used: C:\Users\dobyc\Documents\Arduino\libraries\U8glib
> Multiple libraries were found for "TMCStepper.h"
>  Used: C:\Users\dobyc\Documents\Arduino\libraries\TMCStepper
> Multiple libraries were found for "SPI.h"
>  Used: C:\Users\dobyc\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\SPI
> Multiple libraries were found for "SoftwareSerial.h"
>  Used: C:\Users\dobyc\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\libraries\SoftwareSerial
> exit status 1
> Error compiling for board Arduino Mega or Mega 2560.

### My Configurations

Arduino IDE 1.8.10/1.9.0-beta
RAMPS 1.4 + Mega 2560

#### Additional Information

1.1.x compiles and runs fine.
Set _build.path_ in arduino config didn't help.
Arduino folder in drive's root (path does not contain spaces).

ellensp, 2019-12-30:
This is most likely the windows command line length...  In Marlin/src/HAL remove the unused HAL directories, only keep "HAL_AVR" and "shared" directories. Don't remove the two files in the HAL root directory.

boelle, 2019-12-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

sergiyd, 2019-12-31:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4012170/Marlin.zip)
@ellensp thanks! your solution works.
@boelle configs attached

LinoBarreca, 2020-01-03:
@sergiyd then if you found a solution close the bug.

boelle, 2020-01-05:
seems solved, closing

closed 2020-01-05 by boelle