MarlinFirmware/Marlin #16381: g33 delta probe fails on 2nd interation & only probes 3 points

mtraven, 2019-12-29:
I am working on my scratch built delta, it runs a mega with ramps1.4.  I had been running bugfix1.1.9, printer ran well for the last year or so.  Now that 32bit ARM is now supported, I thought I would upgrade to bugfix2_0_x (downloaded dec29, 2019) and see how the new version is.  I have not yet upgraded to a 32bit controller, so everything is exactly the same as before. All the delta settings are previously known values and I have tried running it with endstop & tower angles reset back to zero, makes no difference.

### Bug Description
g33 only probes 3 points (regardless of fw setting, which is currently 7)
g33 probing error on second iteration, 4th point

### My Configurations
[marlin2_0_bf-confgs.zip](https://github.com/MarlinFirmware/Marlin/files/4008533/marlin2_0_bf-confgs.zip)

### Steps to Reproduce
1. start printer
2. g33 p1   -fine tune height
3. g33 p2 v2 f10

**Expected behavior:** should run 10 iterations of g33 p2, probing 7 points (plus ctr) on each.

**Actual behavior:**  iteration#1 probes only 3 points, but does it successfully, returns home to start next iteration, successfully probes the center & 1st two points, then "probing fails" at the point inline with the y axis.  Machine is still powered & responsive at this point.

#### Additional Information
when I first got the new firmware onto the board, g33 seemed to be failing on the 2nd perimeter point during the 1st iteration.  However, this was not a "probing error".  The machine seemed to just quit and let it drop onto the bed.  Some trial & error led me to figure out it was watchdog restarting the printer.  I thought it might be my DELTA_SEGMENTS_PER_SECOND , which I had bumped up to 220 (worked fine with 1.1.9).  So I put that back to the conservative "190" and saw no change.  So I finally disabled watchdog until I figure out a work around.  And since then, the problem is only as I described in the "actual behavior" section.

Additionally, my probe seems to be working just fine (confirmed by successful m48 & g30 probe calls).  

I have tried expanding the height by setting it via m665, no change in operation.

tried changing the probing radius, which does change the probe positions, but g33 still fails at the same point.

While I wait to hear what you all say, I think I will reload 1.1.9 just to confirm I haven't messed something up.

ps this is my first bug report, I have tried to comply with the standards set forth, but the "code of conduct" link that shows up in this window gives me a 404 not found, so I wasn't able to actually read that. So apologizes if I am not quite in compliance.

If additional information or clarification is required, I am happy to provide it, just ask.

mtraven, 2019-12-29:
ok, I am an idiot....g33p2 is just the towers, so the 3 points is what I should expect. However, the probe failure still remains.  Rolling back to 1.1.9, g33 p2 v2 probes without issue:
g33 p2 v2 
16:56:41.575 : G33 Auto Calibrate
16:56:41.575 : Checking... AC
16:56:41.575 : .Height:591.44  Ex:+0.00  Ey:-0.65  Ez:-0.98  Radius:322.99
16:57:08.977 : Iteration : 01                                std dev:0.070
16:57:08.977 : .Height:591.51  Ex:+0.00  Ey:-0.71  Ez:-1.11  Radius:325.04
16:57:43.995 : Iteration : 02                                std dev:0.021
16:57:43.995 : .Height:591.53  Ex:+0.00  Ey:-0.66  Ez:-1.06  Radius:324.42
16:58:19.002 : Iteration : 03                                std dev:0.021
16:58:19.002 : .Height:591.49  Ex:+0.00  Ey:-0.70  Ez:-1.12  Radius:324.59
16:58:54.000 : Iteration : 04                                std dev:0.008
16:58:54.000 : .Height:591.51  Ex:+0.00  Ey:-0.65  Ez:-1.09  Radius:324.42
16:59:28.957 : Calibration OK                                rolling back.
16:59:28.957 : .Height:591.49  Ex:+0.00  Ey:-0.70  Ez:-1.12  Radius:324.59
16:59:28.973 : Save with M500 and/or copy to Configuration.h

i saw somewhere a g33 A  command, what is the A?  it doesn't seem to be referenced in the marlin gcode guide.  Also, I thought there used to be a link to a more detailed explanation of g33, anyone know what I am talking about?

sjasonsmith, 2019-12-29:
I reworked the delta probing quote quite a bit recently, to better handle offset and dynamically configured probes. Part of this was that it now automatically calculates the calibration probing radius, rather than using a hard-coded value from a calibration file. If it is aborting early and unexpectedly most likely it is considering one of the calculated probe points unreachable. I thought we had dealt with the rounding issues impacting this, but it is possible that in certain edge cases it is still not quite right.

Unfortunately that doesn't currently provide useful prints when it happens, adding those is something I have been meaning to get around to.

mtraven, 2019-12-30:
perhaps we are using different terminology, but I wouldn't say its aborting early.  There is a probe failure that terminates the process.  here is an example running g33 p3 v2 f5:
20:50:08.180 : G33 Auto Calibrate
20:50:08.180 : Checking... AC
20:50:08.180 : .Height:592.34  Ex:-0.61  Ey:-0.04  Ez:+0.65  Radius:323.41
20:50:08.180 : .               Tx:+0.00  Ty:+0.00  Tz:+0.00
20:50:40.039 : Iteration : 01                                std dev:0.436
20:50:40.039 : .Height:591.34  Ex:-0.06  Ey:-2.04  Ez:+0.00  Radius:323.34
20:50:40.039 : .               Tx:-0.30  Ty:+0.00  Tz:+0.30
20:51:06.958 : Error:Probing Failed
20:51:06.958 : Correct delta settings with M665 and M666

This occurs the same way if g33 p2 is used.  So it gets through its first iteration and most of the second. In fact both p2 fails at the last point  and p3 fails at the second to last point, which appears to be the same as the last point for p2 (ie probing fails in the same physical location) p3 have probing failures on the final point for that iteration.  For me that is the point on the +y axis.

I really havn't looked at the g33 code very much, so this is very speculative. but it SEEMS like the process runs through its 1st iteration fine then chooses new endstop offsets & tower angles and then runs into a problem.  Maybe these new values make probing that point 

I tried compiling using arduino ide (normally use sloeber) to see if that changed anything, it did not.
Rolled back to bugfix1.1.9--this all works fine.

I do get a few errors in g33.cpp from sloeber, though they do no prevent compilation.  Mostly stuff like this:
 line 228:    const xy_pos_t center{0};
                center is underlined as the error with the note: "Invalid arguments 'Candidates are: XYval()             XYval(const XYval<float> &)"

sloeber is kind of notorious for spitting out erroneous errors, but I thought I would bring it to your attention.  I get 8 total errors, almost all some form of "invalid arguments"  If it would help to detail them all, let me know and I will do so.

sjasonsmith, 2019-12-30:
You might try adjusting `Z_PROBE_LOW_POINT` to allow it to probe further before erroring out. I think you said you tune the height before anything else, so I would think 3mm would be enough, but maybe there is an error somewhere related to that. On my delta I think  had to adjust it to -5mm, but I didn't separately tune the height first, I just ran `G33` without arguments.

> `const xy_pos_t center{0};`

That probably should be:
`const xy_pos_t center = {0};`

sjasonsmith, 2019-12-30:
That line you mentioned actually doesn't generate any warnings when I build in platformIO. I just tried building it for both AVR and LPC1768 with no warnings in any of the delta-related code.

mtraven, 2019-12-31:
> 
> 
> That line you mentioned actually doesn't generate any warnings when I build in platformIO. I just tried building it for both AVR and LPC1768 with no warnings in any of the delta-related code.

yah, like I said, sloeber loves to spit out errors, a lot of the time they are nothing, just thought you might like to know.

wrt the z_probe_low_point -- When I was desperately trying to get g33 to run I did play with that, its normal (prev working) value is 2.  And I really shouldn't even have been at 3 as my probe offset is -2.65mm.  Then I thought about it a bit & that value isn't relevant as it is the distance it will go PAST the sensor being tripped.  On my probe failures, it never got far enough to trip the probe, so both theoretically and in practice, that variable isn't going to change my probing problem.

Just to make sure this wasn't an issue on my end,  I spent some time re tramming and leveling the bed with bugfix1.1.9.  It is not better than ever before and I plan to give 2.0 another go in......well 6 hours from now when my current print finishes.  I will report then.

Let me know if I can be of any help in correcting this bug.

p.s.  whats the deal with platformio ?  Is it just another IDE?  is it worth getting to know?  I had a quick look at it, didn't love the interface..but, ya know, 1st impressions....

boelle, 2020-01-07:
@mtraven still an issue?

and yes vscode+platformio is better than say arduino, i looked at it and never used arduino ide since

boelle, 2020-01-14:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-14 by boelle