MarlinFirmware/Marlin #16385: [BUG] DGUS_LCD causing compile errors

painteddigital, 2019-12-30:
### Bug Description

Enabling DGUS_LCD causes compile errors in both Arduino IDE and PlatformIO

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Uncomment #define DGUS_LCD in Configuration.h
2. Compile fails

**Expected behavior:** Compile success

**Actual behavior:** 

[Arduino error log](https://pastebin.com/vvZxg41k)
[PlatformIO error log](https://pastebin.com/Wd2jvcyd)

#### Additional Information

I've got a fresh install of Marlin 2.0 with minimal changes to configuration.h, configuration_adv.h, and a pins file. The build will compile once I comment out the LCD line. I've tried enabling SDSUPPORT with no effect.

[Configuration.h](https://github.com/painteddigital/Marlin/blob/2.0.x/Marlin/Configuration.h)
[Configuration_adv.h](https://github.com/painteddigital/Marlin/blob/2.0.x/Marlin/Configuration.h)

ellensp, 2019-12-30:
Take a look at https://github.com/MarlinFirmware/Marlin/pull/16317   Issues are fixed but not in main code yet

painteddigital, 2019-12-30:
@ellensp Thanks! That does seem to compile. Can't test on my machine at the moment but looks promising.

boelle, 2020-01-07:
@painteddigital tested it?

boelle, 2020-01-14:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-14 by boelle