MarlinFirmware/Marlin #16386: [BUG] M112 sent from LCD custom menus ignored if EMERGENCY_PARSER is enabled

interstellarmisfit, 2019-12-30:
### Bug Description

I'm using the latest bugfix-1.1.x on a Creality Ender3 board. I added an EMERGENCY STOP item in the custom user menus to send M112. 

When EMERGENCY_PARSER is disabled it works as expected. The printer immediately stops and goes to the printer killed state. 

With EMERGENCY_PARSER enabled, it does not work. I receive the message "Recv: echo:enqueueing "M112"" over the serial port. However the printer continues on as normal.

Sending M112 from the serial port works correctly in both configurations. 

Other custom menu items work correctly in both configurations.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4009020/Configuration.zip)

### Steps to Reproduce

1. Add a custom menu item EMERGENCY STOP with GCODE M112
2. Select EMERGENCY STOP from custom menu.

**Expected behavior:** Printer stops and kills. 

**Actual behavior:** Printer ignores command and continues operating normally.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

AnHardt, 2019-12-30:
The emergency command parser is located in the serial RX interrupt.
Internally enqueued commands never go the way thru the RX interrupt.

Workaround is: if you are near the machine, to press the kill- or reset-button or pulling the plug. Finding your way true the menus, in a emergency case, is to slow anyway.

A fix would not disable the interpretation of the emergency commands in the normal parser when the emergency command parser is enabled. That might need protection from executing commands twice for some of the emergency command parser interpreted commands (not so for M112 - kill is kill).

boelle, 2020-01-07:
@interstellarmisfit still a problem?

interstellarmisfit, 2020-01-07:
The explanation by AnHardt regarding the emergency parser makes sense on why it's working like it does. I plan to install a reset button on my machine instead. 

Having  M112 parsed in both queues might be a good feature, but not necessary for me at this time.

You can close this. Thanks.

closed 2020-01-07 by boelle