MarlinFirmware/Marlin #16387: Octoprint can't find interface (SKR mini)

Udo-Nbg, 2019-12-30:
SKR mini V 1.1 connection problem:
Marlin 2.0.1 (2019-12-28)
I have the problem that Octoprint (octopi-buster-lite-0.17.0 @ Pi4) cannot find the board / interface.
The connection works perfectly under Windows 10.
Apparently a firmware problem / USB driver / USB driver signature that occurs with Octoprint.
The interface is only recognized very rarely, mostly not.

Tested on two different SKR mini V1.1:
one mounted on Printer, 24V supply powered, with 12824 Display, tmc2208 standalone
other direcht connected / USB-powered / no Display, no Motordrivers attached, nothing else connected (board only)

boelle, 2019-12-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Udo-Nbg, 2019-12-30:
Sorry, 
done.

Lord-Quake, 2019-12-31:
Try setting the following in configuration.h:

`#define SERIAL_PORT 2`
and
`#define SERIAL_PORT_2 -1`

Udo-Nbg, 2020-01-01:
1st happy new Year ;-)

But sorry, that did not solve my prolem.

Issues are exactly the same.
I can connect an control with Windows (10)
But there ist no device found on the Pi4

`Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: []
Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
Failed to autodetect serial port, please set it manually.`

I rebuild the soure completly new, and attched the configuration.h again...

[Marlin_2-0-1.zip](https://github.com/MarlinFirmware/Marlin/files/4014082/Marlin_2-0-1.zip)

boelle, 2020-01-08:
check what serial ports or usb ports that are on the pi when the printer is connected and when not connected

make sure the pi has any needed drivers, this can be (and most likely will be) different than on windoze

google if other have the same problem on the pi

boelle, 2020-01-15:
@Udo-Nbg still the same issue?
have you tried with bugfix 2.0.x ? 2.0.1 is just a frozen copy of 2.0.x

Udo-Nbg, 2020-01-16:
Yes still the same issue,
Windows easily recognizes and uses the driver
Linux (octoprint) doesn't even find an interface.

and I wasn't familiar with the frozen verion.
I will try the current bugfix 2.0.x shortly.

Thank you

boelle, 2020-01-16:
does linux have all needed drivers installed?

Udo-Nbg, 2020-01-17:
which driver i have to install?
chipset oder drivername please

sam2b, 2020-01-22:
@Udo-Nbg what com port are you specifying in OctoPrint? For example, mine is /dev/ttyACM0
(ttyACM zero)

To confirm, have you compared the /dev directory listing without the printer usb cable connected to the Raspberry Pi vs. when not connected? If not, what is your comparison result, please?

Udo-Nbg, 2020-01-22:
@sam2b i can NOT select any port there is nothing  
because NOTHING will recognized :-(

sam2b, 2020-01-22:
@Udo-Nbg I'm asking for you to compare the /dev directory listing in Linux without the printer usb cable connected to the Raspberry Pi vs. when connected. What is your comparison result, please?

Udo-Nbg, 2020-01-22:
I replied: because NOTHING is recognized, I mean that nothing is recognized in the / dev folder either change.

I tested with an ftdi ft232 usb-serial adapter, which is recognized immediately and also appears and disappears in the /dev folder when I plug in or unplug it, 
so please don't consider myself stupid

boelle, 2020-01-23:
so maybe bad hardware since nothing happens?

Udo-Nbg, 2020-01-23:
That would be logical and also my guess
but how does that explain the correct function under windows 10?

Nevertheless, I ordered two new SKR mini V1.1, one as a replacement and the second as a reserve.
But I observe exactly the same behavior on all three boards.

Now you could also consider that the cable could be the cause, I also thought of that and the original cable is actually defective. But I also have this problem with three or four other high quality cables.

In the meantime, I've ordered the Lerdge X board that has closed the firmware, but I've had good experiences with Lerdge. By the way, my Lerdge K-Bard works great with Octoprint *, so it can't be because of the Raspberry Pi 4 *.

As soon as I have received the Lerdge X I will report.

boelle, 2020-01-23:
> but how does that explain the correct function under windows 10?

because win10 prob does at lot in the background automatic without telling, and linux dont do that

just a guess

Udo-Nbg, 2020-01-24:
:-( that helps me a lot (NOT).

boelle, 2020-01-24:
i'm just a user like you so dont expect an expert answer

also remember that everyone working on marlin is working for free and you dont pay for it so you cant expect support like on a real product

Udo-Nbg, 2020-01-25:
Raspberry Pi 4, 4GB supplied with original Raspberry 3A Supply, 
SKR mini V1.1 at USB 2.0 Port, lower left port (black):

`pi@octopi:~ $ ls /dev/tty*
/dev/tty    /dev/tty19  /dev/tty3   /dev/tty40  /dev/tty51  /dev/tty62
/dev/tty0   /dev/tty2   /dev/tty30  /dev/tty41  /dev/tty52  /dev/tty63
/dev/tty1   /dev/tty20  /dev/tty31  /dev/tty42  /dev/tty53  /dev/tty7
/dev/tty10  /dev/tty21  /dev/tty32  /dev/tty43  /dev/tty54  /dev/tty8
/dev/tty11  /dev/tty22  /dev/tty33  /dev/tty44  /dev/tty55  /dev/tty9
/dev/tty12  /dev/tty23  /dev/tty34  /dev/tty45  /dev/tty56  /dev/ttyAMA0
/dev/tty13  /dev/tty24  /dev/tty35  /dev/tty46  /dev/tty57  /dev/ttyprintk
/dev/tty14  /dev/tty25  /dev/tty36  /dev/tty47  /dev/tty58  /dev/ttyS0
/dev/tty15  /dev/tty26  /dev/tty37  /dev/tty48  /dev/tty59
/dev/tty16  /dev/tty27  /dev/tty38  /dev/tty49  /dev/tty6
/dev/tty17  /dev/tty28  /dev/tty39  /dev/tty5   /dev/tty60
/dev/tty18  /dev/tty29  /dev/tty4   /dev/tty50  /dev/tty61`

Udo-Nbg, 2020-01-25:
`OctoPrint version : 1.3.12
OctoPi version    : 0.17.0
------------------------------------------------------------------------------

pi@octopi:~ $ lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@octopi:~ $
pi@octopi:~ $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-FTDI_USB_Serial_Converter_FTFLYVHJ-if00-port0`

i connected a seperate usb serial interface just for test and it appears

boelle, 2020-01-25:
so either the usb on the board is dead or there are no drivers

Udo-Nbg, 2020-01-26:
I had a session with a Linux expert yesterday, about 2 hours.

Obviously, the problem is that there are no drivers from ST for Linux, specially for this kind of Processorchip avaiable.

So it is NOT a problem on Marlin.

Thank you for your patience.

closed 2020-01-26 by Udo-Nbg

boelle, 2020-01-26:
No problemo, but a shame that there is no driver