MarlinFirmware/Marlin #16388: [Question] Marlin github actions

GMagician, 2019-12-30:
Maybe a little OT so please forgive me.

In my Marlin forked repository I always get errors with "bump date" action.
Problem is related to account credentials. I read that to "log in" correctly some "GITHUB_TOKEN" has to be used in action "code" but I can't see where it is used (it seems that action launch ".\buildroot\bin\bump_date" but no token references there)

What's wrong with my repository?

Vertabreak, 2019-12-30:
I just deleted this action no more errors. I suspect it only works on upstream.

sjasonsmith, 2019-12-30:
If you go to the settings of your fork you can disable all actions and it will stop.

GMagician, 2019-12-30:
@sjasonsmith of course it will, but I'll also lose the test build feature...or not?
That action is usefull before posting a PR to check if everything is ok (This is what....sometimes... I was used to do with travis)

sjasonsmith, 2019-12-30:
> @sjasonsmith of course it will, but I'll also lose the test build feature...or not?
> That action is usefull before posting a PR to check if everything is ok (This is what....sometimes... I was used to do with travis)

I didn’t see that action available in my fork? The only action was to bump the distribution date. I don’t know how to run them on my own branch without submitting a PR. I use something called wwtd (What would Travis do) to run them locally.

GMagician, 2019-12-30:
".\workflows\test-builds.yml" isn't this one?

Vertabreak, 2019-12-30:
the test build feature works fine i just disabled the date bump action by deleting it.

GMagician, 2019-12-30:
@Vertabreak how do you live with the fact that when you post a PR, your branch "without" the action will try to remove it as well from Marlin repository?

Vertabreak, 2019-12-30:
my PR branch is seperate from the other branch

sjasonsmith, 2019-12-30:
So do you create a pull request to your own bugfix-2.0.x branch to trigger the test build, or is there some other way to trigger it on a different branch?

@Vertabreak, when you say you delete it, you mean you deleted the yml file? I’d prefer not to do that since my own bf2 branch would then diverge from upstream.

thisiskeithb, 2019-12-30:
The error:
```
Run source ./buildroot/bin/bump_date

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.
```

I already confirmed that those are set and defaults match my info. I tried to add a secret, etc. through Github's UI, but didn't get anywhere.

> the test build feature works fine i just disabled the date bump action by deleting it.

I'm usually pulling down fresh copies of Marlin for new builds or rebasing old copies all the time, so having to delete the action each time I catch my repo up, wouldn't be ideal.

Vertabreak, 2019-12-30:
my default branch is my fork then there are other unmodified branches that i build PR branches from. seems to work fine this way my active PR was not effected at all.

sjasonsmith, 2019-12-30:
There must be some way to restrict this action to only run upstream. A cron job that pushes changes into every fork in GitHub wouldn’t be ideal.

I noticed that GitHub documents actions as being disabled in forks by default. I wonder why it is enabled by default for all of us?

thisiskeithb, 2019-12-30:
> There must be some way to restrict this action to only run upstream. A cron job that pushes changes into every fork in GitHub wouldn’t be ideal.

At least it doesn't work by default 🙂 Think of all the zombie forks...

GMagician, 2019-12-30:
@thisiskeithb did you noticed this, when you tryed with your own define secret?

> With the exception of GITHUB_TOKEN, secrets are not passed to the runner when a workflow is triggered from a forked repository.

> To provide an action with a secret as an input or environment variable, you can use the secrets context to access secrets you've created in your repository. For more information, see "Contexts and expression syntax for GitHub Actions" and "Workflow syntax for GitHub Actions."

I'm not very used to actions and I'm reading something here and there...

sjasonsmith, 2019-12-30:
Even if you could get the job to authenticate, you really don’t want a job to be committing to your bugfix-2.0.x branch in your fork.

GMagician, 2019-12-30:
thanks to all...

closed 2019-12-30 by GMagician