MarlinFirmware/Marlin #16392: [BUG] Error occurs when stm32f4 chip uses mirrored mode

FLYmaker, 2019-12-30:
AUTO_PARK       OK
DUPLICATION   OK
T1 printing alone is OK
Using mirror mode T1 will vibrate and not move left or right.T0  OK
Lpc1769 uses the same firmware, and the image mode is normal
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4009986/Marlin.zip)

InsanityAutomation, 2019-12-30:
I have a build half done with an STM32F4 (Skr 1.1 pro) so ill be testing this with idex eventually. It does sound odd as none of that code should be platform dependent.

boelle, 2020-01-24:
@FLYmaker how often do you update? bugfix is updated almost daily

i wonder how many have the same hardware and can confirm this one

LorenzoSaini, 2020-01-27:
I have a 3d printer with BTT SKR pro 1.1 (also with the Rumba do the same thing) and TMC5160, in duplication mode and auto parking works perfectly, but when i try to use mirror mode the righ extruder only move to the left eventually crashing into the left extruder. (i try different driver with no luck)
Marlin 2.0.1 default branch.
Im sure that with the old bugfix 2.0.x almost 1 year ago both the duplication mode an the mirror worked just fine on a rumba 8-bit board, but i cant find my old firmware.
[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4116281/marlin.zip)

FLYmaker, 2020-01-27:
> @FLYmaker您多久更新一次？错误修正几乎每天更新
> 
> 我想知道有多少具有相同的硬件，并且可以确认这一

This problem also occurs in the latest version

FLYmaker, 2020-01-27:
> 我有一台具有BTT SKR pro 1.1的3d打印机（也与Rumba进行了相同的操作）和TMC5160，在复制模式下，自动泊车效果很好，但是当我尝试使用镜像模式时，righ挤出机只会向左移动，最终崩溃进入左挤出机。（我没有运气尝试其他驱动程序）
> Marlin 2.0.1默认分支。
> 我肯定在大约1年前使用旧的bugfix 2.0.x时，两种复制模式和镜像在rumba 8位板上都能正常工作，但是我找不到我的旧固件。
> [marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4116281/marlin.zip)

The 8-bit motherboard image mode is normal

boelle, 2020-02-02:
> This problem also occurs in the latest version

what version is that?

thinkyhead, 2020-02-02:
> both the duplication mode an the mirror worked just fine on a rumba 8-bit board

And still does. This issue pertains to it not working on STM32F4 but apparently it is demonstrably working on LPC1768.

thinkyhead, 2020-02-03:
Are you able to physically swap the X and X2 stepper drivers to see if the behavior switches and starts to appear on T0 instead of T1?

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle