MarlinFirmware/Marlin #16393: [BUG] Z Endstop not triggering at MKS BASE/GEN L

ikout, 2019-12-30:
### Bug Description
After upgrading from 1.1.9.x to 2.0.0 & 2.0.1 & 2.0.x bugfix releases the Z-Endstop at the MKS Base 1.4 & MKS Gen L 1.0 boards is not working anymore, if downgrade at 1.1.9 release the issue is fixed. So something has been missed with the Z-Endstop.

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4010723/config.zip)

### Steps to Reproduce

Just upgrade and test the Z-Endstop with M119, while for X & Y the endstop gets triggered, for Z-Endstop it never gets triggered.

I'm going to try one more board different type (Geeetech GT2560) and test the Z-endstop there as well, it maybe be an issue on the AVR HAL or endstop handling for 8-bit boards.

I have an SKR 1.3 with 2.0.x bug (pre-2.0 release) and there I have no issue with the Z-Endstop...

ellensp, 2019-12-31:
Can you enable M43  http://marlinfw.org/docs/gcode/M043.html  see what it has to say about Z endstop pins?

ikout, 2019-12-31:
I tried the M43 E1 after enabling the "#define PINS_DEBUGGING" at the Configuration_adv.h

As seen below I send the command and the monitor begins.
>>> M43 E1
SENDING:M43 E1
endstop monitor enabled
  X_MIN:1  Y_MIN:1  Z_MIN:1

Next I send a M119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open

Next I attach a mechanical endstop at the ZMIN port and I press it and nothing happens....

I move the same switch at the YMIN port and press it and the endpoint monitor reports that there was a change:
  Y_MIN:0
  Y_MIN:1

So it seems like the ZMIN port is "ignored" not detected at all...
I can downgrade this board to 1.1.9 if anyone thinks that this will give more details.

ellensp, 2019-12-31:
You have #define SERIAL_PORT_2 1 in your Configuration.h
Serial1 on the mega is pins 19 (RX) and 18 (TX). But Z_Min uses pin 18 
This is in conflict!

ikout, 2019-12-31:
Thanks! I missed that, disabling the SERIAL_PORT_2 and after that **the Z_Min works again**.
This was transferred from my SKR 1.3 board :(

closed 2019-12-31 by ikout