MarlinFirmware/Marlin #16395: [BUG] (short description) Z raised again and again.

PNechNET, 2019-12-31:
Z raised again and again before individual homing X or Y.

Firmware: Marlin bugfix 2.0.x

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4012420/Configuration.zip)

sjasonsmith, 2019-12-31:
Can you add more information to this? We need to know exactly what the unexpected behavior is, along with steps to reproduce it.

tatusah, 2019-12-31:
I guess it's about raise z before homing which happens every time if you home axis individually. So basically every time you home x or y axis the z position climbs up (reproduce by turning on the printer and just home x or y axis multiple times). So maybe there could be done some kind of check if the z axis has already been raised once. However, if you just home all axis at the same time then this isn't an issue. And probably you don't have to home individual axis multiple times anyway. I think only situations where this can cause problem is if the z height is close to the movement limit to start with.

PNechNET, 2019-12-31:
I installed SKR 1.3 on my printer 4 months ago and installed Marlin-bugfix-2.0.x. I worked well. Now I checked marlin.org and saw a new version. I download it and cofigure, printer work well too, but I find bug with individual homing.

sjasonsmith, 2019-12-31:
>  I find bug with individual homing.

Can you provide specific steps to reproduce the problem?

I suspect @tatusah is probably correct, and I probably introduced that behavior recently. We need you to provide more specific details of the problem to make sure it can be fixed correctly.

tatusah, 2019-12-31:
I can confirm this happening with `Z_HOMING_HEIGHT` defined. I've it set to 5 currently and every individual x or y axis homing incrementally lifts the carriage 5 mm on Ender 3 with SKR e3 mini v1.2. And @PNechNET has the homing height set 10 so he must see the height change by 10 every homing cycle.

sjasonsmith, 2019-12-31:
@tatusah, I understand the issue as you describe it. I’m just trying to get the original poster to provide appropriate details. If I or anybody else is going to spend time fixing it, it seems reasonable to expect the reporter to provide a good description of the problem.

PNechNET, 2020-01-01:
Thanks to @tatusah, this problem occurs if the Z axis is unexpected, if you first determine Z, the problem disappears. My standard "Z_HOMING_HEIGHT" is 10, I set 5, the result is a decrease in lift height.

@sjasonsmith My steps:
1) I downloaded the firmware from the site
2) Manually transferred the settings from the old firmware to the new one (without copying files)
3) Installed the firmware on my printer with skr 1.3.
3) Turned on my printer and started checking all axes separately.
4) Checked the correct direction of movement, began to check the limit switches.
At this moment I found this error. I downloaded the firmware 4 days ago. The configuration files are in the description of the first comment.

sjasonsmith, 2020-01-01:
Thanks. This is something I introduced recently. I was fixing an issue that could cause the nozzle to scrape across the bed if homing after motors have been disabled. If the Z position is not known, Z will be raised.

To fix this, the code will need to remember a relative positions since the motors were enabled. I’m not sure whether that already exists, or if more state variables will be needed.

closed 2020-01-09 by PNechNET