MarlinFirmware/Marlin #16397: auto bed leveling issue

gokulvenkat, 2019-12-31:
hi i am using marlin as firmware for my core x y (hbot) with no issue on frame such as unparallel frame (300*300*300 printer)
my issue is only one side of the  mesh is taken wrongly near to orgin during auto homing i tried 4-8 matrix but still starting edge of the matrix is wrong i cant print the first layer only on that edge,i am using proximity sensor for probing

here is my configuration files

### My Configurations

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4012673/Configuration_adv.zip)

stratodream, 2019-12-31:
Same issue from my side.
Ender 3
Skr e3 mini 1.2
Marlin bugfix 2.0.x last commit.

Z movements are not compensated.
Manuale bed levelling Is necessary to have a good print.

tatusah, 2020-01-02:
The bug description is vague and conflicting. The configs you have uploaded @gokulvenkat are for Marlin 1.1.9 with 3 point auto leveling but you mention 4-8 matrix.

Have you tried with either latest bugfix-2.0.x or Marlin 2.0.1 release? What leveling method were you actually trying to use?

stratodream, 2020-01-02:
I have last committed bugfix- 2.0.x, skr e3 mini, originale bl touch.
The z positions are not compensated.
Anyway in issue #13512 there are a lot of people reporting the same issue.
Could i provide more Infos?

tatusah, 2020-01-02:
I use also SKR e3 mini v1.2 with Ender 3 + org. BLTouch. UBL leveling with 9x9 grid has been working fine using 512K memory mod after the EEPROM related issues were corrected a while ago. Currently using firmware updated last time today with all the recent bugfix commits. Ender 3 isn't corexy printer though. However, if the actual probing works for corexy printers in the right positions then it's quite likely that also leveling would work.

For others to actually confirm that there is a bug one should upload correct configurations that were used when the problem occurred and detailed explanation how the problem can be reproduced.

If not using an active leveleving method that's done after homing and before printing then it should be made sure that the leveling is actually activated by having it activated in print start gcode and/or enabling `RESTORE_LEVELING_AFTER_G28`. Also if using UBL care needs to be taken that the mesh is actually created correctly. Depending on the probe offset, probe edge setting and mesh inset there is possibility that after probing is multiple rows/lines of missing probe points. For that there might be need to run smart fill G29 P3 several times and it makes sense to use G29 T to check that actually all mesh points are filled. By following the instructions and documentation online it's not that hard to set the things working but when taking into consideration the hardware, firmware and the print file there is clearly many points where one can also go wrong. If the problem is actually caused by wrong configuration then help is best found from appropriate Facebook forums or Marlin Discord.

reloxx13, 2020-01-03:
can confirm, probing starts at z homing position + edge instead of 0,0+edge

tpruvot, 2020-01-03:
+1, UBL is not compensating at all, and ABL (bilinear) not like it should... or like it was in september

reloxx13, 2020-01-03:
i reverted this and the x,y tart position is fine again now.
https://github.com/MarlinFirmware/Marlin/commit/1c9ccce5209cd1727bf80e632f4f781c651e0c35#r36649413

stratodream, 2020-01-03:
> 1c9ccce#r36649413

Could you please stare tu sw you're using so that i can try it ?

stratodream, 2020-01-03:
I think that in #13512 informations were shared enough.
I think that this issue should be taken into account as It Is no raised
from few people, i see that many of us have the same issue.

tatusah, 2020-01-03:
Was about to mention the same that there is already known thing about the probe edge margins doubling at least with ABL leveling which narrows down the probed area https://github.com/MarlinFirmware/Marlin/pull/16367 but this shouldn't prevent the leveling from working completely. It would make the leveling less accurate in the areas which wasn't reached. Temporary fix for that is using half of the probe edge value you are normally using or copying the changes from that pull request as it hasn't been merged to the bugfix branch at the time of writing this.

As the probe offset typically isn't symmetrical between x and y axis I have preferred using individually set probe edge values for each side of the bed instead of the global value. Been using this type of configuration with UBL. The doubling of the probe edge values hasn't affected this configuration.

Did you @stratodream have also the problem that the margin you have specified for probing (min_probe_edge) wasn't correct or were you not seeing the compensation working at all? I remember your posts in the other issue thread about problems which was before the code that caused probe edge was merged to the Marlin. So the problems you were having then are probably different from this one.

What I was after with the previous message was to understand which of the different leveling methods was the problem with the original issue author and what the problem was. And if others don't have the exact same issue then it makes sense to separate those issues into other threads to better handle them.

stratodream, 2020-01-03:
I'm complaining about the compensation Is not working as It did in Marlin
1.1.9.
I have the same configurations, switching to the new Marlin 2.0 I cared
about taking inside the old configurations.
After automatico bed leveling i see that the first layer is very bad and It
is ok only after the manual bed leveling.

tatusah, 2020-01-03:
Good, it's probably something else than this edge thing as you mentioned your problems 20 days ago and this doubling was caused by a commit 10 days ago. I can check your configurations (configuration.h and the adv) if you upload them @stratodream

stratodream, 2020-01-03:
Give me few data, i will upload them.

gokulvenkat, 2020-01-03:
> The bug description is vague and conflicting. The configs you have uploaded @gokulvenkat are for Marlin 1.1.9 with 3 point auto leveling but you mention 4-8 matrix.
> 
> Have you tried with either latest bugfix-2.0.x or Marlin 2.0.1 release? What leveling method were you actually trying to use?

I TRIED other methods abl linear and bilinear and ubl with 4 to 8 matrix finally three point bed leveling can only manage to print first layer so i kept it as 3 point bed leveling

Aurawin, 2020-01-03:
Guys... The kinematics have indeed recently changed.  The sensor is now the relativity of the homing, G34, and G29 commands.

I'm going to give this thought, but it appears that the programmer shifted logic and is making all probes from point of view of the probe and no longer the nozzle.

stratodream, 2020-01-03:
Sorry, i do not understand, could you please vive us more clarity?

tatusah, 2020-01-03:
Do you @gokulvenkat have something actionable to try reproducing your problems? I would at least do the probe repeatability test with M48 for which you should enable that feature in configuration.h. For BLTouch v3.1 I get reported around 0.0015-0.002 standard deviation. If the reported deviation is high for you this would easily explain why the leveling compensation results inaccurate prints. With inductive proximity sensor the possible bed magnets and probing near bed attachment screws vs other areas can affect the readings.

It makes sense to get the bed level to start with even with auto bed leveling. But just out of curiosity I tried putting thick business cards on one side of the bed under the glass plate so that there was 0.6 mm difference between right and left edge of the bed. I created new UBL mesh for this situation and one layer test print was successfully compensated around the bed. Then did another test with a mesh for normal situation and used UBL mesh tilt correction probing at the start of the print which also resulted in successful print.

ManuelMcLure, 2020-01-03:
> Guys... The kinematics have indeed recently changed. The sensor is now the relativity of the homing, G34, and G29 commands.

This has always been the case, for at least a couple of years (since I started using UBL). If you probe at, for example, (10,10) the _probe_ will move to (10,10) and measure the bed at that position, and that measurement will be used when printing whenever the _nozzle_ is at (10,10). That is the whole point of ``NOZZLE_TO_PROBE_OFFSET``. The _probe_ needs to be able to move to any point that is specified as a probing location.

gokulvenkat, 2020-01-03:
> Do you @gokulvenkat have something actionable to try reproducing your problems? I would at least do the probe repeatability test with M48 for which you should enable that feature in configuration.h. [...]

i am using inductive probe and does it get disturbed by neodymium magnets but even though  its out of the probing way

tatusah, 2020-01-03:
So you have individual neodymium magnets under the bed @gokulvenkat ? Different metal types causes the inductive sensor trigger from different distances. For the inductive probe to work accurately it expects uniform composition of the bed then it could read if the bed is actually in different distance in probing locations. Now the bed doesn't seem uniform and also the magnetic fields from the magnets would probably cause interference. Don't know what kind of error correction and filtering methods the probe possibly has.

Aurawin, 2020-01-04:
@gokulvenkat @tatusah ,

I just had a cold-rolled steel plate cut for me.  I was so tired of BLT errors and inconsistencies I threw it all out and re-designed my probe system.  Rumor had it that conventional ferromagnetic sticky mats are unusable due to loss of magnetism over 50C... So naturally - I had to test :-)  
I ordered a magnetic mat (roll) for use on car magnet signage. I cut the square 330mmx330mm and used 3M high temp spray adhesive.
I coated the steel plate with high-temperature furnace paint.  I regret it somewhat because the first layers remove the paint one print job at a time.  Sooner than later I will remove all paint with a good sanding/polishing.

I use a tiny 2mm inductive sensor that is boosted by the ferrous in the plate and the magnet sheet.  The plate is ~1.5mm thick.

I tested the plate at 90(c) and it worked.  It worked so well I had to wait for the magnetic mat to cool down.  The magnet would not release the plate until it reached ~50c.  Good lesson: don't let impatience get the better of you.  I ran a scrapper through my first magnetic mat. 

The 2mm inductive sensor is by far the most reliable measuring device I have ever experienced.  I used BLTs, Capacitive sensors and I was exploring a dual optical sensor.  

I observe a bit of variation but I think it's acceptable and caused by the deflection/distortion of my bed and not from the repeatability as that came out better than any BLT and Capacitive. Nonetheless, I have a pretty large ABL grid.

I absolutely love the idea of (printing/flexing) the metal plate to remove a print.

gokulvenkat, 2020-01-06:
Yes i am using magnet under the bed but probing points are far from magnet

tatusah, 2020-01-06:
The pull request fixing the double min_probe_edge issue has been merged to the bugfix branch so update your firmware to get that fix in.

But if you are seeing uneven height compensation you might want to continue investigating the issue. Just to rule out different possibilities you might want to do the M48 reliability test. Best way to test the possible effects of the magnets would be to do the test in some position that's away from the magnets. Then if you have some spare magnets place on under the bed and keep the actual carriage position in the same place so that the only thing changing is the magnet you have placed under the probe position or near it. Check if the reported min/max height and deviation change.

gokulvenkat, 2020-01-08:
Which link is the best to get updated bugfixed firmware

tatusah, 2020-01-08:
I guess it's best to learn how to use Vscode and compile the firmware to your specific setup. Then you can always pull in the latest commits from Marlin GitHub repository and get exactly what you want. Additionally you can make your own fork of Marlin in GitHub to manage things.

gokulvenkat, 2020-01-09:
I will try autobed leveling once i got the new firmware bug fix and will post here what happened

gokulvenkat, 2020-01-09:
@tatusah i am new to the programing stuff and can you please explain or make  me clear about vecode and repository means,sorry to ask this but i am new to marlin program editing

boelle, 2020-01-10:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

boelle, 2020-01-10:
but is this just user error and asking for help, or is there really a bug?

stevenvo, 2020-01-26:
I'm on bugfix 2.0 branch and don't see the Z moving even though with `M420 S1` in gcode before printing. So looks like it is not fixed yet.

stratodream, 2020-01-26:
Hi,
here attached my configs files for the same issue:
Ender 3 Pro
Original BlTouch 3.1
Skr E3 mini V1.2
Marlin BugFix 2.0.x (commit available on 
[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/4113477/ConfigFiles.zip)
18/01/2020)

stevenvo, 2020-01-28:
is there a way to simulate the z-compensation in unit test / integration test suite? 

https://github.com/MarlinFirmware/Marlin/blob/e6a37ba2a05131873b4b1d1f2b6b2697c388f944/Marlin/src/core/utility.cpp#L161

boelle, 2020-02-02:
@gokulvenkat please re-test with latest bugfix from here: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

let us know if the problem is still there

stratodream, 2020-02-03:
I tested It, to me It serena ti be fixed.

gokulvenkat, 2020-02-03:
Hi everyone,it's been a long time I tested with bugfix marlin 1.1.9 and the issue is solved now 
Thank you @boelle  @tatusah and every one

gokulvenkat, 2020-02-03:
Sorry for the delay

boelle, 2020-02-03:
1.1.9 or bugfix 2.0.x ?

gokulvenkat, 2020-02-03:
1.1.9 bugfix

stratodream, 2020-02-03:
but the issue was on 2.0.x bugfix if i'm correct, anyway as i told before,
i tested It on 2.0.x bugfix and It seems ti be fixed.

boelle, 2020-02-03:
@gokulvenkat try bugfix 2.0.x also,

there is no 1.1.9 bugfix only 1.1.x bugfix

why i push for bugfix 2.0.x is that its the only one that is actively worked on daily
1.1.9 will not be updated, and the maintainers do not do anything with 1.1.x unless users submits PR's

gokulvenkat, 2020-02-03:
Sorry i used 1.1.x  bugfix

gokulvenkat, 2020-02-03:
Not 1.1.9

boelle, 2020-02-03:
but you should really try 2.0.x bugfix, its the way forward

gokulvenkat, 2020-02-03:
On next upgrade of the machine i move to 2.0.x

gokulvenkat, 2020-02-03:
> but you should really try 2.0.x bugfix, its the way forward

Thank for your tip

chderycker, 2020-02-04:
Still having this problem on ender 3 with Skr e3 mini 1.2 on a config made yesterday on Marlin Bugfix 2.0.3

closed 2020-02-11 by thinkyhead