MarlinFirmware/Marlin #16399: [BUG] Change filament and runout is broken when filament sensor is enabled on skr 1.3

Liger0, 2019-12-31:
### Bug Description

The change filament option, even when the filament runs out, is broken on E0 (unsure about E1).

### My Configurations

Skr 1.3 Tmc 2209 on a Cr10s with dual extruder and dual filament sensor
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4013305/Marlin.zip)

### Steps to Reproduce

1. Change filament
1.a. End filament while printing
2. Purge automatically
3. There is no option to stop purging
4. You either indefinitely purge more or disable the filament sensor

**Expected behavior:** 
IT should show an option to stop purging and continue

**Actual behavior:** [What actually happens]
You can't resume printing with the filament sensor enabled

#### Additional Information
The filament sensor works and detects correctly, in fact if the filament is in, the light in it is on, and also if I just print it starts.

ss89, 2020-01-01:
I have a similar issue on my A10m (also dual extruder but single hotend and dual filament sensor), stock machine but added bondtech extruder and though also updated the firmware to 2.0.1:
I did not print with the machine with the new firmware because i wanted to finish configuring and testing first.
When trying the filament change function and selecting e1/e2 (1 based as in menu), it does a purge but does not eject the filament but wants to start inserting the new filament right away. This does happen with and without having the filament sensor enabled.
The A10m btw. has a GT2560 board, so nothing HAL/architecture related i assume on my side.
Firmware copy: https://git.x2d2.de/3D-Printing/Marlin-Firmware/-/archive/A10m-2.0.1/Marlin-Firmware-A10m-2.0.1.zip

boelle, 2020-01-08:
how do you check if it detects correctly?

Liger0, 2020-01-08:
It seems working with bugfix.

closed 2020-01-08 by Liger0