MarlinFirmware/Marlin #16403: [BUG] Z Max Jerk cannot be set via M205

siteswapjuggler, 2019-12-31:
### Bug Description

Unable to set Z jerk within the firmware limitation via M205 command.

### My Configurations

A custom CR-10S Pro configuration 
[myConfiguration.zip](https://github.com/MarlinFirmware/Marlin/files/4013466/myConfiguration.zip)

### Steps to Reproduce

1. connect via serial command
2. send M205 Z0.4

**Expected behavior:** 

Z max jerk behing set

**Actual behavior:**

Z Max Jerk is not set and I receive "Z Max Jerk limited to 1"

#### Additional Information

This is whith classic Jerk Mode enable instead of Junction Deviation. A hack was applied by changing the value in the firmware and making a M502 te reset default parameters in the eeprom.

GMagician, 2020-01-01:
That's a little weird... warning message is sent only when you have `LIMITED_JERK_EDITING` enabled, but you don't have it. Again limit is twice DEFAULT_ZJERK then limit 1 means DEFAULT_ZJERK should be 0.5...but you have 0.4.
Now I have my printer busy, I'll try it as soon as it will terminate...

siteswapjuggler, 2020-01-01:
Oh sorry for that I was in the process to refine my config files and I sent you the one where I temporarely get rid of the problem by commenting #LIMITED_JERK_EDITING. My bad

GMagician, 2020-01-01:
mmm code is a little bit weird in lkimit check....
`  LIMIT(val, 1, max_limit[lim_axis]);`

jerk seems to be limited between 1 and max...but max is 0.6 when 'LIMITED_JERK_EDITING' is defined...
Need to investigate more

GMagician, 2020-01-01:
@siteswapjuggler please try proposed PR..it should fix

GMagician, 2020-01-04:
I think you may close this

siteswapjuggler, 2020-01-04:
Sorry I was on another case I forgot to reply. It works perfectly. Thanks a lot.

PR #16416 (2020-01-02): [2.0.x] Fix min limit for acc, feed and jerk
Default jerk edit limit is set to 0.6 so it has no sense a limit: 1 < jerk < 0.6

Should fix #16403
closed 2020-01-04 by siteswapjuggler