MarlinFirmware/Marlin #16406: [FR] Use Firmware Retraction before "Stop Print" function

qwewer0, 2020-01-01:
### Description

Marlin could have an option in Configuration_adv.c to apply the Firmware Retraction when the "Stop Print" is pressed in the LCD menu.

### Feature Workflow

Just a little change to the "Stop Print" function.
If Firmware Retraction is enabled, than before it turns off the heating and would park the printer, it could use the Firmware Retraction value first, than do the usual.
It would not leave a string behind the nozzle, and would not ooze.

Lord-Quake, 2020-01-01:
Have you tried it with this option in `Configuration_adv.c`  (example):

`#define EVENT_GCODE_SD_STOP "G1 E-1/nG27" // Retract 1mm of filament then park extruder`

qwewer0, 2020-01-01:
Will work for my problem, thanks.

closed 2020-01-01 by qwewer0

pmjdebruijn, 2020-01-01:
Say for example an Ender3 you'd want something like this:

```
#define NOZZLE_PARK_FEATURE
#define NOZZLE_PARK_POINT { 5, 175, 100 }
#define EVENT_GCODE_SD_STOP "G1 E-3 F3600\nG27 P2"
```

Specifically mind the `G27 **P2**`

qwewer0, 2020-01-01:
Would it work with #define EVENT_GCODE_SD_STOP "G10\nG27 P2", if firmware retraction is enabled?

pmjdebruijn, 2020-01-01:
I've never used firmware based retraction, so I don't know, however doing some brief googling would seems to suggest you might be right.

pmjdebruijn, 2020-01-01:
The other consideration is whether you want to retract the same amount at the end of a print, as opposed to during printing, since that might not per-se be the case.

qwewer0, 2020-01-01:
I will try it out, and will tell the outcome.

qwewer0, 2020-01-01:
For the first test it works great.

```
#define NOZZLE_PARK_POINT { 0, (Y_MAX_POS - 10), 20 }
#define EVENT_GCODE_SD_STOP "G10\nG27 P2\nM84"
```