MarlinFirmware/Marlin #16407: [BUG] Z axis noise)

spooky79, 2020-01-01:
Marlin 2.0.1  - Ender 3 SKR E3 mini V1.2 - very  loud work of the z axis motor , other engines silent

ellensp, 2020-01-01:
No config files. Nothing to do here.

spooky79, 2020-01-01:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration_adv.h
 *
 * Advanced settings.
 * Only change these if you know exactly what you're doing.
 * Some of these settings can damage your printer if improperly set!
 *
 * Basic settings can be found in Configuration.h
 *
 */
#define CONFIGURATION_ADV_H_VERSION 020000

// @section temperature

//===========================================================================
//=============================Thermal Settings  ============================
//===========================================================================

//
// Custom Thermistor 1000 parameters
//
#if TEMP_SENSOR_0 == 1000
  #define HOTEND0_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND0_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND0_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_1 == 1000
  #define HOTEND1_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND1_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND1_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_2 == 1000
  #define HOTEND2_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND2_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND2_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_3 == 1000
  #define HOTEND3_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND3_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND3_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_4 == 1000
  #define HOTEND4_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
[... 2769 lines cut ...]
#endif // PRUSA_MMU2

/**
 * Advanced Print Counter settings
 */
#if ENABLED(PRINTCOUNTER)
  #define SERVICE_WARNING_BUZZES  3
  // Activate up to 3 service interval watchdogs
  //#define SERVICE_NAME_1      "Service S"
  //#define SERVICE_INTERVAL_1  100 // print hours
  //#define SERVICE_NAME_2      "Service L"
  //#define SERVICE_INTERVAL_2  200 // print hours
  //#define SERVICE_NAME_3      "Service 3"
  //#define SERVICE_INTERVAL_3    1 // print hours
#endif

// @section develop

/**
 * M43 - display pin status, watch pins for changes, watch endstops & toggle LED, Z servo probe test, toggle pins
 */
//#define PINS_DEBUGGING

// Enable Marlin dev mode which adds some special commands
//#define MARLIN_DEV_MODE

spooky79, 2020-01-01:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020000

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// config/examples/SCARA and customize for your machine.
//

[... 2114 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

spooky79, 2020-01-01:
[Configuration_advh.txt](https://github.com/MarlinFirmware/Marlin/files/4014105/Configuration_advh.txt)
[Configurationh .txt](https://github.com/MarlinFirmware/Marlin/files/4014106/Configurationh.txt)

spooky79, 2020-01-01:
[platformio.txt](https://github.com/MarlinFirmware/Marlin/files/4014107/platformio.txt)

spooky79, 2020-01-01:
hybrid threshold must be //

closed 2020-01-01 by spooky79

SheelGreg, 2020-01-14:
Hello, same issue here...

Working fine with the firmware.bin of BigTreeTech (with bltouch z homing but when i compile mine, the Z motor make a weird continuous sound...
I'm a real noob, hard to find what happen :p

lamosca01, 2020-02-24:
With the new versions of Marlin, it's like going backwards. I have to use the beta.

nickvanderstel, 2020-04-20:
I still have this issue with Marlin 2.0.5.3 using the example config for the BTT SKR mini E3 v1.2. Commenting out HYBRID_THRESHOLD seems to resolve the issue.

thinkyhead, 2020-04-20:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

SheelGreg, 2020-04-20:
Thx, i'll try this asap to tell you 🙂

SheelGreg, 2020-04-21:
Ok i just put this firmware, adapted my conf files and turned on Hybrid Thresold : same weird sound, exactly (i suppose) the same when i turn off stealshop for my TMC2209 on the appropriate menu in Marlin while printing.

Maybe because i turned on something wrond in my conf files ? 
[Conf-files.zip](https://github.com/MarlinFirmware/Marlin/files/4508655/Conf-files.zip)

thinkyhead, 2020-04-21:
If you've got the ability to do `M122` then enable that and send `M122` to get some reports during different movements. The drivers might report that they have unusual settings.

If you want to try keep Z in spreadCycle because it doesn't move too far or too often anyway then for now you can send [`M569 S0 Z`](https://marlinfw.org/docs/gcode/M569.html) and then `M500` to save it to EEPROM.

Also check out these troubleshooting pages to see if anything clicks:
- https://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting
- https://marlinfw.org/docs/basics/troubleshooting.html#trinamic-stepper-drivers

ashkwil, 2020-04-24:
Hi guys I'm a complete newbie here but I compiled the latest bugfix last night, I was previously running the pre-compiled BTT firmware.

Seemed good, was storing my settings, set a print going and the Z is insanely loud I thought something was wrong I ran upstairs to check haha!

Commenting out HYBRID_THRESHOLD fixed it for me though.

SheelGreg, 2020-04-24:
> If you've got the ability to do `M122` then enable that and send `M122` to get some reports during different movements. The drivers might report that they have unusual settings.
>
> If you want to try [...]

I did that, here is the result (it's big i put it in a txt file). It's hard to understand something, even with the troubleshoting :(. 3 tests in the file, and with / without stealthChop.
If it can help...

[returned.txt](https://github.com/MarlinFirmware/Marlin/files/4531276/returned.txt)

Knifa, 2020-05-01:
Getting the same here using 2.0.5.3 with the example configs for this board. Looking at the BTT config, they also have `HYBRID_THRESHOLD` disabled, so doing this fixes it.

Haven't tried bugfix yet!