MarlinFirmware/Marlin #16408: MKS GEN L 1.0 with TFT 35

vMeph, 2020-01-01:
### Bug Description
branch 2.0.x
im using a board MKS GEN L 1.0 with TFT 35 and the  Serial is working but wont work the USB Port

### My Configurations
```
#define SERIAL_PORT 0

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
//#define SERIAL_PORT_2 2
```
serial port 2 i have tryed with 1 and also wont work
### Steps to Reproduce

i have a bigtreech tft35 when i switch to touch mode it works the screen but if i send commands via usb wont work, but i see it connects
```
Connected to: Serial<id=0x67d3b110, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Send: N0 M110 N0*125
Recv: X:231.00 Y:220.00 Z:0.00 E:0.00 Count X:36960 Y:35200 Z:0
Recv: ok
```

**Expected behavior:** [What you expect to happen]

**Actual behavior:** using the tft directly it works but wont work via usb

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-01-01:
The usb/serial converter is connected to serial0, you cannot use the USB port and the TFT on serial0 at the same time.  The two systems talk over each other.  You need to set Serial port 2 to another port. and connect that port the the TFT. But the Gen L uses the other serial port pins for other things...  Serial0 TX D1/RX D0  attached to USB serial converter. Serial1 TX D18 Z-MIN /RX D19 Z-MAX. Serial2 TX D16/RX D17  On exp2, so usable. Serial3 TX D14 Y-MIN /RX D15 Y-MAX. Your only option is to connect the TFT to serial2 pins (with a custom cable) 
and set #define SERIAL_PORT_2 2

vMeph, 2020-01-01:
@ellensp thanks for the feedback that got it to work thanks alot

closed 2020-01-01 by vMeph