MarlinFirmware/Marlin #16409: [BUG] TMC2208 state corrupts during several hours printing

jj11hh, 2020-01-01:
### Bug Description

I'm using TMC2208 drivers on my MKS SGEN-L board with bugfix-2.0.x branch.
After several hours of printing, my printer makes huge noise with the extruder stepper,
I stopped printing and checked all wire connections but found no problem, but when I
control extruder runs with high speed (with a computer), huge noise happens again, 
and all the axes move abnormally, a 10mm move becomes 20mm and more.
so I powered it off and then on, all is well now. So I restart my print again. 
But after 6 hours, it happens again. It seems the subdivisions of TMC2208 corrupts.
I set all the *_MICROSTEPS as 16, and TMC2208 runs on UART mode.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4014122/Configurations.zip)

### Steps to Reproduce

1. configure TMC2208 with UART mode and set *_MICROSTEPS to 16
2. perform a long-time print
3. and just wait

**Expected behavior:** 

all steppers run smoothly and correctly.

**Actual behavior:**

all steppers corrupt.

#### Additional Information

None

sjasonsmith, 2020-01-02:
You mentioned that you reproduced the issues with manual high-speed extrudes from a computer. can you provide instructions to reproduce the problem that way? That would be an easier way for somebody else to try to reproduce the problem.

One thing you could try to gather more information:
1. Enable the TMC_DEBUG feature in Configuration_adv.h
2. Before print:
    a. Record M122 output
    b. Verify that microsteps are properly reporting 16 microsteps for all drivers
3. Reproduce problem
4. Record M122 output, and verify the reported microstep value.

The M122 command actually reads the microstep value from the hardware. This will tell you if the microstep setting for the hardware has changed during the print.

jj11hh, 2020-01-02:
sorry, it seems a hardware issue, my board’s E0 circuit seems malfunctioned, switch to E1 now it works.

closed 2020-01-02 by jj11hh