MarlinFirmware/Marlin #16410: 1c9ccce5 abl regression

pmjdebruijn, 2020-01-01:
The following commit ( in Marlin 2.0.1 and bugfix-2.0.x ) causes a regression with ABL (bilinear):

https://github.com/MarlinFirmware/Marlin/commit/1c9ccce5209cd1727bf80e632f4f781c651e0c35

Relevant configuration:

```
#define BLTOUCH
#define AUTO_BED_LEVELING_BILINEAR
#define GRID_MAX_POINTS_X 3
#define NOZZLE_TO_PROBE_OFFSET { -44, -6, 0 }
#define MIN_PROBE_EDGE 44
#define EXTRAPOLATE_BEYOND_GRID
```

I have this specifically on an Ender 3 Pro with the SKR mini E3 board.

With that commit applied, my 9 point grid is clustered absurdly closely to each other in the center of the board, almost as-if MIN_PROBE_EDGE had been set to 80 or something along those lines.

Reverting 1c9ccce5 on 2.0.1 returns behavior to normal, so that definitively confirms 1c9ccce5 is causing this issue.

And on a sidenote shouldn't MIN_PROBE_EDGE default to abs(NOZZLE_TO_PROBE_OFFSET[1])

sjasonsmith, 2020-01-01:
This is a duplicate of #16348. PR #16367 fixes it. You can use the changes from that PR until it is merged.

pmjdebruijn, 2020-01-01:
Apologies for filing a duplicate. For the time being I'm just reverting that commit. I'll keep an eye out for when that branch will be merged.

boelle, 2020-01-01:
closing since its a dublicate

closed 2020-01-01 by boelle