MarlinFirmware/Marlin #16411: FWRETRACT causes Z? stepper to whine on boot

pmjdebruijn, 2020-01-01:
When enabling FWRETRACT on Marlin 2.0.1 one of my steppers (I think Z?) starts (an oscillating) whining on startup (and doesn't stop).

Relevant configuration:

```
#define FWRETRACT
#define RETRACT_LENGTH 6
#define RETRACT_FEEDRATE 60
```

Without FWRETRACT the printer boots up just fine.

I have this specifically on an Ender 3 Pro with the SKR mini E3 board (with TMC2209).

As a frame of reference, my firmware build script is available here (with FWRETRACT disabled):
https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2

boelle, 2020-01-01:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

pmjdebruijn, 2020-01-02:
I thought my build script might actually be clearer as it illustrates what I changed based on the Ender 3 base config, however, as per request:

[fwretractwhine_marlin_bugfix_20x_84583fc87f0a26c33.zip](https://github.com/MarlinFirmware/Marlin/files/4016798/fwretractwhine_marlin_bugfix_20x_84583fc87f0a26c33.zip)

The only difference is the FWRETRACT between a SKR mini E3 that works absolutely fine, and one that doesn't.

I had this originally with 2.0.1 but I can reproduce this with bugfix-2.0.x (specifically 84583fc8) now.

There one peculiar thing I just noticed while reproducing, which I didn't notice earlier:
1. So I put the broken fwretract firmware on my SD card
2. I boot the SKR mini E3, and the blank screen during bootup takes longer (which is a sign of it flashing the firmware, entirely as expected)
3. The machine then **boots up fine**.
4. I turn off the machine, then turn it on again, **it boots up fine**.
5. Then I init the EEPROM again, and set my Z-Offset to -2.25, and store the configuration, then turn it off, and on again, then my stepper starts **whining on bootup**, and Marlin **doesn't boot properly anymore**.

The whining almost sounds like a police siren btw. I'm not quite sure how else to describe it.

pmjdebruijn, 2020-01-02:
I think I've just had a minor epiphany, there isn't any issue with FWRETRACT at all.

My firmware.bin is too large (where enabling FWRETRACT puts is just over the threshold), where initializing the EEPROM again, effectively overwrites a bit of executable code, thus breaking Marlin, where the whining is merely a consequential effect.

My "whining" firmware.bin results in 230692 bytes.

However, when I also enable ARC_SUPPORT, I a clear message that it's not going to fit:

arm-none-eabi/bin/ld: .pio/build/STM32F103RC_bigtree/firmware.elf section `.rodata' will not fit in region `rom'
arm-none-eabi/bin/ld: region `rom' overflowed by 3988 bytes

So presumably the usable space calculation is off somewhere, so this warning is only triggered when it's exceeded by a fair margin.

So I know there's the 512KB variant, which works, so that's a viable option.

pmjdebruijn, 2020-01-02:
Presumably the following:
  buildroot/share/PlatformIO/ldscripts/STM32F103RC_SKR_MINI_256K.ld
Is incongruent with:
  ./Marlin/src/pins/stm32/pins_BTT_SKR_MINI_E3.h

pmjdebruijn, 2020-01-03:
I just built a 512K firmware and lo and behold, problem gone! So that confirms my theory.

But that does still mean that STM32F103RC_SKR_MINI_256K.ld is slightly off real world numbers. Presumably this will also be true for the 512K.ld but there it's exceedingly difficult to ever reach a practical problem.

pmjdebruijn, 2020-01-03:
Specifically:
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/buildroot/share/PlatformIO/ldscripts/STM32F103RC_SKR_MINI_256K.ld#L4
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/buildroot/share/PlatformIO/ldscripts/STM32F103RC_SKR_MINI_512K.ld#L4

boelle, 2020-01-07:
@pmjdebruijn still an issue?

boelle, 2020-01-14:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-14 by boelle

pmjdebruijn, 2020-01-14:
Sorry for not responding? But is there any reason to think this issue was resolved only few days after I posted?

I'm keeping track with regard to changes to bugfix-2.0.x and I didn't see anything come along that might fix this.

pmjdebruijn, 2020-01-14:
My best guess is these are misaligned:

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/buildroot/share/PlatformIO/ldscripts/STM32F103RC_SKR_MINI_256K.ld#L4
```
LENGTH = 256K - 28K
```

With:

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/stm32/pins_BTT_SKR_MINI_E3.h#L36
```
#define STM32_FLASH_SIZE 256
#define EEPROM_PAGE_SIZE     uint16(0x800) // 2KB
#define EEPROM_START_ADDRESS uint32(0x8000000 + (STM32_FLASH_SIZE) * 1024 - 2 * EEPROM_PAGE_SIZE)
```

However, where does the - 28K come from?

AnHardt, 2020-01-14:
> However, where does the - 28K come from?

Space used by the bootloader at the start of the flash. User program starts at 0x08007000 0x7000=28K behind the physical start of the flash at 0x08000000.

pmjdebruijn, 2020-01-14:
@AnHardt thanks!

That kinda makes sense of course, but I never thought the bootloader would be 28K, I get that it needs code for SD access and FAT, but still 28K, that's a lot of bootloader. Which is why I expected this to be a composite number.

But that does give me a clue as how to fix this. I'll probably post a pull request in the next few days.