MarlinFirmware/Marlin #16413: [BUG] (power loss recovery)

spooky79, 2020-01-01:
the function does not work, after the power failure and confirmation of resuming printing the printout starts again. Does this function work on SKR E3 mini v1.2? on ender 3 .Tested on different versions of marlin
[configurationh.txt](https://github.com/MarlinFirmware/Marlin/files/4014459/configurationh.txt)
[Configuration_advh.txt](https://github.com/MarlinFirmware/Marlin/files/4014460/Configuration_advh.txt)

Vertabreak, 2020-01-01:
it works on the A10 & A20 but much like run out the resume leaves a gap where the top/start happens making the print look like garbage i consider both features to be nothing more then a gimmick and not viable real world options.

spooky79, 2020-01-01:
on the original board it worked very well :(

heriantougm, 2020-01-02:
I'm trying to use the latest marlin 2.0.0 and 2.01 using Arduino Mega and Ramps. All runs smoothly for the power loss feature. One problem is when the printer resumes directly homing. This does not matter if the power loss is short. if the power loss is long then the nozzle has cooled and stuck to the object. If directly homing, the object is released because it is carried by the nozzle. Another possibility can damage the nozzles if the object adheres firmly. 

The question is how to change so that after the resume is not homing first but the nozzles are heated first.

boelle, 2020-01-07:
@spooky79 still an problem?

boelle, 2020-01-14:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-14 by boelle