MarlinFirmware/Marlin #16415: [BUG] Marlin 2.0.x won't build for Atmega1284p

ricardo-roseiro, 2020-01-01:
Hi

I'm trying to build Marlin 2.0.x for a Sanguinololu 1284p, I'm using PlatformIO and VS code on MacOs 10.15

**Things I've changed:**

platformio.ini
default_envs = sanguino_atmega1284p

configuration.h
#define MOTHERBOARD BOARD_SANGUINOLOLU_12

**The output:**

> Executing task: platformio run --environment sanguino_atmega1284p <

Processing sanguino_atmega1284p (platform: atmelavr; board: sanguino_atmega1284p; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/sanguino_atmega1284p.html
PLATFORM: Atmel AVR 2.0.0 > Sanguino ATmega1284p (16MHz)
HARDWARE: ATMEGA1284P 16MHz, 16KB RAM, 127KB Flash
PACKAGES: framework-arduino-avr 5.0.0, toolchain-atmelavr 1.50400.190710 (5.4.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 16 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.1
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.2.5
|-- <Adafruit MAX31865 library> 1.0.2
|   |-- <SPI> 1.0
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <TMC26XStepper>
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/HAL.cpp.o
Compiling .pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/HAL_SPI.cpp.o
In file included from Marlin/src/HAL/HAL_AVR/../../inc/../pins/pins.h:71:0,
                 from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/HAL_AVR/HAL.cpp:25:
Marlin/src/HAL/HAL_AVR/../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
Compiling .pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/MarlinSerial.cpp.o
Compiling .pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/Servo.cpp.o
In file included from Marlin/src/HAL/HAL_AVR/../../inc/../pins/pins.h:71:0,
                 from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/HAL_AVR/HAL_SPI.cpp:34:
Marlin/src/HAL/HAL_AVR/../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
In file included from Marlin/src/HAL/HAL_AVR/../../inc/../pins/pins.h:71:0,
                 from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/HAL_AVR/MarlinSerial.cpp:39:
Marlin/src/HAL/HAL_AVR/../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
In file included from Marlin/src/HAL/HAL_AVR/../../inc/../pins/pins.h:71:0,
                 from Marlin/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:32,
                 from Marlin/src/HAL/HAL_AVR/Servo.cpp:56:
Marlin/src/HAL/HAL_AVR/../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
*** [.pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/HAL.cpp.o] Error 1
*** [.pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/HAL_SPI.cpp.o] Error 1
*** [.pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/Servo.cpp.o] Error 1
*** [.pio/build/sanguino_atmega1284p/src/src/HAL/HAL_AVR/MarlinSerial.cpp.o] Error 1
================================================ [FAILED] Took 5.25 seconds ================================================

Environment                   Status    Duration
----------------------------  --------  ------------
...
sanguino_atmega1284p          FAILED    00:00:05.250
...
========================================== 1 failed, 0 succeeded in 00:00:05.250 ==========================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

I have no problems building for other boards, I only have this problem with the ones that use the Atmega1284p

Thanks

sjasonsmith, 2020-01-01:
I just  built with the current bugfix-2.0.x branch and it builds just fine. I only changed the motherboard and PIO environment, as you described.

ricardo-roseiro, 2020-01-01:
> I just built with the current bugfix-2.0.x branch and it builds just fine. I only changed the motherboard and PIO environment, as you described.

I tried to build on a Windows 10 machine, I got the same error, I simply installed PlatformIO, cloned the repository, changed the platformio.ini and configuration.h files and clicked build, the first time it downloads and installs some dependencies and when it tries to build, fails.

Do I have to install something manually?

Thanks

Grogyan, 2020-01-01:
Marlin has grown quite a lot since the early days, and many features would have to be disabled for it to fit into the code space.
I doubt it would fit anymore anyways

sjasonsmith, 2020-01-01:
@ricardo-roseiro, which branch are you trying to build?

Vertabreak, 2020-01-02:
is this one of them melzi environment boards as its also atmega1284p?

ellensp, 2020-01-02:
The error says pins_RAMPS   so to me it looks like you have not set motherboard correctly somehow Please provide actual Config files.

sjasonsmith, 2020-01-02:
> The error says pins_RAMPS so to me it looks like you have not set motherboard correctly somehow

This is a good point. I overlooked this, since so many other headers end up including pins_RAMPS.h. Looking again, this configuration should not.

If I build for the sanguino environment without changing the motherboard int the config, I see exactly the same error as reported in the description (Which is correct behavior in this case).

ricardo-roseiro, 2020-01-02:
> This is a good point. I overlooked this, since so many other headers end up including pins_RAMPS.h. Looking again, this configuration should not.
>
> If I build for the sanguino environment without [...]

I have changed the motherboard in the configuration.h to BOARD_SANGUINOLOLU_12, it was the only change I did on that file so far.

sjasonsmith, 2020-01-02:
> I have changed the motherboard in the configuration.h to BOARD_SANGUINOLOLU_12, it was the only change I did on that file so far.

Could there possibly be a simple mistake, such as editing files in a different folder than you are building from? I have certainly done that.

Can you verify that the configuration file you are editing is the file in the `Marlin` folder directly beneath the folder containing  the `platformio.ini` file you changed?

ricardo-roseiro, 2020-01-02:
> Could there possibly be a simple mistake, such as editing files in a different folder than you are building from? I have certainly done that.
>
> Can you verify that the configuration file you are [...]

You are 100% right, I was editing the wrong file, searched for "configuration.h" and edited the first one that appeared, /config/default/configuration.h instead of the on in the root

Thanks

closed 2020-01-02 by ricardo-roseiro