MarlinFirmware/Marlin #16419: [BUG] Change Tool : Offset Z error

yopla31, 2020-01-02:
I built a double head. Now I try to configure the offsets (x, y and z). 

1/ I run a G28 and, for the test, I set the offset of the second tool with M218 T1 X10 Y5 Z5. 
I position the head with G1 X150 Y150 Z0. The LCD displays X150 Y150 Z0. 
2/ I enter the T1 command to change the tool. The head rises and switches to the T1 tool, moves to X140, Y145 and Z0 and displays X150 Y150 **Z5**. For X and Y the result is good, but for Z, it is an error. 
3/ I execute the T0 command to return to the first tool. The head rises, switches to the T0 tool, returns to the initial X and Y, but remains in Z5 and displays X150 Y150 Z5.

A T1 command goto step 2 !
I crash the head in the bed !

Board : BTT skr Pro 1.1
Tested with v2.0.1 and bugfix 2.0.x (2019/12/13)

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4014688/Marlin.zip)

yopla31, 2020-01-03:
With Marlin v2.0.1 see this [vidéo](https://youtu.be/esMwMOzvaEc)
With Marlin v1.0.9 see this old [video](https://www.youtube.com/watch?v=-rRwC1Zt67w)

boelle, 2020-01-07:
@yopla31 so there is a bug? i dont understand if you are reporting a bug or asking for help

yopla31, 2020-01-07:
It is a bug. But without comment or question and stopped in my dual extrusion project, I switched to a Duet board. Unfortunately, I cannot to perform other tests to detail this bug.
My tests were run on a board (BTT SKR Pro 1.1) that has a lot of problems to drive servomotors.
Unfortunaly, I cannot perform other tests because I changed my board by a Duet2.
While waiting for questions or comments, I gave information on the same problem or almost (#16429).

GMagician, 2020-01-07:
Please compile your firmware with `DEBUG_LEVELING_FEATURE`, execute M11 S255, then your "problematic" sequence.
Report the log you get (repetier or other program). When home is done code seems correct

yopla31, 2020-01-07:
As written above, I cannot perform other tests because I changed my board by a Duet2.
I am sorry but I couldn't wait any longer. But the simulation is easy to reproduce. Only a servo correctly connected to the board is necessary.
If I can, I will repeat the tests with my other printer (under Marlin) by simulating as described above and I will keep you informed.
Sorry again.

boelle, 2020-01-07:
would it not make more sense then to close this one and create a new when and if you get time to test on the other printer`?

yopla31, 2020-01-08:
Yes it's better. When I have time to test on my other printer, I will define a specific test plan to isolate the problem.

closed 2020-01-10 by boelle