MarlinFirmware/Marlin #16422: [BUG] (TMC5161:Stop after moving a few times)

lijinlin19970813, 2020-01-02:
Bug Description：
TMC5161 cannot move after moving several axes.
Steps to Reproduce:
1.Configuration.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4015568/Configuration.zip)
2.Move XYZ axis using serial screen.
**Expected behavior:**
Send command to move axis.
**Actual behavior:**
After moving a few times, the corresponding axis can't move.
#### Additional Information
Use M122 to check until the axis stops moving. The information is correct.
[image]
move_x:
[image]
move_x_sometimes:
[image]
Description：This is the case for all axes. After a while, it cannot move, and the motor is not locked.
[＃16365](url)
@everybody，how to solve this problem.

lijinlin19970813, 2020-01-02:
[https://github.com/MarlinFirmware/Marlin/issues/16365](url)

boelle, 2020-01-07:
is this issue and #16365 the same?

lijinlin19970813, 2020-01-07:
> is this issue and #16365 the same?

Yes they are the same question.

boelle, 2020-01-07:
then i close this one, we only need one for the same 

and you said they are questions? so not a bug?

closed 2020-01-07 by boelle