MarlinFirmware/Marlin #16423: [BUG] BL-Touch not working with Sensorless Homing

BoHoehne, 2020-01-02:
### Bug Description

BL-Touch probe pin is not lowered on Z-Home.

### My Configurations

Tevo Tornado with RUMBA+ mainboard.
Sensorless homing on X and Y with tmc2130 & tmc5160 with software endstops.

BL-Touch:
Servo is configured on PWM2.
Sensor pin had to be moved to a digital pin (because of defective z-min-pin).
Selftest is working (OK)
Probe stowing and probe lowering with GCODE is working. (seems OK)
M119 shows according to state of pin TRIGGERED or OPEN (seems OK).
Changing Z_MIN_PROBE_ENDSTOP_INVERTING did not help.

Even tried another BL-Touch to check if it was defektive hardware, but the other one behaved the same.

### Steps to Reproduce

1. Configure Sensorless Homing on X and Y 
2. Configure BL-Touch
3. Modify Sanitycheck.h to allow such configuration (commenting out check blocks)
4. Home X and Y then home Z

**Expected behavior:** [What you expect to happen]

Successfull Z probing with BL-Touch

**Actual behavior:** [What actually happens]

On Z-Home the probe pin is stowed away instead of lowered.
Z-Axis is moving down and ramming into the printbed.

#### Additional Information

I know that, based on the sanity checks, this configuration is considered "unsupported".
Is there any way to get this working ?
Any hints greatly appreciated.

[MB2.0.x_SPI_ES_BL.zip](https://github.com/MarlinFirmware/Marlin/files/4015769/MB2.0.x_SPI_ES_BL.zip)

BoHoehne, 2020-01-04:
After in depth test, i came to the conclusion that this is definitely a bug in Marlin 2.0.x (bugfix).
It is impossible to negate the probe pin with Z_MIN_PROBE_ENDSTOP_INVERTING.
This way the pin is always in the TRIGGERED when stowed and OPEN when deployed.

To my understanding this would not happen if the bl-touch code was implemented in the following way:
The code should detect state changes between OPEN and TRIGGERED, no matter what the state is in stowed or deployed. The selftest should just check whether the two states differ and what reading the two states produce. This is clearly not the case. 

Such a feature would make life much easier for a lot of users.

boelle, 2020-01-24:
how often do you update? bugfix is updated almost daily

boelle, 2020-01-30:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-30 by boelle