MarlinFirmware/Marlin #16427: SD init fail SKR 1.3 onboard-card

Saschko2000, 2020-01-02:
Hello,

I have a SKR 1.3 with TMC2208 and the TFT35 V2.8 from BIGTREETECH.
I am not able to init an sd-card in the onboard-cardreader. 
Not with M21 nor with the Display.
Not with USB connected, nor without.
On my Windows PC it shows as F and works. I can copy data to and from the card.
I also tried another SD-Card, and tried both Cards
in the display-cardreader. 
They work both mounted to the PC, in the onboard reader 
and standalone in the display reader,
but not standalone in the onboard reader :-(

I googeled an read a lot, but can't find a solution.

Please excuse my english and thanks for any help 
  
Sascha :-)

[MarlinConfigs.zip](https://github.com/MarlinFirmware/Marlin/files/4017139/MarlinConfigs.zip)

ellensp, 2020-01-02:
Is sounds to me like you have the SKR 1.3 power jumper set to USB, and then disconnected the USB So the main board has no power.

Saschko2000, 2020-01-03:
The board has power. I double checked this. Otherwise the Display woould not work. 
In other threads I read about setting #define LPC_SD_ONBOARD. But in the actual Configuration_adv.h I can't find anything with LPC. I dont' know what else I can try :-(

Saschko2000, 2020-01-03:
I am a step further: Change 

#define SDCARD_CONNECTION LCD
to
#define SDCARD_CONNECTION ONBOARD

Worked. 

But now I got EPROM VERSION MISSMATCH... :-(

Saschko2000, 2020-01-03:
OK, I managed this with M502 M500 M501.

Now the SD-Card works for one print. 
After the print I get READ ONBOARD SD CARD ERROR :-(
After a Reset everything works again.

boelle, 2020-01-07:
how many sd cards have you tried with?
have you verified the cards as 100% working and if yes how did you verify them?

Saschko2000, 2020-01-07:
I tried with 4 different cards. And they work fine on my other printer. I have ordered a new SKR 1.3 an will test further when I have it here. 
For now I close this issue, because it is no more the mentioned error.

Thanks to all for your help.

closed 2020-01-07 by Saschko2000