MarlinFirmware/Marlin #16428: [BUG] Anycubic i3 Mega Marlin 1.1.9 and 2.0.x BLTouch Original Version 3.1 not working

RedJohn14, 2020-01-02:
Hello,

**I installed Marlin Build from your repo here: (v.1.1.9)**
https://github.com/MNieddu91/Marlin-AI3M-BLTouch

**My settings in configuration.h looks like this:**
```
`//#define PROBE_MANUALLY
//#define MANUAL_PROBE_START_Z 0.2

/**

A Fix-Mounted Probe either doesn't deploy or needs manual deployment.
(e.g., an inductive probe or a nozzle-based probe-switch.)
*/
//#define FIX_MOUNTED_PROBE
/**

Z Servo Probe, such as an endstop switch on a rotating arm.
*/
//#define Z_PROBE_SERVO_NR 0 // Defaults to SERVO 0 connector.
//#define Z_SERVO_ANGLES {10,90} // Z Servo Deploy and Stow angles
/**

The BLTouch probe uses a Hall effect sensor and emulates a servo.
*/
define BLTOUCH
if ENABLED(BLTOUCH)
define BLTOUCH_DELAY 100 // (ms) Enable and increase if needed

/**

BLTouch V3.0 and newer smart series
For genuine BLTouch 3.0 sensors. Clones may be confused by 3.0 command angles. YMMV.
If the pin trigger is not detected, first try swapping the black and white wires then toggle this.
*/
//#define BLTOUCH_V3
if ENABLED(BLTOUCH_V3)
define BLTOUCH_FORCE_5V_MODE
define BLTOUCH_FORCE_OPEN_DRAIN_MODE
endif
endif`
```

**I used the original BLTouch v3.1 from here:**
https://www.amazon.de/…/B075B57CHD/ref=ppx_yo_dt_b_asin_tit…

**I also tested your new repo Marlin 2.0.x:**
https://github.com/MNieddu91/Marlin-Ai3M-2.0.x-BLTouch

I have home everything with G28 and use G29 all works perfekt. BLTouch Pin comes out but if I TRIGGERED it from hand nothing happens Z axis not stop!

**If I home everything again with G28 and looks with M119 I see this:**
< 13:10:36: x_min: TRIGGERED
< 13:10:36: y_min: TRIGGERED
< 13:10:36: z_min: TRIGGERED
< 13:10:36: z2_min: TRIGGERED
< 13:10:36: z_probe: open

The z_probe status won't change. I TRIGGERED the Pin manual but the status won't change. The Server cable on my motherboard is correct because I can use this command without any problem:

M280 P0 S10 ; pushes the pin down
M280 P0 S90 ; pulls the pin up
M280 P0 S120 ; Self test – keeps going until you do pin up/down or release alarm
M280 P0 S160 ; Release alarm

Only the trigger status will not work Z axis not stop! Any idea what the problem is?

**Here is also a video from the printer:**
https://drive.google.com/open?id=1cAup0ubzmK27F0MHrlfSdWe8jBjmkSl9

Attached you find pictures and a video. Ideas? Maybe someone can help me? I tested everything 4 days now 😞 😞I hope someone can help me ...

Attached you find a picture from my BLTouch mount in the printers head. And a picture from the cable on my motherboard. I have the new Board 1.1. I have an Anycubic i3 Mega Printer.
[image]
[image]

boelle, 2020-01-02:
https://github.com/MNieddu91/Marlin-AI3M-BLTouch is not marlin repro

use this one instead: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

boelle, 2020-01-02:
and please insert both configuration.h and configuration_adv.h in a zip file here

RedJohn14, 2020-01-02:
> https://github.com/MNieddu91/Marlin-AI3M-BLTouch is not marlin repro
> 
> use this one instead: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

boelle, 2020-01-02:
i have checked the link and its a fork/copy of another repro that does not even go back to marlin

we do not support forks as we do not know what it is, it could have been changed in many ways

closing this one, create a new issue and use the link above, and please read the issue template first before posting

closed 2020-01-02 by boelle

RedJohn14, 2020-01-02:
> i have checked the link and its a fork/copy of another repro that does not even go back to marlin
>
> we do not support forks as we do not know what it is, it could have been changed in many ways [...]

Okay thanks for the information. How can I configure Marlin for my Anycubic i3 Printer withe Trigorilla Board 1.1 and the newest original BLTouch Sensor Version 3.1? I read some much that the BLTouch 3.1 have problems with Marlin. Many Thanks for any help! :-)

borislavstefanov, 2020-03-19:
do you have any progress with trigorilla's bltouch because i have the same problem

koke21127, 2020-03-20:
> do you have any progress with trigorilla's bltouch because i have the same problem

> Okay thanks for the information. How can I configure Marlin for my Anycubic i3 Printer withe Trigorilla Board 1.1 and the newest original BLTouch Sensor Version 3.1? I read some much that the [...]

> do you have any progress with trigorilla's bltouch because i have the same problem

Hello, I have got that servo works. If you have trigorilla motherboard, in pin_TRIGORILLA_14.h change the servo0 pin to 5. In my case is the unique form that it works.

borislavstefanov, 2020-03-20:
isn't about trigorilla and not trigorilla 1.1 it's about me being commodity. i upload a video to see where i got to and no more
https://drive.google.com/folderview?id=1ct8B_XOW6SDS1EnN32-f0Q02Cr-GXyz6