MarlinFirmware/Marlin #16429: [BUG] Switching nozzle tool change mouvement error

TouffeTouffe, 2020-01-02:
Using a switching nozzle and extruder using the same servo ( https://www.thingiverse.com/thing:673816 ) when executing a tool change, the tool's mouvement are not the ones expected:

Expected moves:
-Raise z by TOOLCHANGE_ZRAISE 
-Rotate servo
-Move head to compensate HOTEND_OFFSET_X ,Y and Z
-Lower z by TOOLCHANGE_ZRAISE 

Actual moves:
-Rotate servo
-Lower z by a certain distance, which isn't TOOLCHANGE_ZRAISE

I am using the latest bugfix 2.0.x 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4017240/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4017241/Configuration_adv.zip)

GMagician, 2020-01-03:
Hi, expected moves should be:
-Raise z by TOOLCHANGE_ZRAISE
-Rotate servo
-Move head to compensate HOTEND_OFFSET_X and Y
-Lower z by TOOLCHANGE_ZRAISE + HOTEND_OFFSET_Z

Some times ago this was working (I had Dondolo) but a lot of changes have been done since then (and I don't use Dondolo anymore).

TouffeTouffe, 2020-01-03:
You're right, I forgot to mention the z offset.
But I had set mine to zero so it hadn't crossed my mind.
The result is that changing multiple times from T0 to T1 and back to T0 causes the z to come crashing down on the bed.

GMagician, 2020-01-03:
@thinkyhead I see you changed some lines in tool_change.cpp (some months ago) line 932 (used with Dondolo) does:
`current_position.z += _MAX(-diff.z, 0.0) + toolchange_settings.z_raise;`
while just below (line 946):
`current_position += diff;`
doesn't this lead real position mismatch?

TouffeTouffe, 2020-01-03:
I reflashed the firmware but commenting all HOTEND_OFFSET_X Y and Z. And the printer behaves like it should during tool change (goes up, rotates servo, goes down).
It appears that when I define a hotend offset, positive or negative, something goes wrong. It doesn't matter if I have an offset stored in eeprom or not.
My "fix" is to comment all the offsets in configuration.h and do all the offset configuration in eeprom but the bug is still here.

GMagician, 2020-01-03:
@TouffeTouffe just to resume, please check out what condition work
1) Home done
   a) TOOLCHANGE_ZRAISE  <> 0
     I) positive HOTEND_OFFSET_Z
     II) negative HOTEND_OFFSET_Z
   b) TOOLCHANGE_ZRAISE  = 0
     I) positive HOTEND_OFFSET_Z
     II) negative HOTEND_OFFSET_Z
2) Home undone
   a) TOOLCHANGE_ZRAISE  <> 0
     I) positive HOTEND_OFFSET_Z
     II) negative HOTEND_OFFSET_Z
   b) TOOLCHANGE_ZRAISE  = 0
     I) positive HOTEND_OFFSET_Z
     II) negative HOTEND_OFFSET_Z

yopla31, 2020-01-03:
In my case #16419 , even with all axes homed, the issue appears. My TOOLCHANGE_ZRAISE is default value (= 2) and I set the offsets with M218 T1 X10 Y5 Z5 to check all offsets. See my video.

Vertabreak, 2020-01-03:
Wonder if the 5 is from z raise of the probe and if it where printing it would go back to 0.

GMagician, 2020-01-04:
@yopla31 and @Vertabreak please note that different tools changer execute different code. Dondolo (switching_nozzle & switching_extruder) are a specific path and if only "not homed" is the issue I may have found how to solve it

GMagician, 2020-01-04:
@TouffeTouffe may you try fix I pushed on
https://github.com/GMagician/Marlin/tree/Fix-16429

yopla31, 2020-01-04:
I use the SWITCHING_NOZZLE :
_// A dual-nozzle that uses a servomotor to raise/lower one (or both) of the nozzles
#if EXTRUDERS > 1
  #define SWITCHING_NOZZLE  //JBL-E2
//#endif
//#if ENABLED(SWITCHING_NOZZLE)
  #define SWITCHING_NOZZLE_SERVO_NR 1
  //#define SWITCHING_NOZZLE_E1_SERVO_NR 1          // If two servos are used, the index of the second
  #define SWITCHING_NOZZLE_SERVO_ANGLES { 72, 27 }   // Angles for E0, E1 (single servo) or lowered/raised (dual servo)
#endif_
and I execute a G28, see at the beginning of my [video](https://youtu.be/esMwMOzvaEc).

I tried a new test : 

- M218 X5 Y0 Z-5          (X = 5 by security)
- M500
- G28                             LCD=> X154 Y190 Z10.63  (Probe in X0 Y-49 Z-0.63)
- G1 X150 Y150 Z0        LCD=> X150 Y150 Z15 (z=15, yes!) and the head raises to 15mm
- T1                                LCD=> X150 Y150 Z15 the move in x is correct and the head raises by TOOLCHANGE_ZRAISE and down to 20 mm, 15 -(offset Z) =>15-(-5)=>15+5
- T0                                LCD=> X150 Y150 Z15 the head raises by TOOLCHANGE_ZRAISE and down to 15

Note : 
_#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES 10 // 5 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE    10 // 5 // Z Clearance between multiple probes
//#define Z_AFTER_PROBING           5 // Z position after probing is done_

8^(

TouffeTouffe, 2020-01-04:
@GMagician I'll try your fix when I get access again at the machine and tell you how it behaves.
Thanks for the support!

yopla31, 2020-01-04:
Compilation error. :-(
I compiled without the updates of libraries and platform (stm32 5.7.0)

Marlin\src\feature\pause.cpp: In function 'bool unload_filament(const float&, bool, PauseMode)':
Marlin\src\feature\pause.cpp:355:20: error: 'FILAMENT_UNLOAD_PURGE_FEEDRATE' was not declared in this scope
                   (FILAMENT_UNLOAD_PURGE_FEEDRATE) * mix_multiplier);
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\feature\pause.cpp:355:20: note: suggested alternative: 'FILAMENT_UNLOAD_PURGE_RETRACT'
                   (FILAMENT_UNLOAD_PURGE_FEEDRATE) * mix_multiplier);
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                    FILAMENT_UNLOAD_PURGE_RETRACT
*** [.pio\build\BIGTREE_SKR_PRO\src\src\feature\pause.cpp.o] Error 1
=================== [FAILED] Took 32.62 seconds 
Environment                   Status    Duration
----------------------------  --------  ------------
megaatmega2560                IGNORED
...
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               FAILED    00:00:32.620
BIGTREE_BTT002                IGNORED
...
=== 1 failed, 0 succeeded in 00:00:32.620

yopla31, 2020-01-04:
With updates, same error.
[image]

GMagician, 2020-01-04:
@yopla31 about compilation error: please upgrade your config files `FILAMENT_UNLOAD_PURGE_FEEDRATE` has been added recently

GMagician, 2020-01-04:
@yopla31 

> I use the SWITCHING_NOZZLE :

Fix works only when both `SWITCHING_NOZZLE` and `SWITCHING_EXTRUDER` are both enabled

yopla31, 2020-01-04:
I have an extruder and its motor per nozzle, so not a dual extruder that uses a single stepper motor.
//#define SWITCHING_EXTRUDER

[image]

GMagician, 2020-01-04:
@yopla31 I correct myself, fix will work when `SWITCHING_NOZZLE` is enabled and not defined `SWITCHING_NOZZLE_E1_SERVO_NR`. But it works onlòy when no axis homed, otherwise it is untouched

GMagician, 2020-01-07:
@TouffeTouffe have tested my fix?

TouffeTouffe, 2020-01-07:
@GMagician Not yet as I only have access to the machine on Wednesdays and Fridays, but I'll be sure to try it out tomorrow evening and tell you how  it went

GMagician, 2020-01-07:
> Not yet as I only have access to the machine on Wednesdays and Fridays, but I'll be sure to try it out tomorrow evening and tell you how it went

Remember also to answer to:

    Home done
    a) TOOLCHANGE_ZRAISE <> 0
    I) positive HOTEND_OFFSET_Z
    II) negative HOTEND_OFFSET_Z
    b) TOOLCHANGE_ZRAISE = 0
    I) positive HOTEND_OFFSET_Z
    II) negative HOTEND_OFFSET_Z
    Home undone
    a) TOOLCHANGE_ZRAISE <> 0
    I) positive HOTEND_OFFSET_Z
    II) negative HOTEND_OFFSET_Z
    b) TOOLCHANGE_ZRAISE = 0
    I) positive HOTEND_OFFSET_Z
    II) negative HOTEND_OFFSET_Z

TouffeTouffe, 2020-01-08:
With Home done, everything works as it should, in all a and b cases.
With Home not done, the machine just rotates the servo and doesn't move up or down so that's slightly better. I didn't have the time to test the effects of changing the z offset in the "not homed" scenario though.

boelle, 2020-01-08:
so we will have to wait a bit more to see if this one is solved?

GMagician, 2020-01-09:
> With Home not done, the machine just rotates the servo and doesn't move up or down

When home is not done there is a "no_move" flag inside Marlin to prevent axis moving. So this behaviour is almost expected. 
I'm expecting that z position is changing by the z_offset. Please confirm that.
I just want to add a "z_max" protection because now movement will be limited but this may lead to a not complete z_offset move and then to a similar bug

boelle, 2020-01-14:
since the PR is merged i will close this one, we can reopen if the issue is still there

PR #16518 (2020-01-10): Prevent Z misaligments on tool change
This should prevent z misaligments when changing tool when axis no homed (Fix #16429) 
This also prevent z misaligment, when axis homed, when tool change is done too near z max limit (when defined)

Only valid for SWITCHING_NOZZLE configuration
closed 2020-01-14 by boelle