MarlinFirmware/Marlin #16438: [BUG] G34 Compile error

Aurawin, 2020-01-03:
Recently there have been some changes experienced with G34.  

Examination of X axis only
#define X_MIN_POS -4
#define Y_MIN_POS -18
#define Z_MIN_POS 0
#define X_MAX_POS 347
#define Y_MAX_POS 345
#define Z_MAX_POS 350
  #define MIN_PROBE_EDGE_LEFT  5
  #define MIN_PROBE_EDGE_RIGHT 5
  #define MIN_PROBE_EDGE_FRONT 5
  #define MIN_PROBE_EDGE_BACK 5
#define NOZZLE_TO_PROBE_OFFSET { 37, 32, -0.75 }

Error "The 1st Z_STEPPER_ALIGN_XY X is unreachable with the default probe X offset."

Fail: #define Z_STEPPER_ALIGN_XY { {  5, 165 }, { 280,  165 } } // <---- Ideal value [0].X
Fail: #define Z_STEPPER_ALIGN_XY { {  20, 165 }, { 280,  165 } }
Fail: #define Z_STEPPER_ALIGN_XY { {  32, 165 }, { 280,  165 } }
Pass: #define Z_STEPPER_ALIGN_XY { {  33, 165 }, { 280,  165 } }

if (Z_STEPPER_ALIGN_XY.X >= NOZZLE_TO_PROBE_OFFSET.X + X_MIN_POS)) {
  //pass
}
if (Z_STEPPER_ALIGN_XY.X>=X_MIN_POS + MIN_PROBE_EDGE_LEFT){
  // should pass
}

I expect that all values greater than X_MIN_POS + MIN_PROBE_EDGE_LEFT are valid and should pass this safety check.

As is - I cannot probe the bed's X_MIN edge to accurately align my steppers.

tpruvot, 2020-01-03:
if you have a Xmin endstop at 0, you cannot go under it, unless you can teleport your nozzle..

NOZZLE_TO_PROBE_OFFSET 37 32 means... Probe on the Right, and on the Front...
MIN_PROBE_EDGE_LEFT and MIN_PROBE_EDGE_FRONT need adjustment...

Aurawin, 2020-01-03:
> if you have a Xmin endstop at 0, you cannot go under it, unless you can teleport your nozzle..
> 
> NOZZLE_TO_PROBE_OFFSET 37 32 means... Probe on the Right, and on the Back*...
> MIN_PROBE_EDGE_LEFT and MIN_PROBE_EDGE_BACK need adjustment...

My X-min endstop is at -4.  It takes 4 mm to get to my bed, from my X-min.  
My Y-min endstop is -18.  It takes 18mm to get to my bed from the Y-min.
Nozzle is indeed right where I specified.  Or the Probe is to the right and behind the nozzle.

I assumed that MIN_PROBE_EDGE_LEFT is 5mm from the start of my bed or 4+5=9mm from the X-min endstop.  

As I see it - something is wonky

ManuelMcLure, 2020-01-03:
The probe needs to reach the point you specify. Your first Z alignment point has an X of 5, but your probe can’t go to less than X=33 (X_MIN_POS + 37).

Aurawin, 2020-01-03:
> (X_MIN_POS + 37)

Yes.  This is the problem.  My Core XY can probe with the nozzle off the bed, however, I should be able to specify where I want my probe to measure.  Ie.) the  X_MIN_POS is NEEDING to be considered in the min assertion calculation.

Previous versions haven't had this issue.  I have been able to probe my left side in the past.

Aurawin, 2020-01-03:
#define X_BED_SIZE 330
#define Y_BED_SIZE 330

On a Core XY the Minimum Extent Calculation should be:
XMINCheck = X_MIN_POS + MIN_PROBE_EDGE_LEFT should be used. Ie the actual lower limit should be -4 + 5 = 1mm.  The nozzle and probe are still on the bed.
Also...
XMAXCheck= X_MIN_POS  + X_BED_SIZE - NOZZLE_TO_PROBE_OFFSET.X - MIN_PROBE_EDGE_RIGHT should be used.  Ie. the actual high limit should be -4 + 330 - 37 -5 =284mm. The nozzle and probe are still on the bed and are exactly where they should be.

And looking at G29 the MAX values are now also miscalculated.  My probe is off the bed... ARRG...

G34 AND G29 - All of the extent values are presently calculated improperly.  

This is going to cause my nozzle to probe beyond my bed on MAX_POS.

sjasonsmith, 2020-01-03:
@Aurawin, the failure is new because the compile-time checks were only added a few days ago. You wouldn't have noticed this previously until you ran G34 and it didn't work. It would execute blazingly fast, and do absolutely nothing due to the unreachable probe points.

No matter how hard you tell Marlin to probe at X=5, it simply cannot move the probe to that location. As @ManuelMcLure mentioned, your X_MIN_POS and NOZZLE_TO_PROBE_OFFSET are telling Marlin the point cannot be reached.

If your probe actually can reach X=5, you need to modify your X_MIN_POS to allow your nozzle to travel further to the left.

sjasonsmith, 2020-01-03:
> XMINCheck = X_MIN_POS + MIN_PROBE_EDGE_LEFT should be used. Ie the actual lower limit should be -4 + 5 = 1mm. The nozzle and probe are still on the bed.

The probe points are the position of the probe, _not_ the position of the nozzle.

Aurawin, 2020-01-03:
> @Aurawin, the failure is new because the compile-time checks were only added a few days ago. You wouldn't have noticed this previously until you ran G34 and it didn't work. It would execute blazingly fast, and do absolutely nothing due to the unreachable probe points.

I have been aligning my steppers daily for a year now.  G34 and G29 don't work as expected now.

> No matter how hard you tell Marlin to probe at X=5, it simply cannot move the probe to that location. As @ManuelMcLure mentioned, your X_MIN_POS and NOZZLE_TO_PROBE_OFFSET are telling Marlin the [...]

Aurawin, 2020-01-03:
> > XMINCheck = X_MIN_POS + MIN_PROBE_EDGE_LEFT should be used. Ie the actual lower limit should be -4 + 5 = 1mm. The nozzle and probe are still on the bed.
> 
> The probe points are the position of the probe, _not_ the position of the nozzle.

Exactly.  This is the problem.

Aurawin, 2020-01-03:
> > XMINCheck = X_MIN_POS + MIN_PROBE_EDGE_LEFT should be used. Ie the actual lower limit should be -4 + 5 = 1mm. The nozzle and probe are still on the bed.
> 
> The probe points are the position of the probe, _not_ the position of the nozzle.

Agreed.  This is the problem.  The probe is not reaching the target MIN and is way overshooting the Max (on G29)

Aurawin, 2020-01-03:
I can no longer issue a G29.

sjasonsmith, 2020-01-03:
I recommend you focus on one problem in this issue. The more problems get piled into this issue, the harder it is for anybody to make sense of it.

sjasonsmith, 2020-01-03:
Let me look over the checks that were added, to see if I spot any issues in the logic.

In the meantime, can you post your actual configuration files?

Also, I think you have a misconception about MIN_PROBE_EDGE. That is from the edge of the bed, not from the min/max travel positions. Normally MIN_PROBE_EDGE is only needed if your probe cannot probe directly at the edge of the bed. 10mm is default, but it can be more or less depending on your probe. NOZZLE_TO_PROBE_OFFSET will be used along with that to determine the actual probing range.

Aurawin, 2020-01-03:
> Let me look over the checks that were added, to see if I spot any issues in the logic.

The logic changes required are specified in a post above.

> In the meantime, can you post your actual configuration files?
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4018412/Configuration.zip)

> Also, I think you have a misconception about MIN_PROBE_EDGE. That is from the edge of the bed, not from the min/max travel positions. Normally MIN_PROBE_EDGE is only needed if your probe cannot [...]

MIN_PROBE_EDGE is for avoiding clips and the like.  My bed is magnetic with no such restrictions.

Aurawin, 2020-01-03:
I just noticed that the actual probe is not in the center of my bed.  The probe while homing, used to be spot on center.  There was a recent logic change to positioning that is in error.

sjasonsmith, 2020-01-03:
OK, I reviewed the checks, and they look fine to me.

> I expect that all values greater than X_MIN_POS + MIN_PROBE_EDGE_LEFT are valid and should pass this safety check.

This is not correct. MIN_PROBE_EDGE_LEFT is how far inside the bed you want to probe, not inside the travel area. Even within your defined probing edges, it can't probe something the probe cannot physically reach.

> Fail: #define Z_STEPPER_ALIGN_XY { { 5, 165 }, { 280, 165 } } // <---- Ideal value [0].X

If you want to probe when the nozzle is at this position, you need to convert these into probe coordinates. That would be:
`#define Z_STEPPER_ALIGN_XY { { 42, 197}, { 317, 197} }`

> The logic changes required are specified in a post above.

But they are wrong...
If you already know all the answers, you should go ahead and submit a pull request with the changes, and they can be examined there.

Aurawin, 2020-01-03:
> This is not correct. MIN_PROBE_EDGE_LEFT is how far inside the bed you want to probe, not inside the travel area. Even within your defined probing edges, it can't probe something the probe cannot physically reach.

I think the problem becomes easier to understand if you look at relative vs absolute coordinates.  My calculations are using relative calculated values from the absolute minimum of travel to the edges of the bed, to the size of the bed.

The checks are NOT made properly here.  A value of 4 is acceptable because the probe is TO THE RIGHT of the nozzle.  ie.) the bed is at X=0 mm the min edge add to that.  The probe offset is NOT relavent if the nozzle is to the left of the probe.

sjasonsmith, 2020-01-03:
> The probe offset is NOT relavent if the nozzle is to the left of the probe.

Sure it is. Your probe cannot move to X=4.

Aurawin, 2020-01-03:
> Sure it is. Your probe cannot move to X=4.

Ok.  If that is how the logic is done, I can understand why probes are falling off beds.  The X,Y coordinate system is based on the nozzle position - not the probe's position.  That is to say when the probe is where it needs to be it's the NOZZLE that Marlin sees.  Any offset calculated is suspect.

sjasonsmith, 2020-01-03:
I think I am done with this issue and turning off notifications for it.

Aurawin, 2020-01-04:
> If your probe actually can reach X=5, you need to modify your X_MIN_POS to allow your nozzle to travel further to the left.

My X_MIN_POS is -4  -> decreasing this value to allow for my G34 to work is an unacceptable solution.  This is what defines my bed's zero location.  If I change this my gantry will smash into my left 2020

Aurawin, 2020-01-04:
@ManuelMcLure 

> The probe needs to reach the point you specify. Your first Z alignment point has an X of 5, but your probe can’t go to less than X=33 (X_MIN_POS + 37).

Ok. I realize this is the problem.  This is what I am saying.  If you look at this from a user's prospective, X is 5mm on the screen and the nozzle is at 5mm and the probe is at 37+5mm  That is where the probe is, and that is where Marlin knows the nozzle (not probe).

Any logical calculations are based on the coreXY's X,Y,Z position - not the probe.  If I can't get Marlin to accept a position of 5mm for probing my bed - that's a problem for probes located to the righ of the nozzle.

What goes on behind the scenes is the transformation from X,Y to ProbeX, and ProbeY.  Somethign needs to give.

ManuelMcLure, 2020-01-04:
If your nozzle is at X=-4, you’re probing at X=33. The measurement taken will be used to correct the nozzle when the nozzle is at X=33. If you tried to say that the position of the nozzle is where you’re probing, then mesh bed leveling would be useless because you’re not measuring the bed at the point where the correction needs to occur. It might sort of work for 3 point leveling but not for mesh leveling (UBL or Bilinear).

Aurawin, 2020-01-04:
I use Bilinear.  The nozzle  is EXACTLY where it should be on the left side of my bed.  The nozzle starts (Note Nozzle) far right of my bed and works its way to the left and right and left etc.  So my constraints are good on the left side of the bed for meshing bilinear.

No complaints here regarding ABL.  The minimum contraint is only issue here.  Looking at the left side of the printer the probe should be able to probe with my nozzle at 5mm.  5mm is my edge offset left.

The bug is that if I specify Edgeoffset + NPO (37mm) my probe nozzle is not at 5mm it's at 37+
5mm.  Or if I specify 5mm on my left it should take care of this on the backend - not in the config file. 

In the end, I expect my nozzle to be at 5mm physically and my probe to be at 5+NPO during a left Z G34

Aurawin, 2020-01-04:
> if you have a Xmin endstop at 0, you cannot go under it, unless you can teleport your nozzle..
> 
> NOZZLE_TO_PROBE_OFFSET 37 32 means... Probe on the Right, and on the Back*...
> MIN_PROBE_EDGE_LEFT and MIN_PROBE_EDGE_BACK need adjustment...

My config file comments show:

 * Override MIN_PROBE_EDGE for each side of the build plate
 * Useful to get probe points to exact positions on targets or
 * to allow leveling to avoid plate clamps on only specific
 * sides of the bed. With NOZZLE_AS_PROBE negative values are
 * allowed, to permit probing outside the bed.

Ok. I fail to see how MIN_PROBE_EDGE_LEFT and MIN_PROBE_EDGE_BACK all of a sudden went to collision avoidance to calculating a transformation from NozzleX,Y to ProbeX,Y.  Did something change?

ManuelMcLure, 2020-01-04:
What do you mean by “probe nozzle”? The nozzle or the probe? If you specify your G34 probing point as X=42 you should see the probe at X=42 and the nozzle at X=5.

Aurawin, 2020-01-04:
Reloaded default config and it started working as expected.

closed 2020-01-04 by Aurawin