MarlinFirmware/Marlin #16441: [BUG] Joystick Feature exceeds printer dimentions

kurtnelle, 2020-01-03:
### Bug Description

It is possible to command the printer/cnc to exceed the constraints of the machine resulting in the axis crashing into the frame. 

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Deflect joystick until the axis crashes into the frame and the steppers begin to stall.

**Expected behavior:** I expect it to stop when it reaches bed max or zero

**Actual behavior:** Axis continues to try to move far beyond the frame, with the xyz position increasing accorgingly. (e.g. my machine has a max x of 480 but the joystick can drive the machine to 1000 or more!)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-01-03:
"Required: Please include a ZIP file containing your Configuration.h and Configuration_adv.h files."

kurtnelle, 2020-01-03:
@ellensp, files attached.

ellensp, 2020-01-03:
now what boundary do your want it to respect from the config files? I guessing SOFTWARE_ENDSTOPS ??  If so is this the same issues as https://github.com/MarlinFirmware/Marlin/issues/16373 ?

kurtnelle, 2020-01-03:
Yes, it would be software endstops. Therefore I should enable software endstops and try again?

boelle, 2020-01-03:
closing as this is a dublicate of #16373
we only need one open for the same issue

closed 2020-01-03 by boelle