MarlinFirmware/Marlin #16446: const_functions.h missing. is it possible?

lospo, 2020-01-03:
When i try to build i have this. downloaded today.
Last time i use atom to compile, but that version doesn't work anymore , i saved my configration.h, configuration_adv.h, platformio.ini and try to rebuild but i have this error.
`

> Executing task: C:\Users\francesco\.platformio\penv\Scripts\platformio.exe run <

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Can not remove temporary directory `c:\Marlin2\.pioenvs`. Please remove it manually to avoid build issues
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; framework: arduino; board: nxp_lpc1768)
------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.0.2 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.80KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.2.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.6.1
|-- <Adafruit NeoPixel> 1.2.4
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or
copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
In file included from c:\marlin2\marlin\src\hal\shared\marduino.h:36,
                 from c:\marlin2\marlin\src\hal\hal_lpc1768\hal.h:39,
                 from c:\marlin2\marlin\src\hal\hal.h:26,
                 from c:\marlin2\marlin\src\inc\marlinconfig.h:30,
                 from c:\marlin2\marlin\src\core\serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
C:\Users\francesco\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:35:10: fatal error: const_functions.h: No such file or directory

*************************************************************************
* Looking for const_functions.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:const_functions.h"
* Web  > https://platformio.org/lib/search?query=header:const_functions.h
*
*************************************************************************

 #include <const_functions.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o] Error 1
In file included from c:\marlin2\marlin\src\hal\shared\marduino.h:36,
                 from c:\marlin2\marlin\src\hal\hal_lpc1768\hal.h:39,
                 from c:\marlin2\marlin\src\hal\hal.h:26,
                 from c:\marlin2\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
C:\Users\francesco\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:35:10: fatal error: const_functions.h: No such file or directory

*************************************************************************
* Looking for const_functions.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:const_functions.h"
* Web  > https://platformio.org/lib/search?query=header:const_functions.h
*
*************************************************************************

 #include <const_functions.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o] Error 1
In file included from c:\marlin2\marlin\src\hal\shared\marduino.h:36,
                 from c:\marlin2\marlin\src\hal\hal_lpc1768\hal.h:39,
                 from c:\marlin2\marlin\src\hal\hal.h:26,
                 from c:\marlin2\marlin\src\inc\marlinconfig.h:30,
                 from Marlin\src\HAL\HAL_LPC1768\HAL_SPI.cpp:51:
C:\Users\francesco\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:35:10: fatal error: const_functions.h: No such file or directory

*************************************************************************
* Looking for const_functions.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:const_functions.h"
* Web  > https://platformio.org/lib/search?query=header:const_functions.h
*
*************************************************************************

 #include <const_functions.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o] Error 1
========================================================== [FAILED] Took 4.74 seconds ==========================================================

Environment           Status    Duration
--------------------  --------  ------------
megaatmega2560        IGNORED
megaatmega1280        IGNORED
at90usb1286_cdc       IGNORED
at90usb1286_dfu       IGNORED
DUE                   IGNORED
DUE_USB               IGNORED
DUE_debug             IGNORED
LPC1768               FAILED    00:00:04.741
LPC1769               IGNORED
melzi                 IGNORED
melzi_optiboot        IGNORED
rambo                 IGNORED
sanguino_atmega644p   IGNORED
sanguino_atmega1284p  IGNORED
STM32F1               IGNORED
STM32F4               IGNORED
ARMED                 IGNORED
mks_robin             IGNORED
black_stm32f407ve     IGNORED
teensy35              IGNORED
malyanm200            IGNORED
esp32                 IGNORED
fysetc_f6_13          IGNORED
linux_native          IGNORED
==================================================== 1 failed, 0 succeeded in 00:00:04.741 ====================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
`

ellensp, 2020-01-03:
Your system is broken. Delete c:\Marlin2\.pioenvs  as the error tells you to do.

boelle, 2020-01-04:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

GMagician, 2020-01-04:
It seems also that platformio.ini is not updated as it should. Consider to download also fresh sources

lospo, 2020-01-04:
i think that the passage from atom to vscode was a trauma...
sorry for the missed attach.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4022105/Marlin.zip)
[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/4022106/platformio.zip)

GMagician, 2020-01-04:
Don't know where have you got your platform.ini but
`env_default = LPC1768` is not correct, this was used in pre 4.0 platformio.
Comparing current 2.0 with yours...nothing match...this is not coming from a fresh downloaded Marlin 2.0.x

lospo, 2020-01-05:
Please forget this and forgive me.
I used old files from old configuration, it is all my fault, after i changed with fresh files with my configuration everithing worked.
Thank you!
And sorry for waste your time.

closed 2020-01-05 by lospo