MarlinFirmware/Marlin #16450: [BUG] Cannot compile Marlin 2.0.X *Tried everything google has to offer*

ThatOneGingie, 2020-01-03:
Hey, so i've been doing research for like 3 hours and cannot find a fix for my compile issue. 
I have done the following:
1. Reinstalled marlin
2. Reinstalled VS Code
3. Reinstalled Platformio
4. Restarted PC after each step
5. Manually search for the missing file

The issue: 
`c:/users/dirtb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file stm32f407xg.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_SKR_PRO\firmware.elf] Error 1`

I followed the error and tried reinstalling platformio as it says the file should be in .platformio - I tried this step 3 different times with no success. 

If someone can please let me know how to fix this compile issue I would really appreciate it. It says i'm missing the file or the directory is wrong, but I cannot find the file on my PC or how a way to download it. 
As you can see by the error, I have a BIGTREE_SKR_PRO mainboard, idk if that information is important but.. yeh.

ellensp, 2020-01-04:
no config files....   so can only make educated guesses.  My guess is this is the windows max command length issue again...   go into the directory Marlin/src/HAL/ and delete the directories for the hardware you don't have. You need to keep the "HAL_STM32_F4_F7", "HAL_STM32", "shared" directories and the two.h files

LinoBarreca, 2020-01-04:
@ThatOneGingie looks like there are some problems on linking. Since we know it's a recent problem (happened on other boards but not on skr-pro) we need to know
- which branch/commit you pulled (and you are trying to build)
- if you modified any file and how (make a zip of the modified files only and upload it here)

TheGadge, 2020-01-04:
Did you solve this?
The most common fix that works that I've offered on FB is move the whole folder closer to the root of C:\ etc and right click on it and remove 'read only' on it, and all within.

Ausildis, 2020-01-04:
It didn't for me.
I tried 2.0.x, downloaded 2.0.1 from the website, the only thing I changed were "default_envs = BIGTREE_SKR_PRO" in platformio.ini and "#define SERIAL_PORT -1", "#define SERIAL_PORT_2 1" and "#define MOTHERBOARD BOARD_BIGTREE_SKR_PRO_V1_1" in configuration.h. 

Heck I even tried the bigtreetech fork for this exact board (https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware/Marlin-bugfix-2.0.x%20SKR%20ProV1.1-Dmeo - with no changes) and I still get the same error.

Building for megaatmega2560 works just fine.

Crashdown1977, 2020-01-04:
Same problem here by me. Closer to c:\ dosent work for me.

"default_envs = BIGTREE_SKR_PRO"
"#define MOTHERBOARD BOARD_BIGTREE_SKR_PRO_V1_1"

c:/users/timo/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file stm32f407xg.ld: No such file or directory

What is wrong??
Thanks

goldensgui, 2020-01-04:
Hello there.
I've received my SKR Pro 1.1 and tried to build Marlin but I got the same problem as described here (word for word, see below) when compiling from 2.0.x branch. I also tried to move the folder closer to C: without success. Where is that linker script file supposed to be found?

c:/users/hiro-dono/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file stm32f407xg.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\BIGTREE_SKR_PRO\firmware.elf] Error 1

ThatOneGingie, 2020-01-04:
> Hello there.
> I've received my SKR Pro 1.1 and tried to build Marlin but I got the same problem as described here (word for word, see below) when compiling from 2.0.x branch. I also tried to move the folder closer to C: without success. Where is that linker script file supposed to be found?
> 
> Thanks
> 
> c:/users/hiro-dono/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file stm32f407xg.ld: No such file or directory
> collect2.exe: error: ld returned 1 exit status
> *** [.pioenvs\BIGTREE_SKR_PRO\firmware.elf] Error 1

Hey buddy i found a fix for this issue.
To fix it do the following
https://github.com/MarlinFirmware/Marlin/commit/04eab3821e6d95fba81970c02b7ba4e7e32bfd9b
follow the instructions on that github. its a different error but fixes the compile issue.

Crashdown1977, 2020-01-04:
This work`s for me

Thank you

goldensgui, 2020-01-04:
> 
> 
> > Hello there.
> > I've received my SKR Pro 1.1 and tried to build Marlin but I got the same problem as described here (word for word, see below) when compiling from 2.0.x branch. I also tried to move the folder closer to C: without success. Where is that linker script file supposed to be found?
> > Thanks
> > c:/users/hiro-dono/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file stm32f407xg.ld: No such file or directory
> > collect2.exe: error: ld returned 1 exit status
> > *** [.pioenvs\BIGTREE_SKR_PRO\firmware.elf] Error 1
> 
> Hey buddy i found a fix for this issue.
> To fix it do the following
> [04eab38](https://github.com/MarlinFirmware/Marlin/commit/04eab3821e6d95fba81970c02b7ba4e7e32bfd9b)
> follow the instructions on that github. its a different error but fixes the compile issue.

That's fixed the issue. Thank you very much. But what is the exact problem??? Will that modification have an impact later?

Ausildis, 2020-01-04:
> Hey buddy i found a fix for this issue.
> To fix it do the following
> [04eab38](https://github.com/MarlinFirmware/Marlin/commit/04eab3821e6d95fba81970c02b7ba4e7e32bfd9b)
> follow the instructions on that [...]

That fixed it. I guess that package changed and now it doesn't integrate fully with marlin for our board so until it's fixed we have to use the older package.

boelle, 2020-01-08:
closing since fixed by @thinkyhead

closed 2020-01-08 by boelle