MarlinFirmware/Marlin #16455: stm32f103vet Arduino ide build error

diver197, 2020-01-04:
I used stm32f103vet and installed STM32duino. It seems that I couldn't find the relevant board or chip to compile. Though I tried to add it to the hardware folder directly, it always ended up with errors .

ellensp, 2020-01-04:
no error logs, no config files,   This look like a Arduino support request not a marlin bug or feature request.   Please try one of the following resources.  

-RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
-[Tom's 3D Forums](https://forum.toms3d.org/)
-Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
-Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
-[Marlin Configuration ](https://www.youtube.com/results?search_query=marlin+configuration)on YouTube
-Marlin Discord server. Join link: [https://discord.gg/n5NJ59y](https://discord.gg/n5NJ59y)

boelle, 2020-01-04:
issue template not read - closing

closed 2020-01-04 by boelle