MarlinFirmware/Marlin #16456: [BUG] Autoleveling stops after 6 probes - Ender 3 + TouchMi

kramer04, 2020-01-04:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4021815/Configuration.zip)
### Bug Description
When the auto leveling starts the probe goes further than the plateau then starts the probe and stops after 6 probes. 

I expected 36

### My Configurations
Ender 3 + TouchMi + SKR-MINI-E3 V1.2

### Steps to Reproduce
launch auto level or a print

**Expected behavior:** [What you expect to happen]
I expect 36 probes
**Actual behavior:** [What actually happens]
probe only 6
#### Additional Information
using AUTO_BED_LEVELING_BILINEAR
#define GRID_MAX_POINTS_X 6
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

kramer04, 2020-01-05:
With M503
G21    ; Units in mm (mm)
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75
M200 D0
Steps per unit:
M92 X80.00 Y80.00 Z400.00 E412.00
Maximum feedrates (units/s):
M203 X500.00 Y500.00 Z5.00 E25.00
Maximum Acceleration (units/s2):
M201 X500.00 Y500.00 Z100.00 E5000.00
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P500.00 R500.00 T500.00
Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
M205 B20000.00 S0.00 T0.00 J0.08
Home offset:
M206 X0.00 Y0.00 Z0.00
Auto Bed Leveling:
M420 S0 Z0.00
G29 W I0 J0 Z-1.98250
G29 W I1 J0 Z-1.27250
G29 W I2 J0 Z-0.55000
G29 W I3 J0 Z0.21000
G29 W I4 J0 Z0.93000
G29 W I5 J0 Z1.64750
G29 W I0 J1 Z0.00000
G29 W I1 J1 Z0.00000
G29 W I2 J1 Z0.00000
G29 W I3 J1 Z0.00000
G29 W I4 J1 Z0.00000
G29 W I5 J1 Z0.00000
G29 W I0 J2 Z0.00000
G29 W I1 J2 Z0.00000
G29 W I2 J2 Z0.00000
G29 W I3 J2 Z0.00000
G29 W I4 J2 Z0.00000
G29 W I5 J2 Z0.00000
G29 W I0 J3 Z0.00000
G29 W I1 J3 Z0.00000
G29 W I2 J3 Z0.00000
G29 W I3 J3 Z0.00000
G29 W I4 J3 Z0.00000
G29 W I5 J3 Z0.00000
G29 W I0 J4 Z0.00000
G29 W I1 J4 Z0.00000
G29 W I2 J4 Z0.00000
G29 W I3 J4 Z0.00000
G29 W I4 J4 Z0.00000
G29 W I5 J4 Z0.00000
G29 W I0 J5 Z0.00000
G29 W I1 J5 Z0.00000
G29 W I2 J5 Z0.00000
G29 W I3 J5 Z0.00000
G29 W I4 J5 Z0.00000
G29 W I5 J5 Z0.00000
Material heatup parameters:
M145 S0 H190 B60 F0
M145 S1 H240 B110 F0
PID settings:
M301 P25.27 I2.01 D79.52
Power-Loss Recovery:
M413 S1
Z-Probe Offset (mm):
M851 X-50.00 Y-1.00 Z0.00
Stepper driver current:
M906 X580 Y580 Z580
M906 T0 E650
Driver stepping mode:
M569 S1 X Y Z
M569 S1 T0 E
Linear Advance:
M900 K0.15

reloxx13, 2020-01-09:
there where a few bugfixes last days for this, try updating fw to 2.0.x-bugfix and retest

kramer04, 2020-01-09:
> there where a few bugfixes last days for this, try updating fw to 2.0.x-bugfix and retest

tried the last marlin bug fix 2.0.x but same issue

robwlakes, 2020-01-11:
I have a proprietary built chassis I have stripped and rebuilding with Marlin2.0.x BugFix (RAMPS etc) and have been having similar trouble using the UBL routines. I have a grid of 5x5 and when Issue a G29 P1, it goes off and does locations 0 to 20, but when it gets to the 21st location this happens:

........
Recv: Probing mesh point 20/25.
Recv:
Recv: T:190.06 /190.00 B:50.04 /50.00 @:64 B@:33
Recv: echo:busy: processing
.......

Recv: T:190.00 /190.00 B:50.07 /50.00 @:64 B@:35
Recv: echo:busy: processing
Recv:
Recv: Probing mesh point 21/25.
Recv:
Recv: T:189.97 /190.00 B:50.01 /50.00 @:64 B@:44
Recv: echo:busy: processing
Recv: X:95.00 Y:70.00 Z:3.99 E:0.00 Count X:12683 Y:9345 Z:12752
Recv: echo:Unknown command: ""
Recv: ok
And moves to X5 (the last RHS column) stalls here with Unknown command:"" error?

Send: G29 P3
Recv: ok
If I issue the G29 P3 command it fills in the missing 21-25 locations., so the mesh must be Ok and not corrupted. The machine (was a CEL RObox) uses the nozzle as a probe, and tilts to activate the Z-Stop. So the boundaries of the probing are definitely inside the printing boundaries.
Is this the same problem as above or a new problem?

robwlakes, 2020-01-11:
I have tried it on the "dev-2.1.x" and I get the same result. I just copied the Configuration and Configuration_adv across and compiled the dev version. Could it be in the config files?
I am new to bug reporting this way so reluctant to open a new report unless necessary, the above problem looked similar to mine.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4048871/configs.zip)

kramer04, 2020-01-14:
With last Marlin bugfix-2.0.x (011420) when i launch G29 it stops at 18th probes and not 6th. It's better ;)
I expect 36
In pronterface i've always
G29
Error:Probing Failed
Error:Probing Failed

MagicSteve, 2020-01-17:
I think I got similar problems with my printer. I'm using an Anet AM8 with the original melzi Board. I'm using an Tronxy XY-08N NPN autolevel sensor. The sensor is wired correct working like a charm in Marlin 1.1.9 using the bilinear autoleveling.
After update to version 2.0.x the G29 doesn't behave like expected. Same problems as above doing a G29 Error:Probing Failed
The G28 works correct everytime.

The weird thing is that sometimes I got until the 6th probing point then it stops but other times it stops right after the first probing point. It seems that it also has to do with the number of probes. If I set MULTIPLE_PROBING = 2 it stops right after the first probing point. If I disable the second probing it sometimes reaches the 6th point and then it stops.
I tested it again yesterday with the newest bugfix Version 2.0.x but same problem.
I don't know what's wrong with G29 in marlin 2.0.x but my printer is not usable with this so if this is not getting fixed soon I have to revert back to the old but working version 1.1.9.

Attached are my configuration files maybe they help in finding the problem.

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4075559/configuration.zip)

kramer04, 2020-01-17:
I was at 20mm but from 40mm it works again. so define MIN_PROBE_EDGE 40.
But I maintain that the G29 is still buggy. This is not normal operation anyway.

MagicSteve, 2020-01-17:
@kramer04  Thanks for the information.
I will check with 40mm if it gets better for me too.
But 40mm is quite a lot so the leveling would be not as accurate as I want it to.

I agree this is not normal operation for the G29 because in 1.1.9 everything works as expected. So there must be something different.

I can‘t understand why they release Version 2.0.1 for stable with such problematic autoleveling...

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

sjasonsmith, 2020-01-17:
@kramer04, can you try adjusting Z_PROBE_LOW_POINT to allow the probe to travel further?

From your debug prints, it appears that the bed is too far away, causing the probe to fail. 

If adjusting the low point fixes the issue, either your bed actually is too far away, or there is a bug in the way Z is tracked as probing continues.

You should certainly not need to increase MIN_PROBE_EDGE for probing to work. That probably only seems to help because the smaller probing mesh has less Z variation.

kramer04, 2020-01-17:
> @kramer04, can you try adjusting Z_PROBE_LOW_POINT to allow the probe to travel further?
>
> From your debug prints, it appears that the bed is too far away, causing the probe to fail.
>
> If adjusting [...]

ok , i'm downloading last Marlin bugfix-2.0.x, redo an adjustment of the plate and the probe then report here

MagicSteve, 2020-01-17:
@sjasonsmith 
I tried your suggestion:

Downloaded the last changes in bugfix-2.0.x.
My Z_PROBE_LOW_POINT was set to -2 I tried it with -3.
Yes it's going further down now after the sensor triggers (so far that my nozzle is almost crashing the bed) but it doesn't change anything in behaviour. First probing point and then it fails for me.

sjasonsmith, 2020-01-17:
@MagicSteve that sounds like a different issue. That sounds like your probe is not working at all. My suggestion was specifically based on the logs provided by @kramer04, where the probe works fine and it consistently fails as it moves across the bed.

MagicSteve, 2020-01-17:
Why do you think it's not working at all when everything was ok with marlin 1.1.9? Can you explain that? Not working at all means to me that the nozzle would crash always into the bed but thats not the point. 

As I said a few posts before:
Doing G28 it always works perfect. That probe is correct wired, the light is flashing when it's getting triggered. When it's triggered the Z axis is moving up correctly but probing fails with G29 Error:Probing Failed

MagicSteve, 2020-01-17:
But if you think it makes more sense and my problem has nothing to do with this bug then I would open a new issue?

sjasonsmith, 2020-01-17:
It might be related, then. My concern is that you say it made it almost crash the nozzle into the bed. Was that definitely further than the point it should have triggered at? How far is the nozzle from the bed when it trigger during homing?

My assumption in my original suggestion was that the probe remained too far from the bed to trigger, not that a trigger was ignored. Extending the move distance if it is ignoring probe triggers could certainly crash the nozzle.

kramer04, 2020-01-17:
@sjasonsmith 
I've redo an adjustment of the plate and the probe and changed Z_PROBE_LOW_POINT to -3
Now it works. tried also Z_PROBE_LOW_POINT to -2 works too.
But i've another bug. When i'm launching a print my nozzle is highest about 0.5/0.65 mm.
I've to tune z-offset to adjust distance between nozzle and plate.
For example i set z-offset to -4,67; but have to offset manually to -5.27 when printing. Otherwise nozzle is too high.
I can open a new issue if you think it's a new problem.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4077852/Configuration.zip)

tatusah, 2020-01-17:
Might be pure geometrics. I mean that if the bed is as tilted as the logs say the slope on the bed causes the difference you are seeing in the offset number. So with the 50 mm offset if you just lower the nozzle to z0 after homing the place where the nozzle is can touch the bed earlier than the position where the actual homing probe was done. So unless you moved the nozzle to the position where the probe touched the bed while homing and measured the offset at that position then this can explain the difference. Even with automatic bed leveling it makes sense to have the bed leveled manually accurately enough.

MagicSteve, 2020-01-17:
@sjasonsmith 
When doing the G29 the sensor triggers and the Z-Axis is going down a bit deeper until it stops. With Z_PROBE_LOW_POINT set to -3 my nozzle tip is just on the bed. This should be the 3mm because I placed my sensor 3mm higher then the tip of the nozzle. So with -3 it’s a bit to deep -2 or -1 is better but changing that value seems not effect the probing error.

And yes this only happens with G29. When doing the G28 it seems that Z_PROBE_LOW_POINT doesn’t have an effect?
To give you precise infos I will test again and try to do some measurements.
Please tell me what I can do to help you finding the problem. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

robwlakes, 2020-01-18:
I have read the above and changed my Z_PROBE_LOW_POINT from -2 to -4. The 5x5 probing still stops at 21/25points and reports 'Recv: echo:Unknown command: "" ' (see above) leaving the last 5 points not probed. I can fill them in with G29 P3 (copies the 16-20 column to the 21-25column).

```
Send: G29 T0
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,139)                      (189,139)
Recv:         0       1       2       3       4
Recv:  4 |   .     -0.212  -0.121  -0.116  -0.536
Recv:    |
Recv:  3 |   .     -0.130  -0.045  +0.018  -0.283
Recv:    |
Recv:  2 |[  .   ] -0.074  +0.002  +0.067  -0.281
Recv:    |
Recv:  1 |   .     -0.074  -0.040  +0.002  -0.294
Recv:    |
Recv:  0 |   .     -0.099  -0.072  -0.089  -0.400
Recv:         0       1       2       3       4
Recv:     (  1,  1)                      (189,  1)
Recv: 
Recv: ok
```

I also enabled M48 to test the accuracy of the nozzle-probe. I ran it through a 3x3 matrix to simplify things and the tilting head/nozzle seemed quite an accurate probe.

```
Range of Probing Measurements (210x150 bed)

`Col1              Col2              Col3

[10,10]           [100,10]          [190,10]
Mean: 0.133962    Mean: 0.161769    Mean: -0.101694
Range 0.006       Range 0.003       Range 0.032

[10,70]           [100,70]          [190,70]
Mean: -0.115031   Mean: 0.003769    Mean: 0.193525 Min
Range 0.007       Range 0.004       Range 0.004

[10,140]          [100,40]          [190,140]
Mean: -0.079463   Mean: 0.176487    Mean: -0.528119
Range 0.006       Range 0.005       Range 0.038
`
So I don't think it is any spurious values being returned by the probe that is crashing the Marlin2.
Are these observations in the right place?

MagicSteve, 2020-01-18:
Installed todays freshest bugfix version and tested again:

Doing G28 the probe triggers, the axis instantly moves up, doing the second probe (sensor triggers again) and moves up. So everything ok.
Because the axis instantly moves up after triggering I guess Z_PROBE_LOW_POINT has no effect for G28.

G29 with Z_PROBE_LOW_POINT set to -2:
Sensor goes to first probing point moves down until it triggers then moves 2mm deeper than trigger point and then moves up => Probe failed.

Then I tested with Z_PROBE_LOW_POINT set to 0:
The same as G28 moves until triggers and instantly moves up again but the same result: Probe failed.

The interesting thing this time (maybe because of todays bugfix changes):
With G28 the light of the probe sensor flashes when it gets triggered. Doing the G29 it triggers but the light is not flashing.

Strange things are happening....

tpruvot, 2020-01-18:
if it stops at 21/25.. i guess a whole line is skipped... adjust the mesh inset in config for this side... generally right one if the probe is on the left of the nozzle

robwlakes, 2020-01-19:
tpruvot: I think you are right, it just doesn't do the last line, which I have seen it do in V1.9.x, so the boundaries are OK. Whatever it should not exit on "Uknown Command", and it does not reveal what that command was or where it came from. Also my head nozzle is the probe so there is no point on the build board it can't access safely, but I have declared a safety 5mm no go zone at the edges anyway.
MagicSteve: Z_PROBE_LOW_POINT set to -2 means that is far as Marlin will allow the head to go in search of a Z-Stop trigger. In other words a slip of the thumb on the LCD knob can't send your print head crashing through to your build bed any further than -2mm. So when it says the probe has failed it means the Z-Stop trigger has not arrived in time, ie before the -2mmZ is reached.

If it is set 0 then it will always fail if the Z-Stop trigger occurs somewhere between 0 and -2mm. Ideally the head should not read any height less than or equal to Z_PROBE_LOW_POINT at anytime. This is a Marlin safety feature. So  Z_PROBE_LOW_POINT should always be slightly more at least than the expected trigger point to trap any "runaway" Z movements.

The discrepancy between G28 working and G29 not working, is that G28 assumes the Z-Offset is set and stable (my Z-Offset is -0.18mm and when I do a G28 it calculates the Z height using the offset, because when I use the LCD to adjust the height it already "knows" where 0mm is) , whereas (I think) the G29 is actually probing and probably does not use the Z-Offset initially, as indirectly that is one of the things it is trying to determine. Someone who knows that part of the software maybe able to help out here. With UBL one can set the Z-Offset afterwards.eg M851 Z1.7 where Z is height of 0mm above the trigger point.

MagicSteve, 2020-01-19:
Thanks for all your tips and information. I tried different things and also went back to Marlin 1.1.9 but there I had suddenly the same problem. Then went back again to 2.0.x and a complete bed leveling  
was done. The next leveling after that failed again on first probing point. Was so confused and thought that it only can be a hardware problem. Then I tested the sensor with M119 and find out that the sensor triggers depending on the cable position! So it looks like the cable going into the sensor housing is broken somewhere.
So my problem is hardware related and not software. 
I will buy a new sensor and try again after that.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

boelle, 2020-01-19:
@kramer04 where you able to solve this one?
if you got another issues then maybe best to create a new issue for that?

kramer04, 2020-01-20:
Thanks to all
After many tests I redid all the adjustment of my printer and I kept Z_PROBE_LOW_POINT to -3 not -2
Now it works
I close this issue

closed 2020-01-20 by kramer04