MarlinFirmware/Marlin #16457: Extruder debug info displaying in Repetier v2.1.6 and Marlin 2.0.1 on GT2560 rev A Plus

e415ldc, 2020-01-04:
I have seen this issue in version 1.1.9 as well. I had hoped it was resolved in 2.0.1. Issuing a M111 S0 does not affect the issue. I have seen nothing close to this issue posted here or in the community areas. I can post log files of the issue if requested.

boelle, 2020-01-04:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

e415ldc, 2020-01-05:
[i3-2020-3dTouch.zip](https://github.com/MarlinFirmware/Marlin/files/4022764/i3-2020-3dTouch.zip)

e415ldc, 2020-01-05:
Sorry for the missing files. I was interrupted while posting. I figured i'd wait to see if I had actually posted prior to amending. I've included the requested files and a copy of the text from the Repetier session that includes the commands and responses from the printer.  I am using Arduino 1.8.10 IDE on a W7 system.
I executed a few 1mm extrudes and a retraction as well. Everything works great physically. These messages are just a large distraction.

boelle, 2020-01-08:
btw, does this also happen on bugfix 2.0.x ?

e415ldc, 2020-01-09:
I have not attempted the bugfix version. I can attempt that as soon as I have my system put back together. Maybe later tomorrow (01/09/2020)

e415ldc, 2020-01-09:
I downloaded the BugFix version just after responding last night. I have it compiled and loaded onto one of my printers. There is no change in the extra info being sent during an extruder operation. But, I now see the bead leveling seems to be different. The previous version of the code moved the print head assembly further towards the middle of the bed (appearing to use more than 10mm as the distance from edge parameter). Are there any files you would like to be attached?

boelle, 2020-01-18:
nope, but did it get worse? or is it a completely new issue?
maybe a video could help tell the smart people (not me) what is going on

did you get marlin from here? https://github.com/MarlinFirmware/Marlin/
asking beacuse you included in the zip more than is normal

e415ldc, 2020-01-19:
No change at all between the GA and BF versions in respect to the (what appears to me) to be extruder debug info. There are couple of other minor differences that are consistent with a bugfix release. The 3d /BLtouch bed leveling is a little different, but I still see an anomaly in the calculations it is doing for defining the edge of the bed and the probes x-offset

The link I used was from the Marlin site directly for both. Technically not from Github and yet from Github. Does that make sense?

The only "extra files I included would have been the logfile (.TXT) from Repetier and the .ino file the Arduio IDE creates. The version.h file was already there from the download.

Vertabreak, 2020-01-19:
not sure i see a problem something executes M111 enables logging then changes the log level then shuts it off and if i had to guess repetier is the one doing it unless you the one sending the commands by hand. http://marlinfw.org/docs/gcode/M111.html

e415ldc, 2020-01-19:
FYI, Repetier defaults to "Errror and Info" on. I then execute the M111 S0 as stated in the original issue report. The messages that seem to be related to the extruder actions keep coming no matter the debug settings via M111.

e415ldc, 2020-01-19:
FYI, using v1.0.x of Marlin via the Geeetech site links there are no messages that cannot be stopped with M111 S0.

boelle, 2020-01-23:
have you tried with bugfix 2.0.x from here: https://github.com/MarlinFirmware/Marlin/

e415ldc, 2020-01-23:
I just downloaded the .zip from your most recent link. It IS a different file size than the one I pulled on the 8th. I will try this new? one today and post ASAP. Thank you.

e415ldc, 2020-01-23:
I recompiled with the code at the link you provided. (using my existing configuration.h and configuration.adv.h files) There appears to be no change. I am still getting the lines of what appears to be debug info every time the extruder engages or retracts. Because I use "wiping" there is a very large number of these sequences being thrown into the log area of Repetier. This makes monitoring a print job very difficult.

I am now also getting "facet" and "vertex" messages that appear to be coming from Slic3r that was not seeing before. i am attaching a sample of this info captured in the Repetier log.
[Sample of log.zip](https://github.com/MarlinFirmware/Marlin/files/4105254/Sample.of.log.zip)

boelle, 2020-01-23:
you cant always reuse config files

you need to use a tool like winmerge and compare new and old and then copy over your settings

config files are also updated from time to time hence you cant always reuse them

e415ldc, 2020-01-24:
Understood. I'll do over. Manually setting the unique parts. I do use the sample for the GT2650 set of configs though.

Also, WINMerge and i have not gotten along very well in the past. it had wiped out a week's worth of work I had done. The ONE time I didn't have a copy of my work saved off system.

I'll try to get everything redone by tomorrow evening and post the results.

e415ldc, 2020-01-24:
An error in my last post...

The config files I am starting with are from the ZIP -- \Marlin-2.0.x\config\examples\Geeetech\Prusa i3 Pro C

Since my printers are all based off the Geeetech i3 Pro C and use the GT2560 board.

e415ldc, 2020-01-24:
I have rebuilt from scratch using the sample files in the ZIP as a starting point. Minimal changes -- number of extruders, board type, bed size, jerk, bltouch, acceleration. No change to the extra stuff being sent back to Repetier during any extruder function.  I think I have found the code that is generating the lines in the Motion.cpp file though.  I just don't know what is calling it. I will attempt to comment out the lines related to this and see what happens. I'm sure I will be breaking a valid debug routine, but at least I'll know what is generating my lines. Would you like a new set of files?

Vertabreak, 2020-01-25:
Might help us better understand.

e415ldc, 2020-01-25:
I commented out the following two lines in motion.cpp and everything looks normal again. As I said I'm sure this broke something else, but I don't have a way to figure out what called these lines. Or is there a conditional missing in front of them?

SERIAL_ECHOPAIR("X:", lpos.x, " Y:", lpos.y, " Z:", lpos.z, " E:", current_position.e);

  stepper.report_positions();

e415ldc, 2020-01-25:
[My_Marlin_GH-BF-2.0.1.zip](https://github.com/MarlinFirmware/Marlin/files/4111489/My_Marlin_GH-BF-2.0.1.zip)

e415ldc, 2020-01-25:
I just uploaded the config and .INO files I am currently using.

Vertabreak, 2020-01-28:
yea i looked over them i don't see anything that jumps out as unusual. 
i think that's the slicer doing that to marlin have you tried running it from another slicer or maybe octoprint. i personally own several variants of the GT2560 board and none of them do this with cura or octoprint.

e415ldc, 2020-01-28:
I have not tried any other slicer or print programs. But the issue is completely gone when I comment out those two lines. Something is causing the Serial_Echo line to be executed in version of Marlin 1.9.x and above with my support software. This did not happen with version 1.0.x of the Marlin code provided by Geeetech. Geeetech has told me they do not support anything newer than 1.0.x on the GT2560 boards. I'm just trying to stay current and get some of the newer features for my 3d Touch and TMC 2208 stepper drivers.

Vertabreak, 2020-01-28:
Would suggest trying a slicer that's been updated more resently then 5 years ago way before marlin 2 was released. How could they anticipate changes that happen years later. I'm geeetech forum admin vert by the way.

e415ldc, 2020-01-28:
I don't see how the Slicer has any affect on the issue I've reported. I get he messages I'm reporting on while there is NO object even in Repetier. All I have to do is heat the extruder to above my minimum of 175 and tell Repetier to extrude or retract via the UI. There is no Slicer involved in any way. I did report some other messages that may have been in error on my part. I think I may have done a slice while I was testing. I do see Slic3r messages in my Repetier logs all the time. This is not that. 

The source of the messages i'm getting is definitely the SERIAL_ECHO line in Motion.cpp. Once it is commented, I do not get any more messages that I should not be getting. I do not have the skills/knowledge to determine what is calling that line of code, so I have opened this "case" for the community to help resolve.

boelle, 2020-02-08:
@e415ldc have you retested with bugfix 2.0.x ?

e415ldc, 2020-02-08:
I'll pull another one today and test. The previous one I had pulled still has the issue. It really appears to be something is causing those two lines in motion.cpp to execute.

e415ldc, 2020-02-08:
I pulled the current BF zip this morning. I then configured and compiled to the Arduino. There is no change. I still get the extra lines posted to the Repetier log.
[My_i3_BF_201_200208.zip](https://github.com/MarlinFirmware/Marlin/files/4174648/My_i3_BF_201_200208.zip)

e415ldc, 2020-02-08:
Should I try the new 2.0.3?

Vertabreak, 2020-02-08:
bugfix is always newer then stable as its made from bugfix.

e415ldc, 2020-02-08:
Wouldn't 2.0.3 be all the fixes (to that point in time) consolidated into a GA style release? Like normal (for profit) software releases.

boelle, 2020-02-09:
@e415ldc no, bugfix 2.0.x is updated each day and the release branch's/tags are just a frozen copy of bugfix 2.0.x so they will never change

e415ldc, 2020-02-22:
I pulled the latest bugfix as of 200220 and tried it. No change. I'm attaching the Configuration.h, Configuration-adv.h and thew other files in the ino directory as well as a txt file of the Repetier log file in a zip file for review.
[BugFix-200220.zip](https://github.com/MarlinFirmware/Marlin/files/4238810/BugFix-200220.zip)

e415ldc, 2020-02-22:
I will attempt to do this biweekly as due diligence for the process of resolving this.

boelle, 2020-06-20:
@e415ldc is the issue gone?

e415ldc, 2020-06-21:
I still see the issue with the latest BF. Since Version 2.0.5, my
modification to motion.cpp no longer keeps these messages from
displaying. The motion.cpp is completely different between 2.0.1 and
2.0.5 and later.
I'll keep trying the newer versions they are available, but I'm using
2.0.1 on all three of my GT2560 A based printers while actually
printing to keep things cleaner on the Repetier console.
Thank you for checking. Its been so long with no responses (other than
sarcasm from others on the use of Slicer and Repetier) I gave up.
e415ldc (Larry)

	-----------------------------------------From: "Bo Herrmannsen" 
To: "MarlinFirmware/Marlin"
Cc: "e415ldc", "Mention"
Sent: Saturday June 20 2020 4:09:27PM
Subject: Re: [MarlinFirmware/Marlin] [BUG] Extruder debug info
displaying in Repetier v2.1.6 and Marlin 2.0.1 on GT2560 rev A Plus
(#16457)

	@e415ldc [1] is the issue gone? 

	—
[3]. 

Links:
------
[1] https://github.com/e415ldc
[2]
https://github.com/MarlinFirmware/Marlin/issues/16457#issuecomment-647045672
[3]

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

e415ldc, 2020-06-23:
I pulled the current version today and tried it. I have the same
result (the extra location information for every extruder movement
command.

I'll upload the Marlin code and the log file from Repetier.
e415ldc

	-----------------------------------------From: "Bo Herrmannsen" 
To: "MarlinFirmware/Marlin"
Cc: "e415ldc", "Mention"
Sent: Tuesday June 23 2020 7:11:15AM
Subject: Re: [MarlinFirmware/Marlin] [BUG] Extruder debug info
displaying in Repetier v2.1.6 and Marlin 2.0.1 on GT2560 rev A Plus
(#16457)

	Please test the bugfix-2.0.x branch [1] to see where it stands. 

	—
[3]. 

Links:
------
[1] https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x
[2]
https://github.com/MarlinFirmware/Marlin/issues/16457#issuecomment-648104716
[3]

e415ldc, 2020-06-23:
[BF-200623-i3ProB.zip](https://github.com/MarlinFirmware/Marlin/files/4822485/BF-200623-i3ProB.zip)

Let me know if I've forgotten something.
e415ldc

thinkyhead, 2020-06-24:
I looked at your log for a second and…
Your host or G-code keeps sending G92 and that is why you see this. It is not a bug in Marlin.

closed 2020-06-24 by thinkyhead