MarlinFirmware/Marlin #16458: Mesh Bed Leveling not working after changes

afaulkner420, 2020-01-04:
My machine has been running lovely fornearly a year now,without the need for any changes at all, been using MBL very successfully.
Now ive made some changes, this seems not the case. As soon as i click on "level bed" through the bed leveling menu, it homes the machine, and then just jumps straight back to the main info screen.

Am i missing something? Is this a bug? it seems to only be present when using "BOARD_RAMPS_14_EEB"

Its ramps 1.4 controlled, and have been running the Marlin2 bugfix branch all  this time.

I decided to treat myself to a dual extruder. 

obviously this meant changes!

I downloaded the latest marlin 2 from themain repo, and implemented all config.h and config_adv.h changes, from previous version of marlin.

I then went through again, and change the motherboard type to BOARD_RAMPS_14_EEB from BOARD_RAMPS_14_EFB, defined the 2 extruders, and re-commented the line #define Z_DUAL_STEPPER_DRIVERS as i cant have that with the dual extruder motors attached.

I also have an RRD fan extender board attached, with pin 6 defined for fan0.
i had to do this through the pins-RAMPS.h file by adding:

#undef FAN_PIN
#define FAN_PIN 6

Ive attached my configs below.

If this is too much "ramble" please say. i wanted to add the history, and what i did to try help.

[Non Working configs.zip](https://github.com/MarlinFirmware/Marlin/files/4022265/Non.Working.configs.zip)
[Old - Working configs.zip](https://github.com/MarlinFirmware/Marlin/files/4022266/Old.-.Working.configs.zip)

boelle, 2020-01-08:
does this also happen with bugfix 2.0.x ?

afaulkner420, 2020-01-09:
I'm not 100% certain on that! I need to test that! I'll be doing that this evening! Is there a reason you ask?

afaulkner420, 2020-01-09:
Yes it happens on bugfix 2.0 as well. I'm using a pre-12th November version. I haven't downloaded and modded the latest version though. Will there be anything different after 12thnovember that would have sorted it? I'll download the latest and test that?

boelle, 2020-01-10:
yes, 2.0.x is updated more often, so it could be that the bugwas fixed

afaulkner420, 2020-01-18:
@boelle I can now confirm that this bug is still present in the latest 2.0 bugfix as of yesterday... 17th jan

boelle, 2020-01-23:
also after PR #16602 was merged ?

afaulkner420, 2020-01-23:
I haven't tried as I was told it was a compiling fix. Not a runtime fix? I'll try tonight.

I'm also having an issue where both my hotend thermistors are showing the same value. If I hear ex0, both ex0 and ex1 are reading the same, both rising. And the same if I hear ex1. Both thermistors are reading the same on the LCD?? Could this be linked? Could there be some reason things aren't working with multiple extruders? 
This all seems to have happened since I've defined "extruders 2"

boelle, 2020-01-23:
i have no idea, and i'm not a coder or maintainer, just the "janitor" sweeping the floor and dusting off things

afaulkner420, 2020-01-25:
Ah on fair enough. Well as a side note. The mesh bed levelling wasn't fixed with the pr no. But it does work with marlin 1 bugfix with the same setup. It's just all variants of marlin 2 that don't work. Its almost as if there's a line of code somewhere missing, telling it to go onto the next part... As all it does is home the machine when I click level bed.

afaulkner420, 2020-01-25:
Oh and to add, the thermistors was a bad thermistor. My bad this end.

sjasonsmith, 2020-01-26:
I just reproduced it on my SKR 1.3 using a hybrid between my configs and yours. I'm just starting to debug to figure out what is going on.

sjasonsmith, 2020-01-26:
Apprently this is a duplicate of #13591. I am going to post my findings over there, since it is the earlier of the two issues.

afaulkner420, 2020-01-26:
Ah ok I'll jump on over there! May close this one if that's the case...

sjasonsmith, 2020-01-26:
> Ah ok I'll jump on over there! May close this one if that's the case...

I was just going to recommend you close it! Apparently there have been at least 4 issues opened about this, but nobody has ever looked into what was happening (until now!)

afaulkner420, 2020-01-26:
I even looked through months worths of issues looking!! Apparently I wasnt looking for the right wording! Haha! I'll close this and keep an eye over there! 

CLOSED DUE TO DUPLICATE ISSUE: MENTIONED ABOVE

closed 2020-01-26 by afaulkner420