MarlinFirmware/Marlin #16462: MKS Gen 1.4 with TMC2130 V1.2 Stepper Drivers

Muskateer10, 2020-01-04:
### Bug Description

Upgraded to Marlin 2.0.1 and I get a TMC2130 connection error only on X and Y axis's.
I have been using Marlin 1.1.9 Bugfix for several months and have observed few problems, other than some layer shifting. The Z and E0 drivers display connection information properly in all of the above Marlin Versions. 

### My Configurations
MKS_GEN V1.3
TMC2130 V1.2 Installed on X,Y,Z and E0.
VS Code
Ender 3 Pro

### Steps to Reproduce

1. Install either Marlin 2.0.1 or Marlin 2.0.x bugfix
2. Modify configuration.h and configuration_adv.h to enable TMC2130 on XYZ, E0
3. Enable TMC_Debug
4. Build and upload files to the MKS Gen 1.4 board through USB.
5. TMC connection error appears on Ender 3 screen.
6. Run M122 from pronterface. 
7: X, Y show connection error all registers are FF. Both Z and E0, display correctly as expected.
####
Expected Behavior:
Same as Marlin 1.1.9 Bugfix 
   XYZ, and E0 should connect properly and not show a connection error on the dispaly. I have printed many, many prints with the same hardware using Marlin 1.1.9 Bugfix.

Actual behavior:
 In Marlin 2.0.1 and Marlin 2.0.x bugfix. XY stepper drivers are failing to connect. All registers read FF. TMC2130 connection error appears on the stock display.

#### Additional Information
All versions of Marlin listed above display good connections on the Z axis and on E0.  The X, Y drivers connect only in Marlin 1.1.9 version. I have looked at the Ramps.h file and I could see no differences in pin assignments. 
I may be missing something obvious. But, I have been working on this for 2 days now.  If I downgrade the MKS_GEN 1.4 to Marlin version 1.1.9 everything appears to work properly.
[MKS_GEN1_3.zip](https://github.com/MarlinFirmware/Marlin/files/4022485/MKS_GEN1_3.zip)

Muskateer10, 2020-01-05:
Solved.

boelle, 2020-01-08:
closing since solved

closed 2020-01-08 by boelle