MarlinFirmware/Marlin #16463: [BUG] FIX Multiple definition of eeprom functions framework-arduino-lpc176x

MakeNVape, 2020-01-04:
POSSIBLE FIX PLEASE CONFIRM

Uncomment /. remove below from .platformio/packages/framework-arduino-lpc176x/cores/arduino/arduino.cpp

as it is duplicated in HAL/shared/eeprom_i2c.cpp

// **************************
// Persistent Config Storage
// **************************
/* MAKE N VAPE SKR V1.4 I2C EEPROM FIX
void eeprom_write_byte(uint8_t *pos, unsigned char value) { }

uint8_t eeprom_read_byte(uint8_t * pos) { return '\0'; }

void eeprom_read_block(void *__dst, const void *__src, size_t __n) { }

void eeprom_update_block(const void *__src, void *__dst, size_t __n) { }
*/

### Bug Description

#define I2C_EEPROM 

on LPC176X causes multiple definitions error

### Steps to Reproduce
using LPC176X chipset in Marlin 2

#define I2C_EEPROM 

**Expected behavior:** 
Compile Without Error

**Actual behavior:** 

.pio/build/LPC1768/libFrameworkArduino.a(arduino.cpp.o): In function `eeprom_write_byte(unsigned char*, unsigned char)':
.platformio/packages/framework-arduino-lpc176x/cores/arduino/arduino.cpp:111: multiple definition of `eeprom_write_byte(unsigned char*, unsigned char)'
.pio/build/LPC1768/src/src/HAL/shared/eeprom_i2c.cpp.o: Marlin/src/HAL/shared/eeprom_i2c.cpp:49: first defined here
.pio/build/LPC1768/libFrameworkArduino.a(arduino.cpp.o): In function `eeprom_read_byte(unsigned char*)':
.platformio/packages/framework-arduino-lpc176x/cores/arduino/arduino.cpp:113: multiple definition of `eeprom_read_byte(unsigned char*)'
.pio/build/LPC1768/src/src/HAL/shared/eeprom_i2c.cpp.o:Marlin/src/HAL/shared/eeprom_i2c.cpp:94: first defined here
.pio/build/LPC1768/libFrameworkArduino.a(arduino.cpp.o): In function `eeprom_read_block(void*, void const*, unsigned int)':
.platformio/packages/framework-arduino-lpc176x/cores/arduino/arduino.cpp:115: multiple definition of `eeprom_read_block(void*, void const*, unsigned int)'
.pio/build/LPC1768/src/src/HAL/shared/eeprom_i2c.cpp.o:Marlin/src/HAL/shared/eeprom_i2c.cpp:108: first defined here
.pio/build/LPC1768/libFrameworkArduino.a(arduino.cpp.o): In function `eeprom_update_block(void const*, void*, unsigned int)':
arduino.cpp:(.text._Z19eeprom_update_blockPKvPvj+0x0): multiple definition of `eeprom_update_block(void const*, void*, unsigned int)'
.pio/build/LPC1768/src/src/HAL/shared/eeprom_i2c.cpp.o:Marlin/src/HAL/shared/eeprom_i2c.cpp:67: first defined here

atoomnetmarc, 2020-01-05:
This issue looks familiar to this one: https://github.com/MarlinFirmware/Marlin/issues/15860

MakeNVape, 2020-01-05:
@atoomnetmarc yes but it wasn't resolved and still exists, hence the need for the fix. But thank you for pointing that one out, I had double checked that one first.

boelle, 2020-01-08:
if this is a fix it should be sent in as a PR to bugfix 2.0.x instead, else it will hang forever

MakeNVape, 2020-01-09:
@boelle thanks will do

boelle, 2020-01-14:
you made 2 PR's but both where based of wrong branch, you need to start out with bugfix 2.0.x

closed 2020-01-14 by boelle

PaschyS, 2020-04-16:
Marlin 2.0.5.3 The Error is still there with the skr 1.4 turbo.

thinkyhead, 2020-04-17:
@PaschyS — Unable to reproduce with `bugfix-2.0.x`. All builds are succeeding.

PaschyS, 2020-04-19:
OK, I will download the last bugfix and try it ;-)

laminarturbulent, 2020-04-21:
@PaschyS I reproduced the error as well with the SKR V1.4 Turbo after trying to define I2C_EEPROM in the Turbo pins file:
`
#pragma once

#define BOARD_INFO_NAME "BIGTREE SKR 1.4 TURBO"
#define SKR_HAS_LPC1769

//
// EEPROM
//
//#define FLASH_EEPROM_EMULATION
//#define SDCARD_EEPROM_EMULATION

#define I2C_EEPROM
#define E2END 0x7FFF //32 kilobytes

//
// Include SKR 1.4 pins
//
#include "../lpc1768/pins_BTT_SKR_V1_4.h"
`
I got it to compile successfully after doing @MakeNVape 's fix though, which is explained in more detail here:
https://www.makenvape.uk/3d-printing/3d-printing-guides/i2c-error-marlin-2/
Interestingly the article says that the problem is resolved past Marlin 2.0.5 but I still needed to comment that stuff out from arduino.cpp to get it to compile.
I don't actually have an I2C EEPROM module to test this with so all I know is that it compiled without errors.

PaschyS, 2020-04-21:
I  have just compiled the bugfix version from April 19, but without success, see:

> c:/users//.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1769\libFrameworkArduino.a(arduino.cpp.o): in function `eeprom_write_byte(unsigned char*, unsigned char)':
> C:\Users\\.platformio\packages\framework-arduino-lpc176x\cores\arduino/arduino.cpp:113: multiple definition of `eeprom_write_byte(unsigned char*, unsigned char)'; .pio/build/LPC1769/src/src/HAL/shared/eeprom_i2c.cpp.o:F:\3DPRINT\Marlin-bugfix-2.0.x_19.04.2020/Marlin\src\HAL\shared/eeprom_i2c.cpp:52: first defined here
> c:/users//.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1769\libFrameworkArduino.a(arduino.cpp.o): in function `eeprom_read_byte(unsigned char*)':
> C:\Users\\.platformio\packages\framework-arduino-lpc176x\cores\arduino/arduino.cpp:115: multiple definition of `eeprom_read_byte(unsigned char*)'; .pio/build/LPC1769/src/src/HAL/shared/eeprom_i2c.cpp.o:F:\3DPRINT\Marlin-bugfix-2.0.x_19.04.2020/Marlin\src\HAL\shared/eeprom_i2c.cpp:97: first defined here
> c:/users//.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1769\libFrameworkArduino.a(arduino.cpp.o): in function `eeprom_read_block(void*, void const*, unsigned int)':
> C:\Users\\.platformio\packages\framework-arduino-lpc176x\cores\arduino/arduino.cpp:117: multiple definition of `eeprom_read_block(void*, void const*, unsigned int)'; .pio/build/LPC1769/src/src/HAL/shared/eeprom_i2c.cpp.o:F:\3DPRINT\Marlin-bugfix-2.0.x_19.04.2020/Marlin\src\HAL\shared/eeprom_i2c.cpp:111: first defined here
> c:/users//.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1769\libFrameworkArduino.a(arduino.cpp.o): in function `eeprom_update_block(void const*, void*, unsigned int)':
arduino.cpp:(.text._Z19eeprom_update_blockPKvPvj+0x0): multiple definition of `eeprom_update_block(void const*, void*, unsigned int)'; .pio/build/LPC1769/src/src/HAL/shared/eeprom_i2c.cpp.o:F:\3DPRINT\Marlin-bugfix-2.0.x_19.04.2020/Marlin\src\HAL\shared/eeprom_i2c.cpp:70: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1

When I doing @MakeNVape 's fix then it will work! I2C EEPROM module works fine, but sometimes comes a EEPROM CRC failure, after reload factory settings is all perfect.
=========== 1 succeeded in 00:00:08.016 ===========

thinkyhead, 2020-04-22:
When you want to use an I2C-based EEPROM in Marlin, but you don't want to use the code that Marlin provides, just enable `EEPROM_SETTINGS` and don't enable `I2C_EEPROM`. Then Marlin will use the one supplied by the LPC176x Arduino framework.

randellhodges, 2020-04-22:
There are a few things at play here. The HAL forces FLASH_EEPROM_EMULATION:
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/LPC1768/eeprom_api.h (and every boards pin file also has a similar define in it)

And the framework just stubs the eeprom calls
https://github.com/p3p/pio-framework-arduino-lpc176x/blob/382459c61b0c0368358be1dc9e4f8b0558763fe2/cores/arduino/arduino.cpp#L111

I chatted with p3p a little about this. I might have misunderstood the direction he was suggesting, but I started something in the above reference PR. I tested it on an SKR 1.4 and I could switch between flash/i2c/sd card just by change my define in one of my config files. I don't have an SPI eeprom so I couldn't test that.

PaschyS, 2020-04-23:
Thanks @randellhodges for your work. I will try it in a few days, because the Skr 1.4 turbo is on the way to my friend. When I configure the firmware for him, then I will test it. 
I have here a Skr pro, maybe there is the same possibility?

randellhodges, 2020-04-23:
TLDR; SKR PRO 1.1 = ¯\_(ツ)_/¯

I know some people have said that they are having problems with flash eeprom emulation after 2.0.3 with the SKR PRO 1.1, specifically with BLTOUCH.

I was using an I2C eeprom with my SKR PRO 1.1 and had an issue where the BLTOUCH stopped working after a print was finished (no PWM signal, aka, no blue led on the BLTOUCH). This is different than what others reported.

People are saying the SKR PRO 1.1 is working for them with SD card based eeprom.

This code change will only apply to LPC based boards, like the SKR 1.4

thinkyhead, 2020-05-01:
Various changes have been made to EEPROM organization in the HALs. Please check out the recent commits and make sure that your EEPROMs are working as expected.