MarlinFirmware/Marlin #16465: [BUG] (Failing to Build - STM32F103RC_bigtree_512K board.cpp not found)

Dracrius, 2020-01-04:
### Bug Description

When trying to build I get a failure and the only error is 
```
"warning: Ignoring missing SConscript 'L:\3D Printing\Marlin Builds\Marlin 2.0.x - Ender 3 with SKR mini e3 & BLtouch\buildroot\share\PlatformIO\scripts\STM32F103RC_SKR_MINI.py'
File "C:\users\dracrius\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 159, in <module>
*** [L:\3D Printing\Marlin Builds\Marlin 2.0.x - Ender 3 with SKR mini e3 & BLtouch\.pio\build\STM32F103RC_bigtree_512K\FrameworkArduinoVariant\board.cpp.o] Source `L:\3D Printing\Marlin Builds\Marlin 2.0.x - Ender 3 with SKR mini e3 & BLtouch\.pio\build\STM32F103RC_bigtree_512K\FrameworkArduinoVariant\board.cpp' not found, needed by target `L:\3D Printing\Marlin Builds\Marlin 2.0.x - Ender 3 with SKR mini e3 & BLtouch\.pio\build\STM32F103RC_bigtree_512K\FrameworkArduinoVariant\board.cpp.o'.". 
```

If I switch to STM32F103RC_bigtree (256k) it fails with almost the same message minus the 512k part.

### Steps to Reproduce

1. change setting for a SKR mini e3 on an Ender 3 with a bltouch
2. Build
3. Failure message

**Expected behavior:** 
A successful build

**Actual behavior:** 
A failure I don't see a reason for as I followed 2 Teaching Tech guides and cross refrenced with other sources and I don't appear to have missed a step so I think something between 2.0 release and 2.0.1 broke skr mini building as all guides were done with 2.0 release

#### Additional Information

TERMINAL Output:
```
> Executing task: C:\Users\Dracrius\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RC_bigtree_512K (platform: ststm32; board: genericSTM32F103RC; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 6.0.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: framework-arduinoststm32-maple 1.10000.190819 (1.0.0), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), tool-stm32duino 1.0.2
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|-- <LiquidCrystal> 1.3.4
|-- <TMCStepper> 0.6.1
|   |-- <SoftwareSerialM>
|-- <Adafruit MAX31865 library> 1.0.3
|-- <LiquidTWI2>
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|-- <SlowSoftI2CMaster>
|-- <SoftwareSerialM>
Building in release mode

warning: Ignoring missing SConscript 'L:\3D Printing\Marlin Builds\Marlin 2.0.x - Ender 3 with SKR mini e3 & BLtouch\buildroot\share\PlatformIO\scripts\STM32F103RC_SKR_MINI.py'
File "C:\users\dracrius\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 159, in <module>
*** [L:\3D Printing\Marlin Builds\Marlin 2.0.x - Ender 3 with SKR mini e3 & BLtouch\.pio\build\STM32F103RC_bigtree_512K\FrameworkArduinoVariant\board.cpp.o] Source `L:\3D Printing\Marlin Builds\Marlin 2.0.x - Ender 3 with SKR mini e3 & BLtouch\.pio\build\STM32F103RC_bigtree_512K\FrameworkArduinoVariant\board.cpp' not found, needed by target `L:\3D Printing\Marlin Builds\Marlin 2.0.x - Ender 3 with SKR mini e3 & BLtouch\.pio\build\STM32F103RC_bigtree_512K\FrameworkArduinoVariant\board.cpp.o'.
===================================================================================== [FAILED] Took 5.12 seconds =====================================================================================

Environment                   Status    Duration
----------------------------  --------  ------------
megaatmega2560                IGNORED
megaatmega1280                IGNORED
rambo                         IGNORED
FYSETC_F6_13                  IGNORED
sanguino_atmega644p           IGNORED
sanguino_atmega1284p          IGNORED
melzi                         IGNORED
melzi_optiboot                IGNORED
at90usb1286_cdc               IGNORED
at90usb1286_dfu               IGNORED
DUE                           IGNORED
DUE_USB                       IGNORED
DUE_debug                     IGNORED
LPC1768                       IGNORED
LPC1769                       IGNORED
STM32F103RC                   IGNORED
STM32F103RC_fysetc            IGNORED
STM32F103RC_bigtree           IGNORED
STM32F103RC_bigtree_USB       IGNORED
STM32F103RC_bigtree_512K      FAILED    00:00:05.122
STM32F103RC_bigtree_512K_USB  IGNORED
STM32F103RE                   IGNORED
STM32F103RE_bigtree           IGNORED
STM32F103RE_bigtree_USB       IGNORED
STM32F4                       IGNORED
STM32F7                       IGNORED
ARMED                         IGNORED
STM32F103VE_longer            IGNORED
mks_robin                     IGNORED
mks_robin_pro                 IGNORED
mks_robin_lite                IGNORED
mks_robin_lite3               IGNORED
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
FLYF407ZG                     IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
rumba32_f446ve                IGNORED
mks_rumba32                   IGNORED
include_tree                  IGNORED
=============================================================================== 1 failed, 0 succeeded in 00:00:05.122 ===============================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

```

ellensp, 2020-01-05:
If you build for this motherboard (BOARD_BTT_SKR_MINI_E3_V1_2) but use the wrong platform (eg default_envs  megaatmega2560) It errors. You then switch to default_envs STM32F103RC_bigtree_512K without doing a platformIO clean first, you get this error.  So click "on platformio clean" (the trash can icon along the bottom of vscode) and try again.

Dracrius, 2020-01-05:
> If you build for this motherboard (BOARD_BTT_SKR_MINI_E3_V1_2) but use the wrong platform (eg default_envs megaatmega2560) It errors. You then switch to default_envs STM32F103RC_bigtree_512K without [...]

While I appreciate the suggestion it wasn't the solution I just gave it a try plus I'd actually never built with vscode and platformio ever before so the only thing I ever asked it to build was this board. My SKR mini E3 just showed up I was using Arduino IDE and 1.1.9 on my Original board.

Dracrius, 2020-01-05:
HAHA I fixed it. Ok I think this is a bug in PlatformIO as I couldn't find an example of it happening in VS code and it happens when you use its open a project button on a folder with a "&" in the title. It will duplicate the folder and create a .pio folder inside the duplicate. The strange things is that it can even do that normally windows will not allow two folders to have the same name. After Removing the "&" from the title it stopped duplicating the folder and complied to an error which is actually user error so I'm fixing that and building now. Should I post somewhere else to pass on the PlatformIO bug?

reloxx13, 2020-01-07:
maybe for @ivankravets

boelle, 2020-01-07:
so this is not a marlin issue?

Dracrius, 2020-01-07:
> 
> 
> so this is not a marlin issue?

I don't think so but I'm hoping someone who know the code better can replicate it and confirm that it is not. At minimum its definitely a bug current users may run into and they will need a solution!

boelle, 2020-01-07:
a bug is defined as when code does not work as it is written to work

boelle, 2020-01-07:
but i see its a problem with platformio and not marlin so will close this one

closed 2020-01-07 by boelle

Dracrius, 2020-01-07:
> 
> 
> a bug is defined as when code does not work as it is written to work

True I wasn't say it was specifically a marlin bug. I more meant its a bug in general. My larger concern after finding the source was that others will waist as much time as I did searching for errors in Marlin if its not obvious it is a known bug with PlatformIO. Hopefully it gets sorted quickly to save other the hassle!